espressif/esp-idf #12281: ESP32-S3 keep rebooting invalid image block, can't boot.

wilian-wagner, 2023-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I made the flashing encrypt process and expected to read a serial message

### What is the actual behavior?

The device is restarting and not booting

### Steps to reproduce.

1. Step espefuse.py --port COM3 burn_key BLOCK_KEY0  C:\Users\Wilian\Desktop\Encrypt_firmware\Encrypt_firmware\flash_encryption_key.bin XTS_AES_128_KEY

2. espsecure.py encrypt_flash_data --aes_xts --keyfile keyfile.bin --output encrypted_partitions.bin partitions.bin --address 0x8000 
3. espsecure.py encrypt_flash_data --aes_xts --keyfile keyfile.bin --output encrypted_bootloader.bin bootloader.bin --address 0x1000 
4.  espsecure.py encrypt_flash_data --aes_xts --keyfile keyfile.bin --output encrypted_firmware.bin firmware.bin --address 0x10000
5. esptool.py --chip ESP32-S3 --port COM3 --baud 921600 --before default_reset --after
                          hard_reset write_flash -z
                          --flash_mode dio --flash_freq 40m --flash_size detect
                          0x1000 encrypted_bootloader.bin 0x8000 encrypted_partitions.bin 0x10000
                          encrypted_firmware.bin

...

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x73e290dd,len:0xdda79069
Invalid image block, can't boot.
ets_main.c 329
ESP-ROM:esp32s3-20210327
```

mahavirj, 2023-09-21:
@wilian-wagner 

>  espsecure.py encrypt_flash_data --aes_xts --keyfile keyfile.bin --output encrypted_bootloader.bin bootloader.bin --address 0x1000

For ESP32-S3 case, the bootloader offset is 0 (`CONFIG_BOOTLOADER_OFFSET_IN_FLASH`). Please update this command and then flash the bootloader at 0 offset.

wilian-wagner, 2023-09-21:
Oh thanks,  the error is not showing anymore, but now i have to understand why the serial monitor is not showing any data... Do you have some idea?
I will let my code here
#include <Arduino.h>

# include "esp_flash_encrypt.h"
void setup() {
  Serial.begin(115200);
}

void loop() {
  if (esp_flash_encryption_enabled())
    Serial.println("Encryption Enabled");
  else
    Serial.println("Encryption not Enabled");
}

wilian-wagner, 2023-09-21:
I've tried via cdc because i read somewhere that with flash encryption the jtag it's diseable

#include <Arduino.h>
#include <USB.h>
#include <USBCDC.h>
#include <HWCDC.h>

#include "esp_flash_encrypt.h"
void setup()
{
  Serial.begin(9600);
}

void loop()
{
  if (esp_flash_encryption_enabled())
  {
    Serial.println("Encryption Enabled");
  }
  else
  {
    Serial.println("Encryption not Enabled");
  }
  delay(1000);
}

Didn't work...

mahavirj, 2023-09-22:
@wilian-wagner 

- Please share the efuse summary of the device
- Please share complete console log, ROM/bootloader level logs were functional, right?

wilian-wagner, 2023-09-22:
Connecting...
Detecting chip type... ESP32-S3

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
TEMP_SENSOR_CAL (BLOCK2)                           ??? Temperature calibration                        = -16.7 R/W (0b110100111)
ADC1_MODE0_D2 (BLOCK2)                             ??? ADC1 calibration 1                             = 300 R/W (0x4b)
ADC1_MODE1_D2 (BLOCK2)                             ??? ADC1 calibration 2                             = 464 R/W (0x74)
ADC1_MODE2_D2 (BLOCK2)                             ??? ADC1 calibration 3                             = -304 R/W (0xcc)
ADC1_MODE3_D2 (BLOCK2)                             ??? ADC1 calibration 4                             = 232 R/W (0x3a)
ADC2_MODE0_D2 (BLOCK2)                             ??? ADC2 calibration 5                             = -352 R/W (0xd8)
ADC2_MODE1_D2 (BLOCK2)                             ??? ADC2 calibration 6                             = 64 R/W (0x10)
ADC2_MODE2_D2 (BLOCK2)                             ??? ADC2 calibration 7                             = -232 R/W (0xba)
ADC2_MODE3_D2 (BLOCK2)                             ??? ADC2 calibration 8                             = 0 R/W (0x00)
ADC1_MODE0_D1 (BLOCK2)                             ??? ADC1 calibration 9                             = 92 R/W (0b010111)
ADC1_MODE1_D1 (BLOCK2)                             ??? ADC1 calibration 10                            = -48 R/W (0b101100)
ADC1_MODE2_D1 (BLOCK2)                             ??? ADC1 calibration 11                            = -56 R/W (0b101110)
ADC1_MODE3_D1 (BLOCK2)                             ??? ADC1 calibration 12                            = 36 R/W (0b001001)
ADC2_MODE0_D1 (BLOCK2)                             ??? ADC2 calibration 13                            = 20 R/W (0b000101)
ADC2_MODE1_D1 (BLOCK2)                             ??? ADC2 calibration 14                            = -48 R/W (0b101100)
ADC2_MODE2_D1 (BLOCK2)                             ??? ADC2 calibration 15                            = 96 R/W (0b011000)
ADC2_MODE3_D1 (BLOCK2)                             ??? ADC2 calibration 16                            = 68 R/W (0b010001)

Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DCACHE (BLOCK0)                                Disables DCache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_DOWNLOAD_DCACHE (BLOCK0)                       Disables Dcache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
DIS_APP_CPU (BLOCK0)                               Disables APP CPU                                   = False R/W (0b0)
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms
DIS_DIRECT_BOOT (BLOCK0)                           Disables direct boot mode                          = False R/W (0b0)
DIS_USB_SERIAL_JTAG_ROM_PRINT (BLOCK0)             Disables USB-Serial-JTAG ROM printing              = False R/W (0b0)
FLASH_ECC_MODE (BLOCK0)                            Configures the ECC mode for SPI flash
   = 16-byte to 18-byte mode R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Disables USB-Serial-JTAG download feature in UART  = False R/W (0b0)
                                                   download boot mode
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Enabled R/W (0b00)
FLASH_TYPE (BLOCK0)                                Selects SPI flash type                             = 4 data lines R/W (0b0)
FLASH_PAGE_SIZE (BLOCK0)                           Sets the size of flash page                        = 0 R/W (0b00)
FLASH_ECC_EN (BLOCK0)                              Enables ECC in Flash boot mode                     = False R/W (0b0)
FORCE_SEND_RESUME (BLOCK0)                         Forces ROM code to send an SPI flash resume comman = False R/W (0b0)
                                                   d during SPI boot
DIS_USB_OTG_DOWNLOAD_MODE (BLOCK0)                 Disables USB-OTG download feature in UART download = False R/W (0b0)
                                                    boot mode
DISABLE_WAFER_VERSION_MAJOR (BLOCK0)               Disables check of wafer version major              = False R/W (0b0)
DISABLE_BLK_VERSION_MAJOR (BLOCK0)                 Disables check of blk version major                = False R/W (0b0)
BLOCK_USR_DATA (BLOCK3)                            User data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS (BLOCK0)                                    Disables programming of individual eFuses          = 8388864 R/W (0x00800100)
[... cut ...]
Writing at 0x0002bfdb... (47 %)
Writing at 0x0002ffd6... (52 %)
Writing at 0x00033fd1... (58 %)
Writing at 0x00037fcc... (64 %)
Writing at 0x0003bfc7... (70 %)
Writing at 0x0003ffc2... (76 %)
Writing at 0x00043fbd... (82 %)
Writing at 0x00047fb8... (88 %)
Writing at 0x0004bfb3... (94 %)
Writing at 0x0004ffae... (100 %)
Wrote 263552 bytes (263643 compressed) at 0x00010000 in 2.1 seconds (effective 994.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

 yep i think they are functional

mahavirj, 2023-09-26:
@wilian-wagner 

> SPI_BOOT_CRYPT_CNT (BLOCK0) Enables encryption and decryption, when an SPI boo = Disable R/W (0b000)
t mode is set. Enabled when 1 or 3 bits are set,di
sabled otherwise

This indicates that flash encryption is not yet enabled on this device. If the config `SECURE_FLASH_ENC_ENABLED` is enabled in the bootloader then it shall attempt to encrypt the artifacts (bootloader, partition-table, application) using the pre-programmed flash enc key. Can you please share the bootloader console log here?

Interim, we have a documentation guide for enabling flash encryption using host assisted workflow [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/host-based-security-workflows.html). Please check this guide once for the correct setup instructions.

wilian-wagner, 2023-09-26:
Hi, I've follow the link you you gave [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/host-based-security-workflows.html) and now i have a new problem hahahahah....  
My console jus show me this, invalid header: 0ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048839
invalid header: 0x15914773
invalid header: 0x15914773

this is on loop, and read that it's a commom issue (If the bootloader partition is re-flashed with a plaintext firmware bootloader image, the ROM bootloader will fail to load the firmware bootloader resulting in the following failure:) but i don't know how to solve it.

I don't know how to get the bootloader log too, if you show me how i can send it...

here it's the summary just toyou see if help:

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
TEMP_SENSOR_CAL (BLOCK2)                           ??? Temperature calibration                        = -16.7 R/W (0b110100111)
ADC1_MODE0_D2 (BLOCK2)                             ??? ADC1 calibration 1                             = 300 R/W (0x4b)
ADC1_MODE1_D2 (BLOCK2)                             ??? ADC1 calibration 2                             = 464 R/W (0x74)
ADC1_MODE2_D2 (BLOCK2)                             ??? ADC1 calibration 3                             = -304 R/W (0xcc)
ADC1_MODE3_D2 (BLOCK2)                             ??? ADC1 calibration 4                             = 232 R/W (0x3a)
ADC2_MODE0_D2 (BLOCK2)                             ??? ADC2 calibration 5                             = -352 R/W (0xd8)
ADC2_MODE1_D2 (BLOCK2)                             ??? ADC2 calibration 6                             = 64 R/W (0x10)
ADC2_MODE2_D2 (BLOCK2)                             ??? ADC2 calibration 7                             = -232 R/W (0xba)
ADC2_MODE3_D2 (BLOCK2)                             ??? ADC2 calibration 8                             = 0 R/W (0x00)
ADC1_MODE0_D1 (BLOCK2)                             ??? ADC1 calibration 9                             = 92 R/W (0b010111)
ADC1_MODE1_D1 (BLOCK2)                             ??? ADC1 calibration 10                            = -48 R/W (0b101100)
ADC1_MODE2_D1 (BLOCK2)                             ??? ADC1 calibration 11                            = -56 R/W (0b101110)
ADC1_MODE3_D1 (BLOCK2)                             ??? ADC1 calibration 12                            = 36 R/W (0b001001)
ADC2_MODE0_D1 (BLOCK2)                             ??? ADC2 calibration 13                            = 20 R/W (0b000101)
ADC2_MODE1_D1 (BLOCK2)                             ??? ADC2 calibration 14                            = -48 R/W (0b101100)
ADC2_MODE2_D1 (BLOCK2)                             ??? ADC2 calibration 15                            = 96 R/W (0b011000)
ADC2_MODE3_D1 (BLOCK2)                             ??? ADC2 calibration 16                            = 68 R/W (0b010001)

Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DCACHE (BLOCK0)                                Disables DCache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_DOWNLOAD_DCACHE (BLOCK0)                       Disables Dcache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
DIS_APP_CPU (BLOCK0)                               Disables APP CPU                                   = False R/W (0b0)
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms
DIS_DIRECT_BOOT (BLOCK0)                           Disables direct boot mode                          = False R/W (0b0)
DIS_USB_SERIAL_JTAG_ROM_PRINT (BLOCK0)             Disables USB-Serial-JTAG ROM printing              = False R/W (0b0)
FLASH_ECC_MODE (BLOCK0)                            Configures the ECC mode for SPI flash
   = 16-byte to 18-byte mode R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Disables USB-Serial-JTAG download feature in UART  = False R/W (0b0)
                                                   download boot mode
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Enabled R/W (0b00)
FLASH_TYPE (BLOCK0)                                Selects SPI flash type                             = 4 data lines R/W (0b0)
FLASH_PAGE_SIZE (BLOCK0)                           Sets the size of flash page                        = 0 R/W (0b00)
FLASH_ECC_EN (BLOCK0)                              Enables ECC in Flash boot mode                     = False R/W (0b0)
FORCE_SEND_RESUME (BLOCK0)                         Forces ROM code to send an SPI flash resume comman = False R/W (0b0)
                                                   d during SPI boot
[... cut ...]
VDD_SPI_FORCE (BLOCK0)                             Force using VDD_SPI_XPD and VDD_SPI_TIEH to config = True R/W (0b1)
                                                   ure VDD_SPI LDO
PIN_POWER_SELECTION (BLOCK0)                       Sets default power supply for GPIO33..37           = VDD3P3_CPU R/W (0b0)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = 0 R/W (0b00)

Flash voltage (VDD_SPI) set to 3.3V by efuse.

thanks in advance

mahavirj, 2023-09-27:
> rst:0x7 (TG0WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048839
invalid header: 0x15914773
invalid header: 0x15914773

Since your device has not yet enabled the flash encryption (`SPI_BOOT_CRYPT_CNT` being 0), it expects plain-text artifacts like bootloader, partition-table and application. If you are flashing encrypted bootloader, then you can get this error. 

Please read this [guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/flash-encryption.html) carefully before you experiment with the flash encryption feature.

mahavirj, 2023-10-09:
Closing due to lack of further update. Please feel to re-open if you are still facing any issue.

closed 2023-10-09 by mahavirj

Shayantika-Dhar, 2024-01-05:
yesterday the module was working fine with my code, but when i start using today the same code is giving error and now this is showing

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

Shayantika-Dhar, 2024-01-05:
step wise help will be very helpful to me