espressif/esp-idf #12282: ESP32 Time Clock runs slow under certain conditions with Bluetooth Classic HF client

danjulio, 2023-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-439-gc25d92074e

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Hardware

### Power Supply used.

USB

### What is the expected behavior?

I expect the system time as reported by gettimeofday() to remain fairly accurate during continuous operation of the ESP32 in a communication device that uses the Bluetooth Classic Handsfree Client Protocol to connect to a cellphone.

### What is the actual behavior?

Under poor Bluetooth conditions (the phone is at the very edge of Bluetooth range) then the time clock running on the ESP32 can lose many tens of seconds per hour of real time. 

This does not occur when the Bluetooth connection is good (e.g. the phone is in the same room as the device) or not present (e.g. the phone is so remote that the ESP32 can never connect).

See "More Information" for additional details.

### Steps to reproduce.

The program is designed to run continuously (the ESP32 is not put to sleep at any time).  When not connected to a Bluetooth device it will attempt a connection every minute.  The following assumes the device has been paired with a cellphone (iPhone 13 iOS 16.6.1).

1. Set the ESP32 time clock using settimeofday().  At the same time set the time in an accurate battery-backed RTC chip attached to the ESP32.
2. Locate the cellphone in another room (perhaps 20-25m away with two intervening walls and an intervening floor)
3. The program attempts to connect and if successful it attempts to maintain a Bluetooth connection with the cellphone.
4. Once per hour compare the ESP32 time clock time obtained with gettimeofday() with that of the external hardware RTC chip.  Display an error message and correct the ESP32 time clock if it is more than 10 seconds different.

The external hardware RTC chip has been verified to hold the correct time.

### Debug Logs.

```plain
ELF file SHA256: b3778f3f1e013f3a

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6968
load:0x40078000,len:15792
load:0x40080400,len:4516
0x40080400: _init at ??:?

[... 367 lines cut ...]
I (28804396) bt_hf: --battery level 5
I (28804396) bt_hf: APP HFP event: SIGNAL_STRENGTH_IND_EVT
I (28804406) bt_hf: -- signal strength: 1
I (28804406) bt_hf: APP HFP event: ROAMING_STATUS_IND_EVT
I (28804416) bt_hf: --ROAMING: inactive
I (28804416) bt_hf: APP HFP event: CALL_HELD_IND_EVT
I (28804426) bt_hf: --Call held indicator NONE held
I (28804436) bt_hf: APP HFP event: CONNECTION_STATE_EVT
I (28804436) bt_hf: --connection state slc_connected, peer feats 0xfef, chld_feats 0x3f
I (28804446) bt_hf: APP HFP event: INBAND_RING_TONE_EVT
I (28804456) bt_hf: --inband ring state Provided
I (28804456) bt_task: DISCONNECTED->CONNECTED-IDLE
I (28804506) app_task: DISCONNECTED->CONNECTED_IDLE
I (28807306) gcore_task: Vusb: 5.11v, Iusb: 235mA, Vbatt: 4.18v, Iload: 237mA, Chg: 2
E (28807306) gcore_task: Correcting ESP32 time (delta = -54)
I (28807316) time_utilities: Set time: Wed Sep 20, 2023  5:49:42  (epoch secs: 1695188982)
```

### More Information.

The log file contains a section of an overnight run showing approximately 1 hour of elapsed time during which the ESP32 loses 54 seconds (it is 54 seconds slower than the hardware RTC).  [NOTE I had to cut out some of the log file to post this].  Interestingly the log file millisecond time stamps do appear to be keeping accurate time.  At time 25205186 mSec the code corrects the ESP32 time clock.  For the next hour you can see it struggles to maintain a connection with the phone.  About one hour later, at 28807306 mSec, it checks the ESP32 time clock against the RTC and finds it is 54 seconds slow.

My Bluetooth code is based on the Espressif Bluetooth Classic Handsfree client demo code and I retained the logging information for the various GAP (bt_gap) and HF (bt_hf) callbacks that you can see in the log file.   My bluetooth handling code can be found here: [BT Task](https://github.com/danjulio/weeBell_bluetooth/blob/main/gcore_pots_bt/main/bt_task.c).

Again, if the devices are physically close and the Bluetooth SLC remains valid or if the devices are too far apart (or the device pairing information is deleted) then the clock maintains accurate time.  It seems only when the Bluetooth stack is struggling to maintain a connection does the clock slow down (it never runs fast).

KonstantinKondrashov, 2023-10-03:
Hi @danjulio!
Could you clarify the two things?
1. Is sleep used in any way by the chip or not?
2. Every hour do you see -54 seconds? How often is it seen?

danjulio, 2023-10-03:
Thanks for looking at this @KonstantinKondrashov.  

1. Sleep is not used.   ESP32 is running 100% of the time.
2. It is seen during periods of time where the Bluetooth stack is having difficulty maintaining a connection with the cellphone (long distance).  It is disconnecting and attempting to reconnect frequently.  If cellphone is close or completely absent then there is no problem.

I realize this is not a very detailed description.  Please let me know if I can provide more data.