espressif/esp-idf #12283: memset in ROM exception when plug one ESP32-S3 in BOOT mode to another ESP32-S3 OTG

Bumanuu, 2023-09-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1890-g28167ea5a3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

BPI-Leaf-S3

### Power Supply used.

USB

### What is the expected behavior?

Using _esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp_ slightly modified example example.
This is normal behavior when I plug device with CP2102 Virtual COM Port chip in :
[CP2102-VCP_.log](https://github.com/espressif/esp-idf/files/12682316/CP2102-VCP_.log)

### What is the actual behavior?

When plug another ESP32-S3 in BOOT mode to flashed with example above firmware first ESP32-S3 OTG port:
[ESP32-S3-VCP.log](https://github.com/espressif/esp-idf/files/12682384/ESP32-S3-VCP.log)

Linux detect those devices in BOOT mode as:
`[17269.589717] usb 1-3.4: new full-speed USB device number 7 using xhci_hcd
[17269.711121] usb 1-3.4: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[17269.711136] usb 1-3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[17269.711143] usb 1-3.4: Product: USB JTAG/serial debug unit
[17269.711149] usb 1-3.4: Manufacturer: Espressif
[17269.711154] usb 1-3.4: SerialNumber: 48:27:E2:0D:7F:C4
[17269.713425] cdc_acm 1-3.4:1.0: ttyACM0: USB ACM device
`

### Steps to reproduce.

1. Move _esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/managed_components/espressif__usb_host_ch34x_vcp_ folder to _esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/components/espressif__usb_host_ch34x_vcp_
2. Modify _esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/components/espressif__usb_host_ch34x_vcp/include/usb/vcp_ch34x.hpp_ like that:
`//#define NANJING_QINHENG_MICROE_VID (0x1A86)
//#define CH340_PID                  (0x7522)
#define NANJING_QINHENG_MICROE_VID (0x303A)
#define CH340_PID                  (0x1001)
`
3. Build and flash one ESP32-S3 (#1) with _esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp_
4. Plug in with USB cable another ESP32-S3 (#2) in BOOT mode (BOOT button is pressed or pin 0 is connected to GND) to #1 ESP32-S3 USB-OTG socket.
5. ESP32-S3 #1 is reporting about exception and rebooting cyclically.

### Debug Logs.

```plain
bumanuu@bumanuux:~/DEV/GIT/esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp$ /home/bumanuu/.espressif/python_env/idf5.2_py3.10_env/bin/python /home/bumanuu/DEV/GIT/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /home/bumanuu/DEV/GIT/esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/build/cdc_acm_vcp.elf
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[... 97 lines cut ...]

Backtrace: 0x4200ac49:0x3fc99410 0x4200b20b:0x3fc99470 0x42009bda:0x3fc994c0 0x42009391:0x3fc994e0 0x420093bd:0x3fc99500 0x4200a4be:0x3fc99520 0x42009969:0x3fc99590 0x4202790b:0x3fc99600 0x4037c55d:0x3fc99630
0x4200ac49: cdc_acm_transfers_allocate at /home/bumanuu/DEV/GIT/esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/managed_components/espressif__usb_host_cdc_acm/cdc_acm_host.c:624

0x4200b20b: cdc_acm_host_open_vendor_specific at /home/bumanuu/DEV/GIT/esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/managed_components/espressif__usb_host_cdc_acm/cdc_acm_host.c:859 (discriminator 6)

0x42009bda: CdcAcmDevice::open_vendor_specific(unsigned short, unsigned short, unsigned char, cdc_acm_host_device_config_t const*) at /home/bumanuu/DEV/GIT/esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/managed_components/espressif__usb_host_cdc_acm/include/usb/cdc_acm_host.h:317
 (inlined by) esp_usb::CH34x::CH34x(unsigned short, cdc_acm_host_device_config_t const*, unsigned char) at /home/bumanuu/DEV/GIT/esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/components/espressif__usb_host_ch34x_vcp/usb_host_ch34x_vcp.cpp:71

0x42009391: esp_usb::VCP::register_driver<esp_usb::CH34x>()::{lambda(unsigned short, cdc_acm_host_device_config_t const*, unsigned char)#1}::operator()(unsigned short, cdc_acm_host_device_config_t const*, unsigned char) const at /home/bumanuu/DEV/GIT/esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/managed_components/espressif__usb_host_vcp/include/usb/vcp.hpp:52

0x420093bd: esp_usb::VCP::register_driver<esp_usb::CH34x>()::{lambda(unsigned short, cdc_acm_host_device_config_t const*, unsigned char)#1}::_FUN(unsigned short, cdc_acm_host_device_config_t const*, unsigned char) at /home/bumanuu/DEV/GIT/esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/managed_components/espressif__usb_host_vcp/include/usb/vcp.hpp:51

0x4200a4be: esp_usb::VCP::open(cdc_acm_host_device_config_t const*, unsigned char) at /home/bumanuu/DEV/GIT/esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/managed_components/espressif__usb_host_vcp/usb_host_vcp.cpp:72

0x42009969: app_main at /home/bumanuu/DEV/GIT/esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_vcp/main/cdc_acm_vcp_example_main.cpp:144

0x4202790b: main_task at /home/bumanuu/DEV/GIT/esp-idf/components/freertos/app_startup.c:217 (discriminator 13)

0x4037c55d: vPortTaskWrapper at /home/bumanuu/DEV/GIT/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: f494c561e

Rebooting...
```

tore-espressif, 2023-11-21:
Hello @Bumanuu 

I'll check the abort condition in cdc_acm_host component.

Meanwhile, the cdc_acm_vcp example is intended for CDC-like devices, that implement vendor specific class. Like the CP2102.

If you want to experiment with the original CDC-ACM driver, this example might work better for you https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_host

tore-espressif, 2023-11-21:
We released version 2.0.2 of [usb_host_cdc_acm](https://components.espressif.com/components/espressif/usb_host_cdc_acm) that will not crash on this condition.

However the device will not be opened. Please try the [cdc_acm_host example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_host). Modify VID and PID in the beginning of the main.c file, it should work OK with your USB device.

tore-espressif, 2023-12-05:
Please feel free to reopen if the problem persists

closed 2023-12-05 by tore-espressif