espressif/esp-idf #12284: PCNT migration from legacy to idf5.1.1 version problem

damian-kurek-wizzdev, 2023-09-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have an existing project and want to migrate it to idf 5.1.1. I ran into some problems with the new PCNT implementation.
My legacy configuration looked like this one:
`    pcnt_unit_t   unit        = PCNT_UNIT_0;
    pcnt_config_t pcnt_config = {
        // Set PCNT input signal and control GPIOs
        .pulse_gpio_num = MOTOR_FEEDBACK_PIN,
        .ctrl_gpio_num  = PCNT_PIN_NOT_USED,
        .lctrl_mode     = PCNT_MODE_KEEP,
        .hctrl_mode     = PCNT_MODE_KEEP,
        .pos_mode       = PCNT_COUNT_INC, // Count up on the positive edge
        .neg_mode       = PCNT_COUNT_DIS, // Keep the counter value on the negative edge
        .counter_h_lim  = PCNT_H_LIM_VAL,
        .counter_l_lim  = PCNT_L_LIM_VAL,
        .unit           = unit,
        .channel        = PCNT_CHANNEL_0,
    };
    // Initialize PCNT unit
    pcnt_unit_config(&pcnt_config);

    // Configure and enable the input filter,
    pcnt_set_filter_value(unit, PCNT_FILTER_VALUE_IN_APB_CYCLES);
    pcnt_filter_enable(unit);

    // Enable event on rising edge
    pcnt_set_event_value(unit, PCNT_EVT_THRES_1, PCNT_THRESH1_VAL);
    pcnt_event_enable(unit, PCNT_EVT_THRES_1);

    // Initialize PCNT's counter
    pcnt_counter_pause(unit);
    pcnt_counter_clear(unit);`

My new configuration:
`    pcnt_unit_config_t unitConfig{
        .low_limit = PCNT_L_LIM_VAL, .high_limit = PCNT_H_LIM_VAL, .flags = {.accum_count = 1}};
    pcnt_chan_config_t chanelConfig{
        .edge_gpio_num  = MOTOR_FEEDBACK_PIN,
        .level_gpio_num = -1,
        .flags          = {
            .invert_edge_input  = 1, /*!< Invert the input edge signal */
            .invert_level_input = 1, /*!< Invert the input level signal */
            .virt_edge_io_level =
                0, /*!< Virtual edge IO level, 0: low, 1: high. Only valid when edge_gpio_num is set to -1 */
            .virt_level_io_level =
                0, /*!< Virtual level IO level, 0: low, 1: high. Only valid when level_gpio_num is set to -1 */
            .io_loop_back = 1,
        }};
    pcnt_glitch_filter_config_t glitchFilterConfig{.max_glitch_ns = GLITCH_FILTER_TIME_NS};

    auto result = pcnt_new_unit(&unitConfig, &motorControl.pPcntUnitHandle);
    if (result != ESP_OK)
    {
        LOG_ERROR("New unit creation failed ");
    }
    result = pcnt_new_channel(motorControl.pPcntUnitHandle, &chanelConfig, &motorControl.pPcntChannelHandle);
    if (result != ESP_OK)
    {
        LOG_ERROR("New channel creation failed ");
    }

    //     Configure and enable the input filter,
    //        result = pcnt_unit_set_glitch_filter(motorControl.pPcntUnitHandle, &glitchFilterConfig);
    //        if (result != ESP_OK)
    //        {
    //            LOG_ERROR("New set glitch filter creation failed ");
    //        }
    //     Enable event on rising edge
    result = pcnt_channel_set_edge_action(
        motorControl.pPcntChannelHandle, PCNT_CHANNEL_EDGE_ACTION_INCREASE, PCNT_CHANNEL_EDGE_ACTION_HOLD);
    if (result != ESP_OK)
    {
        LOG_ERROR("New set edge action creation failed ");
    }

    result = pcnt_unit_enable(motorControl.pPcntUnitHandle);
    if (result != ESP_OK)
    {
        LOG_ERROR("Add unit enable failed ");
    }

    result = pcnt_unit_clear_count(motorControl.pPcntUnitHandle);
    if (result != ESP_OK)
    {
        LOG_ERROR("Add unit clear count ");
    }

    result = pcnt_unit_start(motorControl.pPcntUnitHandle);
    if (result != ESP_OK)
    {
        LOG_ERROR("Start unit ");
    }` 
    
I read the current count using pcnt_unit_get_count -> new idf and pcnt_get_counter_value in legacy code.
The problem I face is that the new implementation always returns 0. Do you have any idea how to fix it or any example of how to migrate from 4.4 config to 5.1. I checked the migration guide already.
I noticed already that new implementation low limit needs to be lower than 0 in legacy it could be set to 0

suda-morris, 2023-09-22:
I think you're setting too many parameters that are not necessary for you. Also I didn't see you call `pcnt_channel_set_level_action`.

Please try the following code:

```c
    pcnt_unit_config_t unit_config = {
        .low_limit = -100,
        .high_limit = 100,
    };
    pcnt_chan_config_t chan_config = {
        .edge_gpio_num = MOTOR_FEEDBACK_PIN, 
        .level_gpio_num = -1,              // no gpio connected t other level signal
    };
    pcnt_unit_handle_t unit = NULL;
    pcnt_channel_handle_t chan = NULL;

    printf("install pcnt unit\r\n");
    pcnt_new_unit(&unit_config, &unit);

    printf("install pcnt channel\r\n");
    pcnt_new_channel(unit, &chan_config, &chan);
    pcnt_channel_set_edge_action(chan, PCNT_CHANNEL_EDGE_ACTION_INCREASE, PCNT_CHANNEL_EDGE_ACTION_HOLD);
    pcnt_channel_set_level_action(chan, PCNT_CHANNEL_LEVEL_ACTION_KEEP, PCNT_CHANNEL_LEVEL_ACTION_KEEP);
    pcnt_unit_enable(unit);

    int count_value = 0;
    printf("start units\r\n");
    // start unit
    pcnt_unit_start(unit);
```

damian-kurek-wizzdev, 2023-09-22:
OK, I will test it but, 
Why do I need pcnt_channel_set_level_action when in channel config I have .level_gpio_num = -1?
Parameters are not necessary, but I run clang static analysis for this project and it will drop errors for uninitialized parameters, but they are set to default and shouldn't matter.

damian-kurek-wizzdev, 2023-09-28:
I figured it out. Loop back should be set to 0.
.io_loop_back = 0,