espressif/esp-idf #12285: esp_ble_gattc_get_all_char wont find any characteristics

Svahnen-Perific, 2023-09-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The project needs to use 4.4.5 since we are using arduino code in other places.

Firstly we use esp_ble_gattc_search_service(gattc_if, param->cfg_mtu.conn_id, NULL); inside ESP_GATTC_OPEN_EVT to get all the services and save them and their start and end handles, after that inside ESP_GATTC_SEARCH_CMPL_EVT we use a for loop that will go through all the services and run esp_ble_gattc_get_all_char. However the char_count is always 0, even though we know there are characteristics to be found.

the handler looks like this:
```
case ESP_GATTC_SEARCH_RES_EVT: {
    // This event is triggered for each service found during service discovery
    ESP_LOGI(GATTC_TAG, "SEARCH RES: conn_id = %x is primary service %d",
             p_data->search_res.conn_id, p_data->search_res.is_primary);
    ESP_LOGI(GATTC_TAG, "start handle %d end handle %d current handle value %d",
             p_data->search_res.start_handle, p_data->search_res.end_handle,
             p_data->search_res.srvc_id.inst_id);

    // TODO: check if its a primary service and save to services, if not it
    // should be a characteristic and saved under the correct service
    if (p_data->search_res.is_primary) {

      // Increment the number of services
      gl_profile_tab[PROFILE_A_APP_ID].num_services++;

      // Reallocate memory for the new number of services
      gl_profile_tab[PROFILE_A_APP_ID].services = (service_t *)realloc(
          gl_profile_tab[PROFILE_A_APP_ID].services,
          gl_profile_tab[PROFILE_A_APP_ID].num_services * sizeof(service_t));

      // Initialize the new service
      service_t *new_service =
          &gl_profile_tab[PROFILE_A_APP_ID]
               .services[gl_profile_tab[PROFILE_A_APP_ID].num_services - 1];
      new_service->start_handle = p_data->search_res.start_handle;
      new_service->end_handle = p_data->search_res.end_handle;
      new_service->characteristics = NULL;
      new_service->num_characteristics = 0;

      // Handle UUID based on its length
      if (p_data->search_res.srvc_id.uuid.len == ESP_UUID_LEN_16) {
        new_service->uuid.len = ESP_UUID_LEN_16;
        new_service->uuid.uuid.uuid16 =
            p_data->search_res.srvc_id.uuid.uuid.uuid16;
        printf("service found uuid 16, uuid is 0x%04x\n",
               p_data->search_res.srvc_id.uuid.uuid.uuid16);
      } else if (p_data->search_res.srvc_id.uuid.len == ESP_UUID_LEN_128) {
        new_service->uuid.len = ESP_UUID_LEN_128;
        memcpy(new_service->uuid.uuid.uuid128,
               p_data->search_res.srvc_id.uuid.uuid.uuid128, ESP_UUID_LEN_128);

        printf("service found uuid 128, uuid is:\n");
        for (int i = 0; i < ESP_UUID_LEN_128; i++) {
          printf("%02x\n", new_service->uuid.uuid.uuid128[i]);
        }
        printf("\n");
      }
    } else {
      printf("Entered ESP_GATTC_SEARCH_RES_EVT when not primary service\n");
      // TODO: If not primary save under characteristic instead
      // Se if we can print what service a characteristic came from
      // Handle UUID based on its length
      if (p_data->search_res.srvc_id.uuid.len == ESP_UUID_LEN_16) {
        printf("Characteristic found uuid 16, uuid is 0x%04x\n",
               p_data->search_res.srvc_id.uuid.uuid.uuid16);
      } else if (p_data->search_res.srvc_id.uuid.len == ESP_UUID_LEN_128) {

        printf("Characteristic found uuid 128, uuid is:\n");
        for (int i = 0; i < ESP_UUID_LEN_128; i++) {
          printf("%02x\n", p_data->search_res.srvc_id.uuid.uuid.uuid128[i]);
        }
        printf("\n");
      }
    }

    break;
  }

  case ESP_GATTC_SEARCH_CMPL_EVT: {
    printf("We got inside ESP_GATTC_SEARCH_CMPL_EVT\n");

    // Debug testing Start ----------------------------------
    // This event is triggered when service discovery is complete
    ESP_LOGI(GATTC_TAG, "SEARCH COMPLETE: conn_id = %x",
             p_data->search_cmpl.conn_id);

    // Loop through all the services and print their details
    for (size_t i = 0; i < gl_profile_tab[PROFILE_A_APP_ID].num_services; ++i) {
      service_t *service = &gl_profile_tab[PROFILE_A_APP_ID].services[i];
      printf("Service #%zu:\n", i + 1);
      printf("  Start Handle: %d\n", service->start_handle);
      printf("  End Handle: %d\n", service->end_handle);

      // Print UUID based on its length
      if (service->uuid.len == ESP_UUID_LEN_16) {
        printf("  UUID: 0x%04x (16-bit)\n", service->uuid.uuid.uuid16);
      } else if (service->uuid.len == ESP_UUID_LEN_128) {
        printf("  UUID: ");
        for (int j = 0; j < ESP_UUID_LEN_128; ++j) {
          printf("%02x", service->uuid.uuid.uuid128[j]);
        }
        printf(" (128-bit)\n");
      }
    }
    // Debug testing Stop ----------------------------------
    if (p_data->search_cmpl.status != ESP_GATT_OK) {
      ESP_LOGE(GATTC_TAG, "search service failed, error status = %x",
               p_data->search_cmpl.status);
      break;
    }

    if (p_data->search_cmpl.searched_service_source ==
        ESP_GATT_SERVICE_FROM_REMOTE_DEVICE) {
      ESP_LOGI(GATTC_TAG, "Get service information from remote device");
    } else if (p_data->search_cmpl.searched_service_source ==
               ESP_GATT_SERVICE_FROM_NVS_FLASH) {
      ESP_LOGI(GATTC_TAG, "Get service information from flash");
    } else {
      ESP_LOGI(GATTC_TAG, "unknown service source");
    }

    ESP_LOGI(GATTC_TAG, "ESP_GATTC_SEARCH_CMPL_EVT");

    esp_gattc_char_elem_t
        char_elem[CONFIG_BT_GATTC_MAX_CACHE_CHAR]; // This value is set in
                                                   // menuconfig

    // Loop through all services
    for (int service_index = 0;
         service_index < gl_profile_tab[PROFILE_A_APP_ID].num_services;
         ++service_index) {
      uint16_t char_count = 0;
      esp_err_t ret = esp_ble_gattc_get_all_char(
          gattc_if, param->search_cmpl.conn_id,
          gl_profile_tab[PROFILE_A_APP_ID].services[service_index].start_handle,
          gl_profile_tab[PROFILE_A_APP_ID].services[service_index].end_handle,
          char_elem, &char_count,
          0 // Offset, set to 0 to get from the beginning
      );

      printf("For service index %d, ret = %d, char_count = %d\n", service_index,
             ret, char_count);

      if (ret != ESP_FAIL) {
        for (int i = 0; i < char_count; ++i) {
          printf("We can now save characteristics for service index %d\n",
                 service_index);
          // Save the characteristics here

          // Print the details of each characteristic element
          printf("Characteristic #%d:\n", i + 1);
          printf("  Handle: %d\n", char_elem[i].char_handle);
          printf("  Properties: %d\n", char_elem[i].properties);
          // Add more fields here if you want to print additional information
        }
      }
    }
```
    
The profile structure looks like this:
```
typedef struct {
  esp_bt_uuid_t uuid;
  uint16_t handle;
} characteristic_t;

typedef struct {
  esp_bt_uuid_t uuid;
  uint16_t start_handle;
  uint16_t end_handle;
  characteristic_t *characteristics; // Dynamic array of characteristics
  size_t num_characteristics;        // Number of characteristics
} service_t;

struct gattc_profile_inst {
  esp_gattc_cb_t gattc_cb;
  uint16_t gattc_if;
  uint16_t app_id;
  uint16_t conn_id;
  esp_bd_addr_t remote_bda;
  service_t *services; // Dynamic array of services
  size_t num_services; // Number of services
};
```

the log looks like this:
```
We got inside ESP_GATTC_SEARCH_CMPL_EVT
[0;32mI (16261) GATTC_ENEGIC: SEARCH COMPLETE: conn_id = 0[0m
Service #1:
  Start Handle: 1
  End Handle: 8
  UUID: 0x1801 (16-bit)
Service #2:
  Start Handle: 9
  End Handle: 15
  UUID: 0x1800 (16-bit)
Service #3:
  Start Handle: 16
  End Handle: 29
  UUID: d44c65547e8f27907c41b18701c17fa5 (128-bit)
Service #4:
  Start Handle: 30
  End Handle: 40
  UUID: 0x180a (16-bit)
Service #5:
  Start Handle: 41
  End Handle: 46
  UUID: 9ecadc240ee5a9e093f3a3b50100406e (128-bit)
Service #6:
  Start Handle: 47
  End Handle: 50
  UUID: 84aa6074528a8b86d34cb71d1ddc538d (128-bit)
[0;32mI (16313) GATTC_ENEGIC: Get service information from remote device[0m
[0;32mI (16320) GATTC_ENEGIC: ESP_GATTC_SEARCH_CMPL_EVT[0m
For service index 0, ret = 4, char_count = 0
For service index 1, ret = 4, char_count = 0
For service index 2, ret = 4, char_count = 0
For service index 3, ret = 4, char_count = 0
For service index 4, ret = 4, char_count = 0
For service index 5, ret = 4, char_count = 0
event: 18
[0;32mI (16350) GATTC_ENEGIC: ESP_GATTC_CFG_MTU_EVT
```

zhp0406, 2023-09-22:
@Svahnen-Perific 
I will check it ASAP.

zhp0406, 2023-09-22:
@Svahnen-Perific 
Since I don't have your complete code, I couldn't perform a full test of your program. Therefore, I conducted testing using `esp-idf/examples/bluetooth/bluedroid/ble/gatt_client` and `esp-idf/examples/bluetooth/bluedroid/ble/gatt_server`. I didn't make any modifications to the `esp-idf/examples/bluetooth/bluedroid/ble/gatt_server` part; I simply compiled and flashed it.

As for `idf/examples/bluetooth/bluedroid/ble/gatt_client`, I made the following modifications:

```
    case ESP_GATTC_SEARCH_CMPL_EVT:
        if (p_data->search_cmpl.status != ESP_GATT_OK){
            ESP_LOGE(GATTC_TAG, "search service failed, error status = %x", p_data->search_cmpl.status);
            break;
        }
        if(p_data->search_cmpl.searched_service_source == ESP_GATT_SERVICE_FROM_REMOTE_DEVICE) {
            ESP_LOGI(GATTC_TAG, "Get service information from remote device");
        } else if (p_data->search_cmpl.searched_service_source == ESP_GATT_SERVICE_FROM_NVS_FLASH) {
            ESP_LOGI(GATTC_TAG, "Get service information from flash");
        } else {
            ESP_LOGI(GATTC_TAG, "unknown service source");
        }
        ESP_LOGI(GATTC_TAG, "ESP_GATTC_SEARCH_CMPL_EVT");
        if (get_server){
            uint16_t count = 0;
            esp_gatt_status_t status = esp_ble_gattc_get_attr_count( gattc_if,
                                                                     p_data->search_cmpl.conn_id,
                                                                     ESP_GATT_DB_CHARACTERISTIC, // the attribute type
                                                                     gl_profile_tab[PROFILE_A_APP_ID].service_start_handle,
                                                                     gl_profile_tab[PROFILE_A_APP_ID].service_end_handle,
                                                                     INVALID_HANDLE,
                                                                     &count);
            if (status != ESP_GATT_OK){
                ESP_LOGE(GATTC_TAG, "esp_ble_gattc_get_attr_count error");
                break;
            }

            if (count > 0) {
                char_elem_result = (esp_gattc_char_elem_t *)malloc(sizeof(esp_gattc_char_elem_t) * count);
                if (!char_elem_result) {
                    ESP_LOGE(GATTC_TAG, "gattc no mem");
                    break;
                }
                else {
                    esp_gatt_status_t ret = esp_ble_gattc_get_all_char(gattc_if,
                                                                       p_data->search_cmpl.conn_id,
                                                                       gl_profile_tab[PROFILE_A_APP_ID].service_start_handle,
                                                                       gl_profile_tab[PROFILE_A_APP_ID].service_end_handle,
                                                                       char_elem_result,
                                                                       &count,
                                                                       0);
                    if (ret != ESP_GATT_OK) {
                        ESP_LOGE(GATTC_TAG, "esp_ble_gattc_get_all_char error");
                        free(char_elem_result);
                        char_elem_result = NULL;
                        break;
                    }

                    /*  Every service have only one char in our 'ESP_GATTS_DEMO' demo, so we used first 'char_elem_result' */
                    if (ret != ESP_FAIL) {
                        for (int i = 0; i < count; i++) {
                            // Print the details of each characteristic element
                            printf("Characteristic #%d:\n", i);
                            printf("  Handle: %d\n", char_elem_result[i].char_handle);
                            printf("  Properties: %d\n", char_elem_result[i].properties);
                            // Add more fields here if you want to print additional information
                        }
                    }
                    /* free char_elem_result */
                    free(char_elem_result);
                }
            }
            else
            {
[... 65 lines cut ...]
I (748) GATTC_DEMO: 

I (748) GATTC_DEMO: searched device ESP_GATTS_DEMO

I (748) GATTC_DEMO: connect to the remote device.
I (758) GATTC_DEMO: stop scan successfully
I (808) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (808) GATTC_DEMO: REMOTE BDA:
I (808) GATTC_DEMO: 84 f7 03 67 48 8e 
I (808) GATTC_DEMO: open success
I (1138) GATTC_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
I (2258) GATTC_DEMO: discover service complete conn_id 0
I (2268) GATTC_DEMO: SEARCH RES: conn_id = 0 is primary service 1
I (2268) GATTC_DEMO: start handle 40 end handle 43 current handle value 40
I (2268) GATTC_DEMO: service found
I (2278) GATTC_DEMO: UUID16: ff
I (2278) GATTC_DEMO: Get service information from remote device
I (2288) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT
Characteristic #0:
  Handle: 42
  Properties: 26
I (2338) GATTC_DEMO: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 500, conn_id 0
```

**I seems ok.**

zhp0406, 2023-09-22:
@Svahnen-Perific 
Could you please print the parameters for esp_ble_gattc_get_all_char to assist with the analysis? For instance:
start_handle, end_handle, count.

Svahnen-Perific, 2023-09-22:
Here are some additional prints:
```
event: 6
We got inside ESP_GATTC_SEARCH_CMPL_EVT
[0;32mI (20653) GATTC_ENEGIC: SEARCH COMPLETE: conn_id = 0[0m
Service #1:
  Start Handle: 1
  End Handle: 8
  UUID: 0x1801 (16-bit)
Service #2:
  Start Handle: 9
  End Handle: 15
  UUID: 0x1800 (16-bit)
Service #3:
  Start Handle: 16
  End Handle: 29
  UUID: d44c65547e8f27907c41b18701c17fa5 (128-bit)
Service #4:
  Start Handle: 30
  End Handle: 40
  UUID: 0x180a (16-bit)
Service #5:
  Start Handle: 41
  End Handle: 46
  UUID: 9ecadc240ee5a9e093f3a3b50100406e (128-bit)
Service #6:
  Start Handle: 47
  End Handle: 50
  UUID: 84aa6074528a8b86d34cb71d1ddc538d (128-bit)
[0;32mI (20705) GATTC_ENEGIC: Get service information from remote device[0m
[0;32mI (20712) GATTC_ENEGIC: ESP_GATTC_SEARCH_CMPL_EVT[0m
Debug github issue: gattc_if=1, conn_id=0, start_handle=1, end_handle=8, char_elem=0x3fff5b70, char_count=0, offset=0
For service index 0, ret = 4, char_count = 0
Debug github issue: gattc_if=1, conn_id=0, start_handle=9, end_handle=15, char_elem=0x3fff5b70, char_count=0, offset=0
For service index 1, ret = 4, char_count = 0
Debug github issue: gattc_if=1, conn_id=0, start_handle=16, end_handle=29, char_elem=0x3fff5b70, char_count=0, offset=0
For service index 2, ret = 4, char_count = 0
Debug github issue: gattc_if=1, conn_id=0, start_handle=30, end_handle=40, char_elem=0x3fff5b70, char_count=0, offset=0
For service index 3, ret = 4, char_count = 0
Debug github issue: gattc_if=1, conn_id=0, start_handle=41, end_handle=46, char_elem=0x3fff5b70, char_count=0, offset=0
For service index 4, ret = 4, char_count = 0
Debug github issue: gattc_if=1, conn_id=0, start_handle=47, end_handle=50, char_elem=0x3fff5b70, char_count=0, offset=0
For service index 5, ret = 4, char_count = 0
event: 18
[0;32mI (20805) GATTC_ENEGIC: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 498, conn_id 0[0m
```

Svahnen-Perific, 2023-09-22:
Regarding "display database in server site using esp_err_t esp_ble_gatts_show_local_database(void)", could you explain how this function is supposed to be used? And does it work in our client even tough its a gatts command? We don't own the code on the server side.

zhp0406, 2023-09-22:
[image]

zhp0406, 2023-09-22:
@Svahnen-Perific 
Based on your prints and your previous code, I have identified an error in your implementation. The count parameter you provided for esp_ble_gattc_get_all_char seems to be unreasonable. I suggest you either set a non-zero value or use my method to determine the value of count before calling esp_ble_gattc_get_all_char.

please test it again...

Svahnen-Perific, 2023-09-22:
Thank you! That seem too have been the cause.

zhp0406, 2023-09-22:
@Svahnen-Perific 
Sure, if the issue has been resolved, please close this issue.

closed 2023-09-22 by Svahnen-Perific