espressif/esp-idf #12286: IRAM0 segment data does not fit

Abhay-2412, 2023-09-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.5

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Wrover Module

### Power Supply used.

External 3.3V

### What is the expected behavior?

In my project, I successfully utilized several ICs that communicate via I2C, as well as a camera. Additionally, the project incorporated both Wi-Fi and Bluetooth functionality, and it worked smoothly in the Arduino IDE. However, when attempting to import the project into the ESP-IDF (VScode), an IRAM error is encountered.

### What is the actual behavior?

IRAM0 segment data does not fit

### Steps to reproduce.

i just turn on BLE functionality and i wrote some functions and tried to test ended up with this error

### Debug Logs.

```plain
[1/9] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4.5\workspace\GAZZELE_MINI\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.5/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/build/bootloader/bootloader.bin"
Bootloader binary size 0x6ae0 bytes. 0x520 bytes (5%) free.
[2/7] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/main.cpp:73:14: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
 char* host = "192.168.149.237";
              ^~~~~~~~~~~~~~~~~
[5/7] Linking CXX executable GAZZELE_MINI.elf
FAILED: GAZZELE_MINI.elf 
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\GAZZELE_MINI.elf.rsp -o GAZZELE_MINI.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: GAZZELE_MINI.elf section `.iram0.text' will not fit in region `iram0_0_seg'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 18216 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

### More Information.

when ever i commented bluetooth part and tried to build its getting build

Abhay-2412, 2023-09-21:
```
Total sizes:
Used static DRAM:   38048 bytes (  86532 remain, 30.5% used)
      .data size:   15192 bytes
      .bss  size:   22856 bytes
Used static IRAM:  119250 bytes (  11822 remain, 91.0% used)
      .text size:  118223 bytes
   .vectors size:    1027 bytes
Used Flash size :  864827 bytes
      .text     :  680243 bytes
      .rodata   :  184328 bytes
Total image size:  999269 bytes (.bin may be padded larger)
```
above is what i get if i comment bluetooth functionality and check

Abhay-2412, 2023-09-21:
i hope below info helps
```
Executing "ninja size-files"...
[1/1] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4.5\wo...rks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/build/GAZZELE_MINI.map"Total sizes:
Used static DRAM:   55456 bytes (  69124 remain, 44.5% used)
      .data size:   21656 bytes
      .bss  size:   33800 bytes
Used static IRAM:  149286 bytes ( -18214 remain, 113.9% used) Overflow detected!
      .text size:  148259 bytes
   .vectors size:    1027 bytes
Used Flash size : 1272892 bytes
      .text     : 1037215 bytes
      .rodata   :  235421 bytes
Total image size: 1443834 bytes (.bin may be padded larger)
Per-file contributions to ELF file:
             Object File DRAM .data & 0.bss .rtc_noinit IRAM0 .text & 0.vectors ram_st_total Flash .text & .rodata & .rodata_noload & .appdesc flash_total
      spiram_psram.c.obj          9      48           0        8747           0         8804       12808     16159                0          0       37723
           btm_sec.c.obj          0       0           0           0           0            0       14704      2884                0          0       17588
                wl_cnx.o          2    3855           0         261           0         4118       13403      1373             2540          0       17579
       btm_ble_gap.c.obj          0      50           0           0           0           50       14845      2326                0          0       17171
         ieee80211_sta.o          0      34           0        1440           0         1474       11265      1233             1810          0       15748
        lib_a-vfprintf.o          0       0           0           0           0            0       14746       700                0          0       15446
           phy_chip_v7.o        721     614           0        1634           0         2969       13009         0                0          0       15364
       lib_a-svfprintf.o          0       0           0           0           0            0       14319       750                0          0       15069
        bta_dm_act.c.obj          0       0           0           0           0            0       12005      2196                0          0       14201
      ieee80211_output.o          2      20           0        1869           0         1891       11391       476                0          0       13738
                    pp.o       1142      53           0        6764           0         7959        5193       637                0          0       13736
           sockets.c.obj          0     648           0           0           0          648       12254       886                0          0       13140
              uart.c.obj         56      13           0           0           0           69       10776      2288                0          0       13120
               lc_task.o         32       0           0           0           0           32       11795       736                0          0       12563
       ieee80211_ioctl.o          4     104           0         271           0          379       10593      1210              440          0       12518
               nd6.c.obj          8     932           0           0           0          940       12028       251                0          0       12287
       phy_chip_v7_cal.o        474      53           0        3499           0         4026        8136         0                0          0       12109
               sae.c.obj          0       0           0           0           0            0       11588       475                0          0       12063
      lib_a-svfiprintf.o          0       0           0           0           0            0       10414      1172                0          0       11586
               lld_evt.o         53      96           0        6999           0         7148        4171       284                0          0       11507
       lib_a-vfiprintf.o          0       0           0           0           0            0       10610       700                0          0       11310
        ieee80211_scan.o         18     288           0           0           0          306        9027       336             1836          0       11217
                    pm.o         36     404           0        2868           0         3308        7070       941                0          0       10915
            tcp_in.c.obj          0      52           0           0           0           52        9631      1242                0          0       10873
          gatt_api.c.obj          0       0           0           0           0            0        7022      3772                0          0       10794
     bta_gattc_act.c.obj          0       0           0           0           0            0        8619      2090                0          0       10709
                   lld.o          2       0           0          59           0           61       10317       318                0          0       10696
        lib_a-svfscanf.o          0       0           0           0           0            0       10228       387                0          0       10615
          ieee80211_ht.o          0       4           0        1147           0         1151        8216      1117               47          0       10527
     WiFiGeneric.cpp.obj          6      73           0           0           0           79        7265      2893                0          0       10164
[... cut ...]
          mschapv2.c.obj          0       0           0           0           0            0           0         0                0          0           0
       sha1-tlsprf.c.obj          0       0           0           0           0            0           0         0                0          0           0
     sha256-tlsprf.c.obj          0       0           0           0           0            0           0         0                0          0           0
     sha384-tlsprf.c.obj          0       0           0           0           0            0           0         0                0          0           0
```

listout, 2023-09-21:
What partition table and RAM size were you using in Arduino IDE? Try to use that/choose that from the menuconfig.

Abhay-2412, 2023-09-22:
https://gist.github.com/Abhay-2412/42345dc9e5226baf3f363f9624e8605e

Please refer above link for my sdkconfig file

listout, 2023-09-22:
I tried building a hello world project with your sdkconfig and it built fine, http://ix.io/4H31 and the size is http://ix.io/4H33.

> char* host = "192.168.149.237";

Are you using a lot of heap? I suspect enabling wifi and bluetooth has very little to do with your error.

listout, 2023-09-22:
Few things you can try:
- Try build with optimization for size (In menuconfig Compiler options -> Optimization Level -> Optimize for size)
- Have a look at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html
- Have a look at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#optimize-iram-usage

Abhay-2412, 2023-09-22:
Basically my esp32 is using camera and some ICs like GPIO expander ,Battery charger IC i donno which functionality using heap

Abhay-2412, 2023-09-22:
the reason why i assumed its because of wifi or bluetooth because i commented bluetooth related function then it got build
```
Total sizes:
Used static DRAM:   38048 bytes (  86532 remain, 30.5% used)
      .data size:   15192 bytes
      .bss  size:   22856 bytes
Used static IRAM:  119250 bytes (  11822 remain, 91.0% used)
      .text size:  118223 bytes
   .vectors size:    1027 bytes
Used Flash size :  864827 bytes
      .text     :  680243 bytes
      .rodata   :  184328 bytes
Total image size:  999269 bytes (.bin may be padded larger)
```

Abhay-2412, 2023-09-22:
> Few things you can try:
> 
> * Try building with optimization for size (In menuconfig Compiler options -> Optimization Level -> Optimize for size)
> * Have a look at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html
> * Have a look at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#optimize-iram-usage

i set optimization level to size still same error

Dazza0, 2023-09-22:
@Abhay-2412

```
Used static IRAM:  138090 bytes (  -7018 remain, 105.4% used) Overflow detected! You can run idf.py size-files for more information.
```
This doesn't mean your overall binary size is too large. It just means that your binary uses too much IRAM. By default, most of your binary (i.e., functions) will be stored in flash memory. However, certain functions are placed in IRAM for performance or behavioral reasons.

You can use `idf.py size-components` and `idf.py size-files` to get a summary of IRAM usage on a per-component and per-file level respectively. This should help you track down which feature is using the most IRAM.

To reduce IRAM usage, please see the [optimizing IRAM usage doc](https://docs.espressif.com/projects/esp-idf/en/v4.4.5/esp32/api-guides/performance/ram-usage.html#optimizing-iram-usage). This should list the various menuconfig options to move particular component functions from IRAM back into flash memory, thus saving some IRAM.

Abhay-2412, 2023-09-25:
Hello, I successfully reduced the storage utilization of IRAM through optimization techniques. However, even after these optimizations, the IRAM is still occupying approximately 99.8 percent of its capacity. I would greatly appreciate further guidance or suggestions on how to address this situation. Thank you.

```
Total sizes:
Used static DRAM:   42936 bytes (  81644 remain, 34.5% used)
      .data size:   21272 bytes
      .bss  size:   21664 bytes
Used static IRAM:  130862 bytes (    210 remain, 99.8% used)
      .text size:  129835 bytes
   .vectors size:    1027 bytes
Used Flash size : 1158408 bytes
      .text     :  943531 bytes
      .rodata   :  214621 bytes
Total image size: 1310542 bytes (.bin may be padded larger)
```

listout, 2023-09-25:
> You can use idf.py size-components and idf.py size-files to get a summary of IRAM usage on a per-component and per-file level respectively. This should help you track down which feature is using the most IRAM.

listout, 2023-09-25:
> libarduino.a

Are you using arduino as a component? If it's because of some libraries try finding alternatives for native native ESP-IDF?

Abhay-2412, 2023-09-25:
Yes i'm using Arduino as component ,I have successfully compiled and tested the Arduino code, and now I want to import it into ESP-IDF without spending additional time rewriting the code in ESP-IDF.

Dazza0, 2023-11-07:
@Abhay-2412 I'm closing this issue as there aren't any clear action items. Please try some of the suggestions posted above in this thread.

Use `idf.py size-components` or `idf.py size-files` to find out what components/files are consuming the most IRAM.

- If it is an IDF component that needs IRAM reduction, try some of the steps in the [optimizing IRAM usage doc](https://docs.espressif.com/projects/esp-idf/en/v4.4.5/esp32/api-guides/performance/ram-usage.html#optimizing-iram-usage)
- If it is non-IDF provided components consuming the most IRAM, please check what attributes or [linker placements](https://docs.espressif.com/projects/esp-idf/en/v4.4.5/esp32/api-guides/linker-script-generation.html) those components are using, and see if their functions can be moved into flash.

closed 2023-11-07 by Dazza0

Abhay-2412, 2023-12-05:
Hello just a small doubt Why no IRAM overflow issues in Arduino IDE build, but overflow occurs using Arduino as a component?

igrr, 2023-12-05:
You probably don't have exactly the same configuration (sdkconfig) in these two cases. You can update your sdkconfig used in arduino-as-component build to match the sdkconfig used in Arduino.