espressif/esp-idf #12287: ESP32-S3 build in windows and docker does not recognise ESP_EXT1_WAKEUP_ANY_LOW

ttkkhhaann, 2023-09-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Expected behaviour is to build with ESP_EXT1_WAKEUP_ANY_LOW

### What is the actual behavior?

Build with ESP_EXT1_WAKEUP_ALL_LOW but not with ESP_EXT1_WAKEUP_ANY_LOW

### Steps to reproduce.

We have two machines and one cloud docker. 
ON windows machine we have to replace ESP_EXT1_WAKEUP_ANY_LOW with ESP_EXT1_WAKEUP_ALL_LOW to build
ON docker we also have to replace ESP_EXT1_WAKEUP_ANY_LOW with ESP_EXT1_WAKEUP_ALL_LOW to build
ON MacOS we build with ESP_EXT1_WAKEUP_ANY_LOW and it build as expected

Windows is ESP-IDF v5.1. 

Docker is https://hub.docker.com/layers/espressif/idf/release-v5.1/images/sha256-8c4fbd6821489d37e99a25f2fcac728f40e8165944137812f5b4110f6aa23fd9?context=explore

### Build or installation Logs.

```plain
In file included from C:/Espressif/frameworks/esp-idf-v5.1-2/components/esp_hw_support/include/esp_sleep.h:10,
                 from C:/x/x.espFirmware5/x/main/include/app_main.h:6,
                 from C:/x/x.espFirmware5/x/main/app_main.c:1:
C:/x/x.espFirmware5/x/main/app_main.c:625:73: error: 'ESP_EXT1_WAKEUP_ANY_LOW' undeclared (first use in this function); did you mean 'ESP_EXT1_WAKEUP_ALL_LOW'?
  625 |     ESP_ERROR_CHECK(esp_sleep_enable_ext1_wakeup(ext_wakeup_pin_1_mask, ESP_EXT1_WAKEUP_ANY_LOW));
      |                                                                         ^~~~~~~~~~~~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.1-2/components/esp_common/include/esp_err.h:117:30: note: in definition of macro 'ESP_ERROR_CHECK'
  117 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
C:/flare/Flare.espFirmware5/flareApp/main/app_main.c:625:73: note: each undeclared identifier is reported only once for each function it appears in
  625 |     ESP_ERROR_CHECK(esp_sleep_enable_ext1_wakeup(ext_wakeup_pin_1_mask, ESP_EXT1_WAKEUP_ANY_LOW));
      |                                                                         ^~~~~~~~~~~~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.1-2/components/esp_common/include/esp_err.h:117:30: note: in definition of macro 'ESP_ERROR_CHECK'
  117 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
[107/108] Generating binary image from built executableesptool.py v4.7.dev1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Generated C:/x/x.espFirmware5/x/build/bootloader/bootloader.bin
[108/108] cmd.exe /C "cd /D C:\x\x.espFirmware5\fl...are.espFirmware5/x/build/bootloader/bootloader.bin"Bootloader binary size 0x51f0 bytes. 0x2e10 bytes (36%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\x\x.espFirmware5\x\build\log\idf_py_stderr_output_50708 and C:\x\x.espFirmware5\x\build\log\idf_py_stdout_output_50708
```

igrr, 2023-09-22:
@ttkkhhaann In release/v5.1 branch, `ESP_EXT1_WAKEUP_ANY_LOW` was added in commit b27e57db7bf6dc810f3603e19045c93afdc428e4, with follow-up changes in badef66538ec6deb494687c9a14eeab883eadb83 and 5859b8323285fefed8a0ebdb479ca46aa864a8f4. These commits are not part of v5.1 release tag. They are included in v5.1.1 bugfix release.

I think the issue you are seeing is due to the fact that you aren't running IDF v5.1.1 (or a more recent version or release/v5.1 branch) on all of the platforms. To troubleshoot this, please run `git describe` command in ESP-IDF directory on all 3 platforms, and post the results here.

ttkkhhaann, 2023-09-28:
Hi @igrr 

git describe on Windows machine  where id doest build with ESP_EXT1_WAKEUP_ANY_LOW

 git describe
v5.1

git describe at MAC where  it does build with ESP_EXT1_WAKEUP_ANY_LOW
v5.1.1-305-g697e3e8eac

cannot figure out how to run git describe on docker image
but the build results in pipeline show

[2/2] idf (5.1.1) and it doesnt build due to same error

ttkkhhaann, 2023-09-29:
@igrr 
I have changed the docker to the latest v5.1.1 and it is working now. The only issue remaining is how to fix it on windows,
I have reinstaleed 5..1.1 on windows but it doesn't build. I cannot find any instructions on how to update windows to latest branch . git status inside windows says its not on any branch. I used windows installer for this
thanks

Dazza0, 2023-11-23:
@ttkkhhaann You need to make sure that you are downloading the correct docker image for your selected version of ESP-IDF. 

The docker image you've [linked above](https://hub.docker.com/layers/espressif/idf/release-v5.1/images/sha256-92accd9c68964d18035a5a582cf0e80885eaa405837cb3b84acaa5591c435a50?context=explore) is for the ESP-IDF `release/v5.1` branch which is progressively updated over multiple bugfix releases.

If you are using ESP-IDF v5.1.1 (i.e., `git checkout tags/v5.1.1` on your MAC), then you should download the [v5.1.1 docker image](https://hub.docker.com/layers/espressif/idf/v5.1.1/images/sha256-5e8e9e02a7f41ca549a03e0df4ab48d1b9f7aaabf040797a5963e8dc548a8e52?context=explore) on your corresponding windows PC.

Hope the pointers above resolves your issue.

closed 2023-11-23 by Dazza0