espressif/esp-idf #12290: Read multiple registers in one CID

PatrickV82, 2023-09-22:
When reading multiple registers in one CID only the first data is copied !! 
this is different then when using RTU master modbus.

example CID 
const mb_parameter_descriptor_t device_parameters[] = {
	// { CID		   , Param Name		  , Units	  , Modbus Slave Addr	, Modbus Reg Type, Reg Start, Reg Size, Instance Offset, Data Type		 , Data Size, Parameter Options	, Access Mode}
		// { CID		   												, Param Name		  									, 	Units	  , 	Modbus Slave Addr, 		Modbus Reg Type		, Reg Start					, Reg Size	, Instance Offset, Data Type		 , Data Size	 , Parameter Options	, Access Mode}
{	CID_SLAVE_INIT_RW_ADDR										, STR("Slave ID")										, 	STR("Data"), 	MB_DEVICE_SLAVEID,		MB_PARAM_HOLDING	,MODBUS_HREG_SLAVEID				,5			,0				 , PARAM_TYPE_U16	 , PARAM_SIZE_U16, OPTS(0, 0, 0)		,	PAR_PERMS_READ_WRITE_TRIGGER},

}

https://github.com/espressif/esp-idf/blob/f21cbb625f1c9b05f94aa774c885b1389a8c4f8b/components/freemodbus/tcp_master/modbus_controller/mbc_tcp_master.c#L424

alisitsyn, 2023-09-25:
The documentation describes the mode [here](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/master_api_overview.html#configuring-master-data-access). 
```
Specifies type of the characteristic. PARAM_TYPE_U8, PARAM_TYPE_U16, PARAM_TYPE_U32 - Unsigned integer 8/16/32 bit type; PARAM_TYPE_FLOAT - IEEE754 floating point format; PARAM_TYPE_ASCII - ASCII string or binary data;
```
However, the example in the documentation does not address how to use this type of parameter in data dictionary.
I think it worth to add its description to the documentation. I will try to do it in next release or a bit later. 
Thank you for issue and comments.

alisitsyn, 2023-10-10:
The documentation is updated in esp-modbus v1.0.12 to address the [PARAM_TYPE_ASCII](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/master_api_overview.html#configuring-master-data-access) type. The issue will be closed.

closed 2023-11-01 by tomassebestik