espressif/esp-idf #12294: int_wdt breaks dumping core

KaeLL, 2023-09-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Even on v5.2-dev-2934-g3b748a6cb7, the code
```c
#include "driver/gptimer.h"
#include "esp_attr.h"

IRAM_ATTR bool gpt_cb( gptimer_handle_t timer, const gptimer_alarm_event_data_t *edata, void *user_ctx )
{
	while ( 1 )
		;
}

void app_main( void )
{
	gptimer_handle_t gpt = NULL;

	ESP_ERROR_CHECK( gptimer_new_timer(
		&( const gptimer_config_t ) {
			.clk_src = GPTIMER_CLK_SRC_DEFAULT,
			.intr_priority = 3,
			.resolution_hz = 40000000 },
		&gpt ) );
	ESP_ERROR_CHECK( gptimer_register_event_callbacks( gpt, &( const gptimer_event_callbacks_t ) { gpt_cb }, NULL ) );
	ESP_ERROR_CHECK( gptimer_enable( gpt ) );
	ESP_ERROR_CHECK( gptimer_set_alarm_action( gpt, &( const gptimer_alarm_config_t ) { .alarm_count = 80000000 } ) );
	ESP_ERROR_CHECK( gptimer_start( gpt ) );
}

```
with this defconfig
```
# This file was generated using idf.py save-defconfig. It can be edited manually.
# Espressif IoT Development Framework (ESP-IDF) 5.2.0 Project Minimal Configuration
#
CONFIG_APP_NO_BLOBS=y
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF=y
CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG=y
CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS=y
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_PARTITION_TABLE_OFFSET=0xC000
CONFIG_COMPILER_OPTIMIZATION_NONE=y
CONFIG_COMPILER_WARN_WRITE_STRINGS=y
CONFIG_SPI_MASTER_IN_IRAM=y
CONFIG_SPI_SLAVE_ISR_IN_IRAM=n
CONFIG_GPTIMER_CTRL_FUNC_IN_IRAM=y
CONFIG_GPTIMER_ISR_IRAM_SAFE=y
CONFIG_GPTIMER_ENABLE_DEBUG_LOG=y
CONFIG_ESP32_REV_MIN_3=y
CONFIG_RTC_CLK_SRC_INT_8MD256=y
CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE=n
CONFIG_ESP_PHY_RF_CAL_FULL=y
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y
CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT=y
CONFIG_ESP_CONSOLE_UART_CUSTOM=y
CONFIG_ESP_CONSOLE_UART_BAUDRATE=2000000
CONFIG_ESP_INT_WDT_TIMEOUT_MS=30
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_DEBUG_OCDAWARE=n
CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH=y
CONFIG_ESP_COREDUMP_CHECKSUM_SHA256=y
[... 119 lines cut ...]
Core  1 register dump:
PC      : 0x40087664  PS      : 0x00060334  A0      : 0x800d2fb0  A1      : 0x3ffb8b10  
0x40087664: xt_utils_wait_for_intr at /esp-idf/components/xtensa/include/xt_utils.h:82
 (inlined by) esp_cpu_wait_for_intr at /esp-idf/components/esp_hw_support/cpu.c:132

A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00060023  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x3ffb8b10  A8      : 0x800dea38  A9      : 0x3ffb8ac0  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x000000c6  
A14     : 0x00000000  A15     : 0x3ffb8ac0  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40087661:0x3ffb8b10 0x400d2fad:0x3ffb8b30 0x4008b9c4:0x3ffb8b60
0x40087661: xt_utils_wait_for_intr at /esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at /esp-idf/components/esp_hw_support/cpu.c:132

0x400d2fad: esp_vApplicationIdleHook at /esp-idf/components/esp_system/freertos_hooks.c:59

0x4008b9c4: prvIdleTask at /esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4058 (discriminator 1)

ELF file SHA256: f0a061c0123f418e

Re-entered core dump! Exception happened during core dump!
CPU halted.
```
impossibilitating dumping core and consequently making debugging more difficult, which is a big issue.

igrr, 2023-09-22:
> CONFIG_ESP_INT_WDT_TIMEOUT_MS=30

Can you please try increasing this to 300ms?

KaeLL, 2023-09-25:
[Done](https://termbin.com/5uge), and result is the same.

KaeLL, 2023-09-27:
Same thing using the generic_gpio example.
<details>
<summary>defconfig</summary>

```
# This file was generated using idf.py save-defconfig. It can be edited manually.
# Espressif IoT Development Framework (ESP-IDF) 5.2.0 Project Minimal Configuration
#
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_COREDUMP_ENABLE_TO_UART=y
CONFIG_ESP_COREDUMP_CHECKSUM_SHA256=y
CONFIG_ESP_COREDUMP_LOGS=n
CONFIG_ESP_COREDUMP_MAX_TASKS_NUM=10
CONFIG_ESP_COREDUMP_STACK_SIZE=1024
```

</details>
<details>
<summary>diff</summary>

```
*** /esp-idf/examples/peripherals/gpio/generic_gpio/main/gpio_example_main.c	2023-09-15 13:14:34.602192047 -0300
--- main/gpio_example_main.c	2023-09-27 15:31:40.710017870 -0300
*************** static QueueHandle_t gpio_evt_queue = NU
*** 61,66 ****
--- 61,67 ----
  
  static void IRAM_ATTR gpio_isr_handler(void* arg)
  {
+     while (1);
      uint32_t gpio_num = (uint32_t) arg;
      xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);
  }
```

</details>
<details>
<summary>Log</summary>

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7136
load:0x40078000,len:15780
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3884
entry 0x40080650
I (31) boot: ESP-IDF v5.2-dev-2934-g3b748a6cb7 2nd stage bootloader
I (31) boot: compile time Sep 27 2023 15:18:38
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v3.0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0a79ch ( 42908) map
I (121) esp_image: segment 1: paddr=0001a7c4 vaddr=3ffb0000 size=02714h ( 10004) load
[... 143 lines cut ...]
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffaf840  A5      : 0x3ffaf820  
A6      : 0x400845a0  A7      : 0x00000001  A8      : 0x800d9702  A9      : 0x3ffb66c0  
0x400845a0: ipc_task at /esp-idf/components/esp_system/esp_ipc.c:48

A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffaf820  A13     : 0x3ffaf7f0  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400847fb:0x3ffb6700 0x400d225b:0x3ffb6720 0x40086766:0x3ffb6740 0x40088041:0x3ffb6760
0x400847fb: xt_utils_wait_for_intr at /esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at /esp-idf/components/esp_hw_support/cpu.c:132

0x400d225b: esp_vApplicationIdleHook at /esp-idf/components/esp_system/freertos_hooks.c:59

0x40086766: prvIdleTask at /esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4058 (discriminator 1)

0x40088041: vPortTaskWrapper at /esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 3fa76c448

Re-entered core dump! Exception happened during core dump!
Rebooting...
```

</details>

erhankur, 2025-12-28:
@KaeLL This should be fixed with commit 083999290e45b7f5478fd1af042e17269ec7f6e0.

The fix has been backported to all branches except v5.5. That backport is currently in progress.

Feel free to reopen the issue if you see the same problem on the latest branches.

closed 2025-12-28 by erhankur