espressif/esp-idf #12295: blufi panic

Leadrive, 2023-09-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

no panic

### What is the actual behavior?

panic reboot

### Steps to reproduce.

微信小程序，通过blufi连上设备后，断开连接，重新连，发生panic。
应该跟应用无关，应该是blufi本身的一个bug

### Debug Logs.

```plain
`
CORRUPT HEAP: Bad tail at 0x3fcd6ce4. Expected 0xbaad5678 got 0x63414141

assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace: 0x40375cb6:0x3fcc4520 0x4037f991:0x3fcc4540 0x40385f71:0x3fcc4560 0x40385b7c:0x3fcc4680 0x403761b9:0x3fcc46a0 0x40385fa1:0x3fcc46c0 0x42032605:0x3fcc46e0 0x42054312:0x3fcc4710 0x42063bca:0x3fcc49a0 0x4204c2fd:0x3fcc49e0 0x4204adf6:0x3fcc4a00 0x4204b2f5:0x3fcc4cb0 0x4204b682:0x3fcc4d10 0x42049f9e:0x3fcc4d60 0x42049fda:0x3fcc4d80 0x42050fa9:0x3fcc4da0 0x420465e0:0x3fcc4dd0 0x42032f7f:0x3fcc4df0
0x40375cb6: panic_abort at /home/develop/esp/esp-idf/components/esp_system/panic.c:408
                                                                                                                                                                                                                                   
0x4037f991: esp_system_abort at /home/develop/esp/esp-idf/components/esp_system/esp_system.c:137
                                                                                                                                                                                                                                   
0x40385f71: __assert_func at /home/develop/esp/esp-idf/components/newlib/assert.c:85
                                                                                                                                                                                                                                   
0x40385b7c: multi_heap_free at /home/develop/esp/esp-idf/components/heap/multi_heap_poisoning.c:253
 (inlined by) multi_heap_free at /home/develop/esp/esp-idf/components/heap/multi_heap_poisoning.c:245                                                                                                                              
                                                                                                                                                                                                                                   
0x403761b9: heap_caps_free at /home/develop/esp/esp-idf/components/heap/heap_caps.c:361
                                                                                                                                                                                                                                   
0x40385fa1: free at /home/develop/esp/esp-idf/components/newlib/heap.c:39
                                                                                                                                                                                                                                   
0x42032605: btc_blufi_recv_handler at /home/develop/esp/esp-idf/components/bt/common/btc/profile/esp/blufi/blufi_prf.c:170
                                                                                                                                                                                                                                   
0x42054312: blufi_profile_cb at /home/develop/esp/esp-idf/components/bt/common/btc/profile/esp/blufi/bluedroid_host/esp_blufi.c:211
                                                                                                                                                                                                                                   
0x42063bca: bta_gatts_send_request_cback at /home/develop/esp/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gatts_act.c:949
                                                                                                                                                                                                                                   
0x4204c2fd: gatt_sr_send_req_callback at /home/develop/esp/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_utils.c:1532
                                                                                                                                                                                                                                   
0x4204adf6: gatts_process_write_req at /home/develop/esp/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_sr.c:1161
                                                                                                                                                                                                                                   
0x4204b2f5: gatts_process_attribute_req at /home/develop/esp/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_sr.c:1470
                                                                                                                                                                                                                                   
0x4204b682: gatt_server_handle_client_req at /home/develop/esp/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_sr.c:1665
                                                                                                                                                                                                                                   
0x42049f9e: gatt_data_process at /home/develop/esp/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_main.c:1000
                                                                                                                                                                                                                                   
0x42049fda: gatt_le_data_ind at /home/develop/esp/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_main.c:587
                                                                                                                                                                                                                                   
0x42050fa9: l2c_rcv_acl_data at /home/develop/esp/esp-idf/components/bt/host/bluedroid/stack/l2cap/l2c_main.c:275
                                                                                                                                                                                                                                   
0x420465e0: btu_hci_msg_process at /home/develop/esp/esp-idf/components/bt/host/bluedroid/stack/btu/btu_task.c:144
                                                                                                                                                                                                                                   
0x42032f7f: osi_thread_run at /home/develop/esp/esp-idf/components/bt/common/osi/thread.c:165
````
```

### More Information.

ver 4.4.4
blufi_prf.c
````c
        if (blufi_env.offset > 0) {   /* if previous pkt is frag */
            memcpy(blufi_env.aggr_buf + blufi_env.offset, hdr->data, hdr->data_len);

            btc_blufi_protocol_handler(hdr->type, blufi_env.aggr_buf, blufi_env.total_len);
            blufi_env.offset = 0;
            osi_free(blufi_env.aggr_buf);
            blufi_env.aggr_buf = NULL;
        } else {
            btc_blufi_protocol_handler(hdr->type, hdr->data, hdr->data_len);
            blufi_env.offset = 0;
        }
````
第170行就是
osi_free(blufi_env.aggr_buf);

Leadrive, 2023-09-23:
比较难复现，blufi不稳定，经常出现一些莫名的panic，希望厂家能多测试。

zhp0406, 2023-09-25:
@Leadrive 
感谢你报告问题，我会尝试复现。如果期间你有其他发现，麻烦告诉我一下。

zhp0406, 2023-09-25:
@Leadrive 
你可以把完整的 log 给我一下吗？另外，你有没有使用 menuconfig 做其他的配置。

Leadrive, 2023-09-25:
@zhp0406 我昨天更新到最新的4.4.4
代码已修改成
```c
        if (blufi_env.offset > 0) {   /* if previous pkt is frag */
            /* blufi_env.aggr_buf should not be NULL */
            if (blufi_env.aggr_buf == NULL) {
                BTC_TRACE_ERROR("%s buffer is NULL\n", __func__);
                btc_blufi_report_error(ESP_BLUFI_DH_MALLOC_ERROR);
                return;
            }
            /* payload length should be equal to total_len */
            if ((blufi_env.offset + hdr->data_len) != blufi_env.total_len) {
                BTC_TRACE_ERROR("%s payload is longer than packet length, len %d \n", __func__, blufi_env.total_len);
                btc_blufi_report_error(ESP_BLUFI_DATA_FORMAT_ERROR);
                return;
            }
            memcpy(blufi_env.aggr_buf + blufi_env.offset, hdr->data, hdr->data_len);

            btc_blufi_protocol_handler(hdr->type, blufi_env.aggr_buf, blufi_env.total_len);
            blufi_env.offset = 0;
            osi_free(blufi_env.aggr_buf);
            blufi_env.aggr_buf = NULL;
        } else {
            btc_blufi_protocol_handler(hdr->type, hdr->data, hdr->data_len);
            blufi_env.offset = 0;
        }
```
不知道是不是修复了此问题

zhp0406, 2023-09-25:
@Leadrive 
我不太清楚最新的 v4.4.4 有没有修复该问题，你之前使用的是哪个 commit？你使用最新的 v4.4.4 多测试几次试试，我这面也会进行测试。

Leadrive, 2023-09-25:
@zhp0406
```
git log
commit e8bdaf91986a41678adc6be13888fc037b1acb68 (HEAD, tag: v4.4.4)
Author: Xiao Xufeng <xiaoxufeng@espressif.com>
Date:   Fri Jan 20 02:12:49 2023 +0800

    versions: Update version to 4.4.4

commit 800dec96e9432e96f436918d4298aba3fd60d8a6
Merge: d4080633f9 850ac23d14
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Sat Jan 14 01:01:20 2023 +0800

    Merge branch 'bugfix/clear_pmk_cache_for_missing_reason_codes' into 'release/v4.4'
    
    Fix missing disconnected callback (Backport v4.4)
    
    See merge request espressif/esp-idf!22036

commit 850ac23d1468e7a65d7b6ad02f23234aad0c588a
Author: Shyamal Khachane <shyamal.khachane@espressif.com>
Date:   Fri Jan 13 11:38:27 2023 +0530

    esp_wifi: Fix missing disconnected callback

commit d4080633f9a3498c8245118844b9126550c66162
Merge: 552beab8a9 e59269efa0
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Thu Jan 12 16:24:59 2023 +0800

    Merge branch 'feature/efuse_rev_major_minor_v4.4' into 'release/v4.4'
    
    efuse: Adds major and minor versions (v4.4)
    
    See merge request espressif/esp-idf!19541

commit 552beab8a94a7787c5a210e5dc6bf04354741707
Merge: 9ee3c8337d 79807ec6ab
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Tue Jan 10 17:19:01 2023 +0800

    Merge branch 'bugfix/sae_auth_and_pmk_issues_v4.4' into 'release/v4.4'
    
    Fix SAE open auth and pmk issues (Backport v4.4)
    
    See merge request espressif/esp-idf!21911

commit 79807ec6ab784fbc9ee5a7b9f4d9f52f77db24fb
Author: Shyamal Khachane <shyamal.khachane@espressif.com>
```

Alvin1Zhang, 2024-03-10:
感谢反馈，后续若该问题还出现，将会 Reopen 该问题，谢谢。

closed 2024-03-10 by Alvin1Zhang