espressif/esp-idf #12300: using touch_pad_read_raw_data() with touch sensors on esp32s3 reads high, unchanging values

saillingaway, 2023-09-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-2756-g2bc1f2f574

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 M5Stack

### Power Supply used.

USB

### What is the expected behavior?

The touch_sensor_v2 touch_pad_read example should output values that fluctuate similarly to the example output, and values should go up when respective sensors are touched.

### What is the actual behavior?

Running the example exactly as it is results in the following output:
```
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]

```
The values are all exceptionally higher than the ones from the example output, and none of the values ever change no matter which sensors are touched.

### Steps to reproduce.

1. Install and set up esp-idf for MacOS following [[espressif doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/linux-macos-setup.html)](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/linux-macos-setup.html)
2. Open the example project `esp-idf/examples/peripherals/touch_sensor/touch_sensor_v2/touch_pad_read`
3. Open terminal with working directory of the example project
4. Build the project with `idf.py build`
5. Attach esp32s3 via USB and flash with `idf.py flash -p PORT`
6. Use `idf.py monitor` to see the output

### More Information.

During my troubleshooting I found that:
1. adding `touch_pad_fsm_start();` right before `touch_pad_read_raw_data();` makes it so some of the raw values will fluctuate a little and respond to touch, but some remain unchanging.

```for (int i = 0; i < TOUCH_BUTTON_NUM; i++) {
            touch_pad_fsm_start();
            touch_pad_read_raw_data(button[i], &touch_value);    // read raw data.
            printf("T%d: [%4"PRIu32"] ", button[i], touch_value);
        }
```
[... 185 lines cut ...]
    touch_pad_init();
    // instead of for loop:
    touch_pad_config(button[0]);
    touch_pad_config(button[1]);
    touch_pad_config(button[2]);
    touch_pad_config(button[3]);
    touch_pad_config(button[4]);
    touch_pad_config(button[5]);
    touch_pad_config(button[6]);
    touch_pad_config(button[7]);
    touch_pad_config(button[8]);
    touch_pad_config(button[9]);
    touch_pad_config(button[10]);
    touch_pad_config(button[11]);
...
```
Which gives me the below output most similar to the example output, all the sensors fluctuate, react to touch and the value for T4 is also lower when I don't include `TOUCH_PAD_NUM3` in ` touch_pad_t button`).
```
T2: [14475] T4: [13773] T5: [13353] T6: [13335] T7: [13413] T8: [12834] T9: [13283] T10: [13745] T11: [14867] T12: [15214] T13: [14918] T14: [15670] 
T2: [14468] T4: [13770] T5: [13330] T6: [13327] T7: [13407] T8: [12827] T9: [13283] T10: [13756] T11: [14868] T12: [15209] T13: [14916] T14: [15692] 
T2: [14471] T4: [13774] T5: [13351] T6: [13330] T7: [13409] T8: [12825] T9: [13278] T10: [13762] T11: [14878] T12: [15216] T13: [14917] T14: [15689] 
T2: [14472] T4: [13767] T5: [13347] T6: [13329] T7: [13404] T8: [12824] T9: [13280] T10: [13774] T11: [14874] T12: [15216] T13: [14912] T14: [15691] 
```

I'm new to embedded development so I may have missed some additional necessary configuration, but I've looked at the docs and application note and didn't see anything so posting an issue here in hopes to get some insight onto what's going on. A little confused as to why I had to make these adjustments to get raw data like in the example--especially since the way I'm using `touch_pad_fsm_start()` now to get those values seems wrong/different from it's intended use. Is there possibly a bug here? Either with the higher values or why the raw values don't update unless I  restart the finite state machine each time I want to read values. Thanks in advance.

leeebo, 2023-09-26:
@saillingaway Hi, please refer #12198

saillingaway, 2023-09-27:
@leeebo Thanks for the reply. I saw this issue already while I was searching for a solution and I don't think this is the same issue I'm having now. The way I understand it, if _one_ touch sensor gets too high, **all of the other touch sensors will hang as well**. Is this right? I do think that was happening at first--one of the first high readings from `touch_pad_read_raw_data()` would hang the rest of the sensors and so none of the values would ever change. It makes sense that calling `touch_pad_fsm_start()` right before `touch_pad_read_raw_data()` would 'fix' things so most of the values update each time. However, I don't understand why `T3`, `T4` and `T5` would still be stuck. After the weird fix of getting rid of the for loops, `TOUCH_PAD_NUM3/T3` is still showing a high unchanging value, even though it's only configured as a touch sensor, so I can't use it. Are you saying that `T3` is hanging every single time, and reads the same value despite resetting the fsm?

In #12198 you said if one of the pins was accidentally connected to another device the touch sensor would be trapped in that pin, causing abnormal readings in that channel and maybe in other channels too, but I've only plugged the chip into my machine, in order to flash and monitor. Still, if that is what's happening here, **is there more configuration for `T3` that's necessary to use it as a touch sensor again?** I already have a smaller sample size (100) like you mentioned in the other thread and what I had above was the result of that. I've also tried resetting the pin using [gpio_reset_pin(GPIO_NUM_3)](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/gpio.html#_CPPv414gpio_reset_pin10gpio_num_t) and then configuring them as a touch_sensor just like in the example.

I understand any GPIO pins enabled and configured as touch sensors can't be connected to any other function or hardware, which I don't believe I'm doing. Also, the other functionality T1 will be used for that I mentioned before **isn't configured**, I only removed `TOUCH_PAD_NUM1` from `button[]` preemptively as a reminder to myself. I also experimented with `touch_pad_set_voltage()` to try to see if I could get the value of `T3` to come down but didn't have any luck.

I was reading through the esp32-s3 [data sheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf) again and saw on page 16 that `GPIO3` is a strapping pin. It says after start up the strapping pins are freed up to be used as I/O pins after reset, but I'm curious if there's some artifact that's interfering with the reading?

leeebo, 2023-09-27:
Hi @saillingaway , after checking the [StampS3 schematic](https://docs.m5stack.com/en/core/StampS3), I found the root cause: the GPIO3 (T3) of StampS3 is pulled up outside, this is forbidden for the touch sensor because the pulled-up pin is not able to charge and discharge as normal. 

1. Removing T3 from your button list will work around. (and `touch_pad_fsm_start();` only need be called once)

leeebo, 2023-09-27:
The [StampS3](https://docs.m5stack.com/en/core/StampS3) pullup GPIO3 by default, could be an attempt to switch `USB-JTAG` and `Standard JTAG` interface through the strapping pin. which causes touch sensor T3 to become unusable.

[image]

saillingaway, 2023-09-27:
Thank you! I looked at the schematic and I see that now. Thankfully the word around will work. like I said, if I don't call `touch_pad_fsm_start();` every single time, no values will update. If I only call it once, the output of raw data will like this:

```
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
T1: [14721] T2: [14516] T3: [2438734] T4: [4116735] T5: [4158749] T6: [1539359] T7: [1538649] T8: [1585147] T9: [4035765] T10: [244164] T11: [948200] T12: [1998005] T13: [3530295] T14: [4177858]
```

So the only way I can get any new readings is to call it at the beginning of the while loop before all the `touch_pad_read_raw_data();` calls. Is this maybe because T3 was causing the sensors to hang? Also, is there any reason you can think why moving the config and init outside of for loops would fix `T4` and `T5`?

leeebo, 2023-09-27:
Yes, If you enable and try to read from T3, the sensor will hang, causing other channels not work also.

After remove T3, the example [touch_sensor_v2/touch_pad_read](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/touch_sensor/touch_sensor_v2) should work as normal

saillingaway, 2023-09-27:
> 2. The reason call `touch_pad_fsm_start();`  before each read makes some difference, is because the FSM will restart. In normal case please only call it once

@leeebo yeah I understand that the values are different because the FSM restarts, normally I would follow the example and only call it once but I need to use it every loop to get different readings because of the issue with `T3` I think. (Even if `T3` is removed, calling `touch_pad_fsm_start();` at the beginning once outputs data that never changes)

leeebo, 2023-09-27:
@saillingaway After removing T3 (not config and not read), the touch should work as normal, `touch_pad_fsm_start();` only needs to be [called once after touch init](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/touch_sensor/touch_sensor_v2/touch_pad_read/main/tp_read_main.c#L86). Does the examples work for you then?

saillingaway, 2023-09-27:
@leeebo I tested it again and it works 🎉  Never took out the config but I think before I must not have actually removed `TOUCH_PAD_NUM3` from `button[]`, I may have only removed the `touch_pad_read_raw_data()` for `button[2]` which wouldn't fix anything lol. Thanks for your help on this!

closed 2025-01-20 by Alvin1Zhang