espressif/esp-idf #12302: 无法安装ESP_IDF V4.4.5

us666666, 2023-09-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.5

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

正常安装 

### What is the actual behavior?

相反

### Steps to reproduce.

1.Vscode 中安装 ESP-IDF Extension for Visual Studio Code
2.选择安装源为Github
3.点击install

### Build or installation Logs.

```plain
pip 22.3.1 from C:\Users\kingo\.espressif\tools\idf-python\3.11.2\Lib\site-packages\pip (python 3.11)

Collecting virtualenv
  Using cached virtualenv-20.24.5-py3-none-any.whl (3.7 MB)
Collecting distlib<1,>=0.3.7
  Using cached distlib-0.3.7-py2.py3-none-any.whl (468 kB)
Collecting filelock<4,>=3.12.2
  Using cached filelock-3.12.4-py3-none-any.whl (11 kB)
Collecting platformdirs<4,>=3.9.1
  Using cached platformdirs-3.10.0-py3-none-any.whl (17 kB)
Installing collected packages: distlib, platformdirs, filelock, virtualenv
Successfully installed distlib-0.3.7 filelock-3.12.4 platformdirs-3.10.0 virtualenv-20.24.5
Installing virtualenv
Creating a new Python environment in C:\Users\kingo\.espressif\python_env\idf4.4_py3.11_env
  WARNING: The script virtualenv.exe is installed in 'C:\Users\kingo\AppData\Roaming\Python\Python311\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.

[notice] A new release of pip available: 22.3.1 -> 23.2.1
[notice] To update, run: C:\Users\kingo\.espressif\tools\idf-python\3.11.2\python.exe -m pip install --upgrade pip
C:\Users\kingo\.espressif\tools\idf-python\3.11.2\python.exe: No module named virtualenv
Traceback (most recent call last):
  File "C:\Users\kingo\esp\esp-idf\tools\idf_tools.py", line 1976, in <module>
    main(sys.argv[1:])
  File "C:\Users\kingo\esp\esp-idf\tools\idf_tools.py", line 1972, in main
    action_func(args)
  File "C:\Users\kingo\esp\esp-idf\tools\idf_tools.py", line 1598, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv',
  File "subprocess.py", line 413, in check_call
subprocess.CalledProcessError: Command '['C:\\Users\\kingo\\.espressif\\tools\\idf-python\\3.11.2\\python.exe', '-m', 'virtualenv', '--python', 'C:\\Users\\kingo\\.espressif\\tools\\idf-python\\3.11.2\\python.exe', '--seeder', 'pip', 'C:\\Users\\kingo\\.espressif\\python_env\\idf4.4_py3.11_env']' returned non-zero exit status 1.
Command failed: "C:\Users\kingo\.espressif\tools\idf-python\3.11.2\python.exe" "C:\Users\kingo\esp\esp-idf\tools\idf_tools.py" install-python-env
  WARNING: The script virtualenv.exe is installed in 'C:\Users\kingo\AppData\Roaming\Python\Python311\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.

[notice] A new release of pip available: 22.3.1 -> 23.2.1
[notice] To update, run: C:\Users\kingo\.espressif\tools\idf-python\3.11.2\python.exe -m pip install --upgrade pip
C:\Users\kingo\.espressif\tools\idf-python\3.11.2\python.exe: No module named virtualenv
Traceback (most recent call last):
  File "C:\Users\kingo\esp\esp-idf\tools\idf_tools.py", line 1976, in <module>
    main(sys.argv[1:])
  File "C:\Users\kingo\esp\esp-idf\tools\idf_tools.py", line 1972, in main
    action_func(args)
  File "C:\Users\kingo\esp\esp-idf\tools\idf_tools.py", line 1598, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv',
  File "subprocess.py", line 413, in check_call
subprocess.CalledProcessError: Command '['C:\\Users\\kingo\\.espressif\\tools\\idf-python\\3.11.2\\python.exe', '-m', 'virtualenv', '--python', 'C:\\Users\\kingo\\.espressif\\tools\\idf-python\\3.11.2\\python.exe', '--seeder', 'pip', 'C:\\Users\\kingo\\.espressif\\python_env\\idf4.4_py3.11_env']' returned non-zero exit status 1.
Command failed: "C:\Users\kingo\.espressif\tools\idf-python\3.11.2\python.exe" "C:\Users\kingo\esp\esp-idf\tools\idf_tools.py" install-python-env
  WARNING: The script virtualenv.exe is installed in 'C:\Users\kingo\AppData\Roaming\Python\Python311\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.

[notice] A new release of pip available: 22.3.1 -> 23.2.1
[notice] To update, run: C:\Users\kingo\.espressif\tools\idf-python\3.11.2\python.exe -m pip install --upgrade pip
C:\Users\kingo\.espressif\tools\idf-python\3.11.2\python.exe: No module named virtualenv
Traceback (most recent call last):
  File "C:\Users\kingo\esp\esp-idf\tools\idf_tools.py", line 1976, in <module>
    main(sys.argv[1:])
  File "C:\Users\kingo\esp\esp-idf\tools\idf_tools.py", line 1972, in main
    action_func(args)
  File "C:\Users\kingo\esp\esp-idf\tools\idf_tools.py", line 1598, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv',
  File "subprocess.py", line 413, in check_call
subprocess.CalledProcessError: Command '['C:\\Users\\kingo\\.espressif\\tools\\idf-python\\3.11.2\\python.exe', '-m', 'virtualenv', '--python', 'C:\\Users\\kingo\\.espressif\\tools\\idf-python\\3.11.2\\python.exe', '--seeder', 'pip', 'C:\\Users\\kingo\\.espressif\\python_env\\idf4.4_py3.11_env']' returned non-zero exit status 1.
```

us666666, 2023-09-25:
c:\Espressif\frameworks\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.htm

dobairoland, 2023-09-25:
Duplicate of https://github.com/espressif/vscode-esp-idf-extension/issues/1047

closed 2023-09-25 by dobairoland