espressif/esp-idf #12303: efuse C file generation that does not modify existing auto-generated files

rretanubun, 2023-09-25:
### Is your feature request related to a problem?

Each time we update esp-idf version, we need to manually port `components/efuse` since part of its use case is to generate (and overwrite) a auto-generated file.

### Describe the solution you'd like.

A way to cleanly use `components/efuse` for each esp-idif update and still build with a custom efuse table.

rretanubun, 2023-10-04:
Hi @KonstantinKondrashov 

I was not aware of a `custom efuse csv file` -- thanks for pointing that out.
(using default KConfig options) I was able to generate
-  `main/esp_efuse_custom_table.c` 
- `main/include/esp_efuse_custom_table.h`

However, I am not sure how to get the other env/esp-idf/components/efuse C files to
`#include "esp_efuse_custom_table.h"` instead of `#include "esp_efuse_table.h"`
for example, on: https://github.com/espressif/esp-idf/blob/master/components/efuse/esp32/esp_efuse_fields.c#L9

what am I missing?

KonstantinKondrashov, 2023-10-04:
@rretanubun 
There are two efuse tables:
- common efuse table in "esp_efuse_table.h", it is available in the IDF.
- custom efuse table in "esp_efuse_custom_table.h", it is available only in your project.
Do you want to use the customer table in IDF too? I guess that you can `#include "esp_efuse_custom_table.h"`.

rretanubun, 2023-10-04:
@KonstantinKondrashov : 

Does it make sense for the `components/efuse/` C files to have something like:

```
#ifdef CONFIG_EFUSE_CUSTOM_TABLE
#include esp_efuse_custom_table.h // ideally use the name picked by CONFIG_EFUSE_CUSTOM_TABLE_FILENAME
#else
#include esp_efuse_table.h
#endif 
```

That way customers can use the custom_efuse without needed to modify files inside `component/efuse`

For context, the reason we need a custom esp efuse table is that we are re-using some of the unused bits available in BLOCK3, for example:
```
ADC1_TP_LOW,            EFUSE_BLK3,    96,    7,    TP_REG EFUSE_RD_ADC1_TP_LOW
ADC2_TP_LOW,            EFUSE_BLK3,    112,   7,    TP_REG EFUSE_RD_ADC2_TP_LOW
ADC1_TP_HIGH,           EFUSE_BLK3,    103,   9,    TP_REG EFUSE_RD_ADC1_TP_HIGH
ADC2_TP_HIGH,           EFUSE_BLK3,    119,   9,    TP_REG EFUSE_RD_ADC2_TP_HIGH

to

CUSTOM_001,             EFUSE_BLK3,    60,   95,    Sample Custom Field
```
to a custom group of fields.

KonstantinKondrashov, 2023-10-05:
@rretanubun 
No, we can not do that due to, in this case, the efuse component should depend on the main component, which we need to avoid. But I have a solution for that. I guess that you have faced the issue that common and custom efuses overlap each other, this is why you decided to copy the efuse component into your project. It can be solved by using the following command (see below), the cmd is the same as for the generation of custom efuse files but without the common CSV file, it helps to skip the check of overlapping and crossing efuses. (see the cmds [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/efuse.html#efuse-table-gen-py-tool))

```bash
cd $IDF_PATH/components/efuse/
./efuse_table_gen.py --idf_target esp32 PROJECT_PATH/main/esp_efuse_custom_table.csv
```

rretanubun, 2023-10-05:
@KonstantinKondrashov
Ah okay. I think you idea should work for our use case. Thanks very much for your time!