espressif/esp-idf #12305: error: 'btc_ble_mesh_time_scene_client_arg_deep_free' undeclared

hipal31, 2023-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

I'm using Scene client in ble-mesh. While compiling my code i'm getting error 
`D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/api/models/esp_ble_mesh_time_scene_model_api.c: In function 'esp_ble_mesh_time_scene_client_get_state':
D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/api/models/esp_ble_mesh_time_scene_model_api.c:45:17: error: 'btc_ble_mesh_time_scene_client_arg_deep_free' undeclared (first use in this function); did you mean 'btc_ble_mesh_time_scene_client_arg_deep_copy'?
   45 |                 btc_ble_mesh_time_scene_client_arg_deep_free) == BT_STATUS_SUCCESS ? ESP_OK : ESP_FAIL);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 btc_ble_mesh_time_scene_client_arg_deep_copy`
I think there's declaration missing for function `btc_ble_mesh_time_scene_client_arg_deep_free`

### What is the actual behavior?

Should compile fine

### Steps to reproduce.

1. Enable scene client from menuconfig
2. add scene client model in code
3. Compile

### Debug Logs.

```plain
D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/api/models/esp_ble_mesh_time_scene_model_api.c: In function 'esp_ble_mesh_time_scene_client_get_state':
D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/api/models/esp_ble_mesh_time_scene_model_api.c:45:17: error: 'btc_ble_mesh_time_scene_client_arg_deep_free' undeclared (first use in this function); did you mean 'btc_ble_mesh_time_scene_client_arg_deep_copy'?
   45 |                 btc_ble_mesh_time_scene_client_arg_deep_free) == BT_STATUS_SUCCESS ? ESP_OK : ESP_FAIL);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 btc_ble_mesh_time_scene_client_arg_deep_copy
D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/api/models/esp_ble_mesh_time_scene_model_api.c:45:17: note: each undeclared identifier is reported only once for each function it appears in
D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/api/models/esp_ble_mesh_time_scene_model_api.c: In function 'esp_ble_mesh_time_scene_client_set_state':
D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/api/models/esp_ble_mesh_time_scene_model_api.c:70:17: error: 'btc_ble_mesh_time_scene_client_arg_deep_free' undeclared (first use in this function); did you mean 'btc_ble_mesh_time_scene_client_arg_deep_copy'?
   70 |                 btc_ble_mesh_time_scene_client_arg_deep_free) == BT_STATUS_SUCCESS ? ESP_OK : ESP_FAIL);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 btc_ble_mesh_time_scene_client_arg_deep_copy
D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/api/models/esp_ble_mesh_time_scene_model_api.c: In function 'esp_ble_mesh_time_scene_client_get_state':
D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/api/models/esp_ble_mesh_time_scene_model_api.c:46:1: error: control reaches end of non-void function [-Werror=return-type]
   46 | }
      | ^
D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/api/models/esp_ble_mesh_time_scene_model_api.c: In function 'esp_ble_mesh_time_scene_client_set_state':
D:/Frameworks/esp-idf-v5.1.1/components/bt/esp_ble_mesh/api/models/esp_ble_mesh_time_scene_model_api.c:71:1: error: control reaches end of non-void function [-Werror=return-type]
   71 | }
      |
```

hipal31, 2023-10-04:
@JiaLWang Can you help?

gnameg, 2023-10-11:
Hi,
    This problem has been fixed up, but It would take some time to sync the code to Github.

closed 2023-10-11 by hipal31