espressif/esp-idf #12306: WebSocket transport crash on HTTP 500 response from the server

FcoFabra, 2023-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

The problem is related to transport_ws.c (WebSocket transport for TCP).

During the HTTP handshake required to change to WebSocket protocol, the server can respond with a HTTP document.
When conneting to a specific server, the response was HTTP 500 server error, with HTML content with the specific error dump of the server application (Java stack).
The provided content was about 8KB, and WS_BUFFER_SIZE was set to 4096 bytes. There is no header accepting the protocol update to WebSocket.

The device should fail trying to connect, and if we try again to connect, it should keep failing without rebooting.

### What is the actual behavior?

The first time I try to connect, everything seems to work, but the second time I try to connect the system reboots because some memory was filled beyond ws->buffer.
If I change WS_BUFFER_SIZE to 10240 bytes, the issue disapears.

### Steps to reproduce.

Step 1. Select a WebSocket server that responds with a very long payload during the protocol update negotiation. Configure a sample proyect to connect to that server in infinite loop.

Step 2. Configure WS_BUFFER_SIZE with a size lower than the payload that sends the server. Configure heap poisoning to comprehensive and stack check to ALL.

Step 3. Program and test the application. The first time it tries to connect it fails, but it tries again to connect and the device reboots due to corruption.

gabsuren, 2023-10-06:
@FcoFabra Thank you for sharing the report. I followed the steps you outlined to try and replicate the issue. To summarize, I performed the following actions:
```
esp_websocket_client_start(client); // Initiated the websocket client
...
Received a large 8kb data packet from the server once a connection was established
...
esp_websocket_client_close(client, portMAX_DELAY); // Terminated the websocket client
esp_websocket_client_destroy(client); // Freed up the client's memory

esp_websocket_client_start(client); // Restarted the websocket client
```
However, I didn't encounter any crashes during this process. To better understand and replicate the issue, could you kindly share the client code?"

Thanks you
Suren

FcoFabra, 2023-10-06:
I'm sorry but the code is protected by law.
The error does not happen in the first connection, but when I try to reconnect after unsuccess.
It seems that the first execution corrupts data beyond the websocket buffer.
I am using the last official version of WebSocket found in the online Component Manager.
The first response does not have an "Accepted" response, because it is a regular HTTP error web page... but very big.

Remember... the error only happens when the connection FAILS. In your response you say that you receive a large data packet once the connection is established!

Thank you for your support.

Best regards

gabsuren, 2023-10-10:
@FcoFabra thank you for you response. I attempted to run the same code, but the initial attempt failed. It then tried to reconnect in approximately 10 seconds. Please take a look at the code provided below.
```
#include <stdio.h>
#include "esp_wifi.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_event.h"
#include "protocol_examples_common.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "freertos/event_groups.h"

#include "esp_log.h"
#include "esp_websocket_client.h"
#include "esp_event.h"
#include <cJSON.h>

#define NO_DATA_TIMEOUT_SEC 5

static const char *TAG = "websocket";

static TimerHandle_t shutdown_signal_timer;
static SemaphoreHandle_t shutdown_sema;

static void log_error_if_nonzero(const char *message, int error_code)
{
    if (error_code != 0) {
        ESP_LOGE(TAG, "Last error %s: 0x%x", message, error_code);
    }
}

static void shutdown_signaler(TimerHandle_t xTimer)
{
    ESP_LOGI(TAG, "No data received for %d seconds, signaling shutdown", NO_DATA_TIMEOUT_SEC);
    xSemaphoreGive(shutdown_sema);
}

#if CONFIG_WEBSOCKET_URI_FROM_STDIN
static void get_string(char *line, size_t size)
{
    int count = 0;
    while (count < size) {
        int c = fgetc(stdin);
        if (c == '\n') {
            line[count] = '\0';
            break;
        } else if (c > 0 && c < 127) {
            line[count] = c;
            ++count;
        }
        vTaskDelay(10 / portTICK_PERIOD_MS);
    }
}

#endif /* CONFIG_WEBSOCKET_URI_FROM_STDIN */

static void websocket_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data)
{
    esp_websocket_event_data_t *data = (esp_websocket_event_data_t *)event_data;
    switch (event_id) {
    case WEBSOCKET_EVENT_CONNECTED:
        ESP_LOGI(TAG, "WEBSOCKET_EVENT_CONNECTED");
        break;
    case WEBSOCKET_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "WEBSOCKET_EVENT_DISCONNECTED");
        log_error_if_nonzero("HTTP status code",  data->error_handle.esp_ws_handshake_status_code);
        if (data->error_handle.error_type == WEBSOCKET_ERROR_TYPE_TCP_TRANSPORT) {
            log_error_if_nonzero("reported from esp-tls", data->error_handle.esp_tls_last_esp_err);
[... 151 lines cut ...]
        for nr, conn in self.server.connections.items():
            conn.sendMessage(data)

    def run(self):
        self.server = SimpleWebSocketServer('', self.port, WebsocketTestEcho)
        while not self.exit_event.is_set():
            self.server.serveonce()

    def __init__(self, port):
        self.port = port
        self.exit_event = Event()
        self.thread = Thread(target=self.run)
        self.thread.start()

    def __enter__(self):
        return self

    def __exit__(self, exc_type, exc_value, traceback):
        self.exit_event.set()
        self.thread.join(10)
        if self.thread.is_alive():
            print('Thread cannot be joined', 'orange')

ws=Websocket(8080)
```

closed 2024-05-29 by gabsuren