espressif/esp-idf #12307: Heap corruption in WiFi driver

KStopp, 2023-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1709-g4fc2e5cb95

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROVER-IE | Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

We have a web application running on a custom server with the ESP32-WROVER-IE MCU. Several situations should run like this:

1. When changing the wifi settings the ESP should not crash.
2. When logging in to the app via wifi the ESP should not crash (data transmission).
3. When reloading the Web App browser page the ESP should not crash (data transmission).

### What is the actual behavior?

The wifi driver causes the ESP to crash in several situations. By changing a setting like the wifi channel or IP the wifi is unreachable for a moment, which is normal as I guess. Instead of a re-establishment the ESP often crashes (see Logs below). If there are client side changes like logging in and loading the main web page or refreshing the web page the the ESP crashes sometimes as well (please find respective Logs attached below). 

### Steps to reproduce.

**Changing wifi settings**

1. Changed a setting like the channel (number does not matter)
2. Waited until wifi connection is stable again
3. Repeated several times until crash occured with _DEBUG_LOG_1_

**Reloading page / logging in**
- actual data transmission over webserver component of esp idf causes crash. _Here we got two different crash _logs.__

### Debug Logs.

```plain
Debug_Log_1 (re-establishing the wifi):
CORRUPT HEAP: Bad tail at 0x3fff1b49. Expected 0xbaad5678 got 0x6f630373

assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)

Backtrace: 0x40081f2a:0x3ffe5e20 0x4008b405:0x3ffe5e40 0x400959b9:0x3ffe5e60 0x40090e55:0x3ffe5f90 0x4008276e:0x3ffe5fb0 0x400959f9:0x3ffe5fd0 0x40097dc5:0x3ffe5ff0 0x401ac410:0x3ffe6010 0x400972b5:0x3ffe6030
0x40081f2a: panic_abort at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/esp_system/panic.c:452

0x4008b405: esp_system_abort at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/esp_system/port/esp_system_chip.c:93

0x400959b9: __assert_func at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/newlib/assert.c:81

0x40090e55: multi_heap_free at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/multi_heap_poisoning.c:276 (discriminator 1)

0x4008276e: heap_caps_free at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/heap_caps.c:388

0x400959f9: free at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/newlib/heap.c:39

0x40097dc5: esf_buf_recycle at ??:?

0x401ac410: ppProcTxDone at ??:?

0x400972b5: ppTask at ??:?

Debug_Log_2 (occured during data transmission):
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40090e54  PS      : 0x00060c33  A0      : 0x80090698  A1      : 0x3ffe2fa0
0x40090e54: remove_free_block at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/tlsf/tlsf.c:332
 (inlined by) block_locate_free at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/tlsf/tlsf.c:567
 (inlined by) tlsf_malloc at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/tlsf/tlsf.c:1004

A2      : 0x3ffe4364  A3      : 0x3fff2db0  A4      : 0xce58beef  A5      : 0xc8bdfb2a
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x00000001  A9      : 0x00000005
A10     : 0x00020000  A11     : 0x0000002c  A12     : 0x3ffe440c  A13     : 0x3ffe4388
A14     : 0x3ffe43b8  A15     : 0x3ffda318  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d
EXCVADDR: 0xce58befb  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000
0x4000c349: memcpy in ROM

0x4000c36b: memcpy in ROM

Backtrace: 0x40090e51:0x3ffe2fa0 0x40090695:0x3ffe2fc0 0x400907f5:0x3ffe2fe0 0x40082676:0x3ffe3000 0x40082689:0x3ffe3030 0x400826b5:0x3ffe3050 0x400953f1:0x3ffe3070 0x401245bd:0x3ffe3090 0x40089f00:0x3ffe30b0 0x4008a094:0x3ffe30d0 0x40089edd:0x3ffe30f0 0x400844b1:0x3ffe3110 0x401549d5:0x3ffe3150 0x40156a32:0x3ffe3170 0x40154b63:0x3ffe31d0 0x40098f61:0x3ffe31f0 0x40096cdc:0x3ffe3210
0x40090e51: remove_free_block at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/tlsf/tlsf.c:331
 (inlined by) block_locate_free at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/tlsf/tlsf.c:567
 (inlined by) tlsf_malloc at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/tlsf/tlsf.c:1004

0x40090695: multi_heap_malloc_impl at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/multi_heap.c:207

0x400907f5: multi_heap_malloc at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/multi_heap_poisoning.c:251
 (inlined by) multi_heap_malloc at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/multi_heap_poisoning.c:240

0x40082676: heap_caps_malloc_base at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/heap_caps.c:176

0x40082689: heap_caps_malloc at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/heap_caps.c:197

0x400826b5: heap_caps_malloc_default at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/heap/heap_caps.c:223

0x400953f1: malloc at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/newlib/heap.c:24

0x401245bd: mem_malloc at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/lwip/lwip/src/core/mem.c:209

0x40089f00: esp_pbuf_allocate at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/esp_netif/lwip/netif/esp_pbuf_ref.c:52

0x4008a094: wlanif_input at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/esp_netif/lwip/netif/wlanif.c:151

0x40089edd: esp_netif_receive at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1224

0x400844b1: interceptor_ap_input at C:/Users/user296/Documents/projects/PoE-Switch-outdoor-mini/webinterface/main/network/interceptor.c:514

0x401549d5: hostap_deliver_data at ??:?

0x40156a32: hostap_input at ??:?

0x40154b63: ap_rx_cb at ??:?

0x40098f61: ppRxPkt at ??:?

0x40096cdc: ppTask at ??:?

Debug_Log_3 (after rebooting from Debug_Log_2 with another data transmission):
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4009695d  PS      : 0x00060a30  A0      : 0x800959c5  A1      : 0x3ffdcc90
0x4009695d: ppResortTxAMPDU at ??:?

A2      : 0x0000000a  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffef67c  A7      : 0x3fff3c90  A8      : 0x3fff3c90  A9      : 0x000000c7
A10     : 0x00000000  A11     : 0x0000003f  A12     : 0x00000020  A13     : 0x3ffb1038
A14     : 0x046c3570  A15     : 0x3ffb1624  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000003  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
0x4000c2e0: memcpy in ROM

0x4000c2f6: memcpy in ROM

Backtrace: 0x4009695a:0x3ffdcc90 0x400959c2:0x3ffdcd50 0x4016e649:0x3ffdcd90 0x4016ee7e:0x3ffdcdc0 0x40096d16:0x3ffdce00
0x4009695a: ppResortTxAMPDU at ??:?

0x400959c2: lmacEndFrameExchangeSequence at ??:?

0x4016e649: lmacProcessTxSuccess at ??:?

0x4016ee7e: lmacProcessTxComplete at ??:?

0x40096d16: ppTask at ??:?
```

### More Information.

I need to mention that the points described above do not occur every time. Sometimes it is possible to change the wifi settings several times without causing a crash. This is also the case for loggin in and reloading. I tried the release/v5.1 as well, still got the same issue. Using ethernet everything works fine.

Xiehanxin, 2023-09-27:
hi @KStopp ,  I would like to confirm that the same code can work on release/4.X , right? and if you can provide the sdkconfig of your poject

KStopp, 2023-09-27:
Please find the sdkconfig attached.

We will try to get the same code running on release/4.X . As soon as I have the result I will provide it here.

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/12736270/sdkconfig.zip)

Xiehanxin, 2023-09-28:
thank you for your feedback!

zinke-ct-video, 2023-09-28:
I'm a team mate of @KStopp, we work on the same project and I want to provide some additional insights on this problem. The culprit seems to be `CONFIG_ESP_WIFI_AMSDU_TX_ENABLED` .

Our project grew over a couple of years, and we have used different versions of ESP-IDF. For a long time we had the AMSDU feature enabled for performance reasons and rarely touched the sdkconfig when updating ESP-IDF. Now I'm on ESP-IDF release 5.1.1 and I can reproduce the bug if AMSDU is enabled. The bug disappears if I change only this one config option from enabled to disabled.

Here is another [sdkconfig](https://github.com/espressif/esp-idf/files/12751184/sdkconfig.zip) example file. Just enable `CONFIG_ESP_WIFI_AMSDU_TX_ENABLED` to make it crash very soon when there is traffic on the WiFi AP.

Xiehanxin, 2023-10-11:
hi  @KStopp  @zinke-ct-video   could you please provide a small reproduce codes for us, if we can reproduce it on our side I think it is better for debug

AxelLin, 2023-11-05:
@KStopp @zinke-ct-video 
Any update regarding the issue, and any feedback for comment https://github.com/espressif/esp-idf/issues/12307#issuecomment-1756633951 ?

zinke-ct-video, 2023-11-06:
I just tried the WiFi AP example, added a simple webserver sending a few MB of random data to the browser. Unfortunately I am not able to reproduce the error with a small example project. The AMSDU TX feature seems not to cause any problems in this example. I don't know how to figure out what exactly the reason for the crashes is in our case because it's really complicated to strip our project code down into smaller pieces until we find the exact cause.

At the moment we are fine with the solution to disable the AMSDU TX feature in our project. From our side it would be okay, if you close the issue because I don't see any possibility to provide a small error reproducing code.

mhdong, 2024-03-25:
Hi @KStopp 
Issue may solved with https://github.com/espressif/esp-idf/commit/ec16fa7dd55cd71e8736d5097ddd13a611b06115.
Please confirm if the isue still exists.

closed 2024-05-08 by Sherry616