espressif/esp-idf #12308: Connection problems with BLE nimble with low power

agustinebrecht, 2023-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I am trying to run an example of Nimble ble modified to work at low energy with the esp32H2. When the device enters light sleep, the consumption is indicated in the example of bluetooth/nimble/power_save/ around 20uA, the problem is that when I exit light sleep and try to connect a mobile device to my BLE server, it gives me the following mistake.

[image]

by which it can be seen that the connection is never made.

With various tests that I carried out, I discovered that if I have defined the following macro in the sdkconfig CONFIG_BT_LE_LP_CLK_SRC_DEFAULT it is impossible to connect to the server with the cellular device, however if I remove this macro I can connect correctly to the server, with the exception that in light mode sleep started to have an average consumption of 3mA which is not acceptable for my use.

I attach the example code used to reproduce my case, I am using release 5.1.1 on the esp32H2.

```

/*
 * SPDX-FileCopyrightText: 2015-2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include "esp_log.h"
#include "nvs_flash.h"
/* power management */
#include "esp_pm.h"
/* BLE */
#include "nimble/nimble_port.h"
#include "nimble/nimble_port_freertos.h"
#include "host/ble_hs.h"
#include "host/util/util.h"
#include "console/console.h"
#include "services/gap/ble_svc_gap.h"
#include "nimble/nimble_port.h"
#include "nimble/nimble_port_freertos.h"
#include "host/ble_hs.h"
#include "host/util/util.h"
#include "console/console.h"
#include "services/gap/ble_svc_gap.h"
#include <stdbool.h>
#include "nimble/ble.h"
#include "modlog/modlog.h"
#include "esp_bt.h"
#include "esp_sleep.h"
#include "nimble/nimble_npl.h"
#include "host/ble_hs.h"
#include "host/ble_uuid.h"
#include "services/gap/ble_svc_gap.h"
#include "services/gatt/ble_svc_gatt.h"

#include "services/ans/ble_svc_ans.h"
#include "esp_bt.h"

static const char *tag = "NimBLE_BLE_PRPH";

#define GATT_SVR_SVC_ALERT_UUID               0x1811

DRAM_ATTR esp_pm_lock_handle_t service_light_sleep_pm_lock = NULL;

/**
 * @brief Bloqueo que entre en light sleep
 * 
 */
void systemBlockLightSleep(void){
    if(NULL != service_light_sleep_pm_lock ){

[... 557 lines cut ...]

CONFIG_BT_LE_WAKEUP_SOURCE_BLE_RTC_TIMER=y

CONFIG_BT_LE_LP_CLK_SRC_DEFAULT=y

#
# Power Management
#

CONFIG_PM_POWER_DOWN_CPU_IN_LIGHT_SLEEP=y
# end of Power Management

#
# RTC Clock Config
#

CONFIG_RTC_CLK_SRC_EXT_CRYS=y
# end of RTC Clock Config

I need to add that I only try to connect when the device is not in light sleep
I attach an image of the consumption

[image]

I would appreciate it if anyone could help me with my problem.

rahult-github, 2023-10-11:
Hi @agustinebrecht ,

As mentioned in the description, if BT_LE_LP_CLK_SRC_DEFAULT is set, the BT LE timer uses the same slow clock source as the RTC module, among which only the external 32kHz crystal has the accuracy to establish and maintain an ACL connection. Any other clock source can only support legacy ADV and SCAN in sleep mode.

So if you want to set BT_LE_LP_CLK_SRC_DEFAULT and make a connection, you need to set RTC_CLK_SRC_EXT_CRYS as well (provided that your chip has the external 32.768 kHz crystal).

dashxdr, 2023-11-05:
Could the error code be based on the BLE_HS_EDISABLED value of 30? 30+32+512 = 574
It's defined in ble_hs.h. Just guessing, I don't really know.
BLE_HS_ERR_HCI_BASE is the 512, I don't know what the 32 is, and nothing is 62.

rahult-github, 2023-11-06:
> Could the error code be based on the BLE_HS_EDISABLED value of 30? 30+32+512 = 574 It's defined in ble_hs.h. Just guessing, I don't really know. BLE_HS_ERR_HCI_BASE is the 512, I don't know what the 32 is, and nothing is 62.

Error 574 is decimal value, so its hex value is 23E .  This translates to 200 + 3E in stack error assignment, so actual error code is 3E. This usually implies LL sync couldn't be made when connection was attempted as mentioned in BT spec.

agustinebrecht, 2024-03-30:
By adding the glass and making the necessary configurations, consumption is reduced

closed 2024-03-30 by agustinebrecht