espressif/esp-idf #12309: Power consumption

weiwi, 2023-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I am getting similar power consumption to what is in the readme file. I was wondering if a lower power consumption can be achieved using auto lightsleep with wifi monitoring, like shown in the table from expressif. Thanks.  
[image]

esp-lis, 2024-01-17:
@weiwi You can use the esp32c3, esp32c2 or esp32c6 for the same function with lower current consumption, The following gives the power test report of esp32c3:
[image]

esp-lis, 2024-01-19:
esp32c6 current consumption reference is as follows:
[image]

closed 2024-08-16 by Sherry616