espressif/esp-idf #12310: The correct way to set openthread-specific configuration variables

vke, 2023-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Some of openthread-specific options are set through Kconfig menu. For example, the `OPENTHREAD_CONFIG_DUA_ENABLE` thread option is set using the `OPENTHREAD_DUA_ENABLE` Kconfig option. The build system includes a corresponding header file (`components/openthread/private_include/openthread-core-esp32x-ftd-config.h` for example) which translates Kconfig's `CONFIG_OPENTHREAD_DUA_ENABLE` into a thread's `OPENTHREAD_CONFIG_DUA_ENABLE`. This header file is included by the openthread library through the `OPENTHREAD_CONFIG_FILE` definition.

But what if the required option is not in the Kconfig menu and corresponding `components/openthread/private_include/openthread-core-esp32x-***-config.h` header file? How to define a option in this case?

For example, I want to set the openthread option `OPENTHREAD_CONFIG_RADIO_STATS_ENABLE`, which enables the `radio stats` openthread cli command (`examples/openthread/ot_cli` project). But neither in the Kconfig menu nor in the settings translation header file is there any mention of radio statistics. What is the correct way to enable this option without changing the SDK source code?

zwx1995esp, 2023-10-12:
Hi vke, Sorry for the delay, you can just add the config in the file `components/openthread/private_include/openthread-core-esp32x-***-config.h` manually.

For example: add this line in the file `components/openthread/private_include/openthread-core-esp32x-radio-config.h`

```
#define OPENTHREAD_CONFIG_RADIO_STATS_ENABLE 1
```

vke, 2023-10-12:
Thanks for your reply! Yes it is possible and I will do as you said, thanks! But in this case, I will change the SDK files (not my own project files) and I will have to merge this changes with each SDK update.

vke, 2023-10-16:
> Ok, right, I will create a MR in our internal branch add this config to the IDF and make it as a menuconfig, but it need to wait for one or two weeks for getting the changes on the github.

Just for `OPENTHREAD_CONFIG_RADIO_STATS_ENABLE`, or something general that allows to change any option, such as including a user config to override any openthread config defines not included in menuconfig?

zwx1995esp, 2023-10-16:
Also I can add a menuconfig for enable custom openthread config, if you enable this feature, you can use your own config.

zwx1995esp, 2023-10-16:
> Just for `OPENTHREAD_CONFIG_RADIO_STATS_ENABLE`, or something general that allows to change any option, such as including a user config to override any openthread config defines not included in [...]

Yah, the same as your thought.

vke, 2023-10-16:
> Sorry, I read your commit again, and. I think I can add a config for eanbling the custom openthread config feature, and if the user enable this config, they can use there own config(the file name can be configured still via menuconfig), how about this way?

In my opinion, this would be the ideal solution, and also the simplest. Just an option to include a user's openthread config. The user will be able to independently set any definitions, even new ones that will appear in future versions of openthread.

Thank you!

chshu, 2025-02-26:
Now the ESP Thread SDK supports the custom header file feature, which allows users to define their own openthread-specific configurations, please refer to this link for more details:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/openthread.html#openthread-macro-definitions

closed 2025-02-26 by chshu