espressif/esp-idf #12311: https_request => esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700

greenpdx, 2023-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Seeed Xiao ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

I expect to connect to webserver using TLS. 
I am using examples/protocols/https_request

### What is the actual behavior?

I get a cert verify error
I (539) example_connect: Connecting to Gringo24...
I (539) example_connect: Waiting for IP(s)
I (2949) wifi:new:<6,2>, old:<1,0>, ap:<255,255>, sta:<6,2>, prof:1
I (3619) wifi:state: init -> auth (b0)
I (3619) wifi:state: auth -> assoc (0)
I (3629) wifi:state: assoc -> run (10)
I (3639) wifi:<ba-add>idx:0 (ifx:0, d8:50:e6:cf:e4:da), tid:0, ssn:0, winSize:64
I (3649) wifi:connected with Gringo24, aid = 4, channel 6, 40D, bssid = d8:50:e6:cf:e4:da
I (3649) wifi:security: WPA2-PSK, phy: bgn, rssi: -42
I (3649) wifi:pm start, type: 1

I (3649) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (3689) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (5399) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:3685:18ff:fe05:e8c8, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (7659) esp_netif_handlers: example_netif_sta ip: 192.168.42.47, mask: 255.255.255.0, gw: 192.168.42.1
I (7659) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.42.47
I (7669) example_common: Connected to example_netif_sta
I (7669) example_common: - IPv4 address: 192.168.42.47,
I (7679) example_common: - IPv6 address: fe80:0000:0000:0000:3685:18ff:fe05:e8c8, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (7689) example: Start https_request example
I (7689) example: https_request using crt bundle
I (7699) main_task: Returned from app_main()
E (8399) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
E (8399) esp-tls: Failed to open new connection
E (8399) example: Connection failed...

### Steps to reproduce.

1.  idf.py set-target esp32c3
2. idf.py menuconfig (add mbedtls debug, change wifi connection)
3.  idf.py build
4. idf.py -p /dev/ttyACM0 flash monitor
 

### Debug Logs.

```plain
I (5424) example: Start https_request example
I (5434) example: https_request using crt bundle
I (5434) main_task: Returned from app_main()
I (5644) mbedtls: ssl_tls.c:3939 => handshake

I (5654) mbedtls: ssl_msg.c:2124 => flush output

I (5654) mbedtls: ssl_msg.c:2133 <= flush output

I (5654) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_HELLO_REQUEST

I (5664) mbedtls: ssl_msg.c:2124 => flush output

I (5664) mbedtls: ssl_msg.c:2133 <= flush output

I (5674) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_CLIENT_HELLO

I (5684) mbedtls: ssl_client.c:906 => write client hello

W (5684) mbedtls: ssl_client.c:258 got supported group(001d)

W (5694) mbedtls: ssl_client.c:258 got supported group(0017)

W (5704) mbedtls: ssl_client.c:258 got supported group(0018)

W (5704) mbedtls: ssl_client.c:258 got supported group(0019)

W (5714) mbedtls: ssl_client.c:258 got supported group(001a)

W (5714) mbedtls: ssl_client.c:258 got supported group(001b)

W (5724) mbedtls: ssl_client.c:258 got supported group(001c)

I (5734) mbedtls: ssl_msg.c:2554 => write handshake message

I (5734) mbedtls: ssl_msg.c:2714 => write record

I (5744) mbedtls: ssl_msg.c:2851 <= write record

I (5754) mbedtls: ssl_msg.c:2675 <= write handshake message

I (5754) mbedtls: ssl_client.c:994 <= write client hello

I (5764) mbedtls: ssl_msg.c:2124 => flush output

I (5764) mbedtls: ssl_msg.c:2138 message length: 250, out_left: 250

I (5774) mbedtls: ssl_msg.c:2145 ssl->f_send() returned 250 (-0xffffff06)

I (5784) mbedtls: ssl_msg.c:2172 <= flush output

I (5784) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_SERVER_HELLO

I (5794) mbedtls: ssl_tls12_client.c:1195 => parse server hello

I (5804) mbedtls: ssl_msg.c:3887 => read record

I (5804) mbedtls: ssl_msg.c:1926 => fetch input

I (5814) mbedtls: ssl_msg.c:2066 in_left: 0, nb_want: 5

I (5954) mbedtls: ssl_msg.c:2086 in_left: 0, nb_want: 5

I (5954) mbedtls: ssl_msg.c:2089 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (5954) mbedtls: ssl_msg.c:2111 <= fetch input

I (5964) mbedtls: ssl_msg.c:1926 => fetch input

I (5964) mbedtls: ssl_msg.c:2066 in_left: 5, nb_want: 58

I (5974) mbedtls: ssl_msg.c:2086 in_left: 5, nb_want: 58

I (5984) mbedtls: ssl_msg.c:2089 ssl->f_recv(_timeout)() returned 53 (-0xffffffcb)

I (5984) mbedtls: ssl_msg.c:2111 <= fetch input

I (5994) mbedtls: ssl_msg.c:3959 <= read record

I (6004) mbedtls: ssl_tls12_client.c:1446 server hello, total extension length: 9

I (6004) mbedtls: ssl_tls12_client.c:1660 <= parse server hello

I (6014) mbedtls: ssl_msg.c:2124 => flush output

I (6014) mbedtls: ssl_msg.c:2133 <= flush output

I (6024) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_SERVER_CERTIFICATE

I (6034) mbedtls: ssl_tls.c:7559 => parse certificate

I (6034) mbedtls: ssl_msg.c:3887 => read record

I (6044) mbedtls: ssl_msg.c:1926 => fetch input

I (6054) mbedtls: ssl_msg.c:2066 in_left: 0, nb_want: 5

I (6054) mbedtls: ssl_msg.c:2086 in_left: 0, nb_want: 5

I (6064) mbedtls: ssl_msg.c:2089 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (6074) mbedtls: ssl_msg.c:2111 <= fetch input

I (6074) mbedtls: ssl_msg.c:1926 => fetch input

I (6084) mbedtls: ssl_msg.c:2066 in_left: 5, nb_want: 4024

I (6084) mbedtls: ssl_msg.c:2086 in_left: 5, nb_want: 4024

I (6094) mbedtls: ssl_msg.c:2089 ssl->f_recv(_timeout)() returned 4019 (-0xfffff04d)

I (6104) mbedtls: ssl_msg.c:2111 <= fetch input

I (6134) mbedtls: ssl_msg.c:3959 <= read record

W (6334) mbedtls: ssl_tls.c:7378 x509_verify_cert() returned -9984 (-0x2700)

I (6334) mbedtls: ssl_msg.c:4868 => send alert message

I (6334) mbedtls: ssl_msg.c:2714 => write record

I (6344) mbedtls: ssl_msg.c:2124 => flush output

I (6354) mbedtls: ssl_msg.c:2138 message length: 7, out_left: 7

I (6354) mbedtls: ssl_msg.c:2145 ssl->f_send() returned 7 (-0xfffffff9)

I (6364) mbedtls: ssl_msg.c:2172 <= flush output

I (6364) mbedtls: ssl_msg.c:2851 <= write record

I (6374) mbedtls: ssl_msg.c:4880 <= send alert message

I (6384) mbedtls: ssl_tls.c:3950 <= handshake

E (6384) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
E (6394) esp-tls: Failed to open new connection
E (6394) example: Connection failed...
```

### More Information.

The https_example fails with original configuration.
I have tried setting allow insecure connections but it still fails. 

I have tried using my own  CA chain and keys but always the same error.
mbedtls: ssl_tls.c:7378 x509_verify_cert() returned -9984 (-0x2700)

mahavirj, 2023-09-27:
@greenpdx 

- Did you enable `MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE` to add your own certs to the default bundle?
- What is the endpoint that you are trying to connect to?
- Can you please share your sample application (or patch) and sdkconfig here?

greenpdx, 2023-09-28:
I did in the past I added my signing certs to the bundle, but I wanted to start with a clean https_request, so not on this try.

In the original it tries to access
#define WEB_SERVER "www.howsmyssl.com"
#define WEB_PORT "443"
#define WEB_URL "https://www.howsmyssl.com/a/check"

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/12744173/sdkconfig.txt)

mahavirj, 2023-09-28:
@greenpdx Do you mean that the failure is happening with the IDF example and its default (`howsmyssl.com`) endpoint?

greenpdx, 2023-09-28:
yes

greenpdx, 2023-09-29:
I am now using https_mbedtls with a local net nginx server.
I the root cert is a EC P-256 cert
I created three intermediate certs, EC P-256 Ed2551 and rsa
I then created 9 leaf certs
EC, Ed2551 and RSA for each of the intermediate certs

I     L
rsa rsa   works
rsa ec    works
rsa ed    fails

ec  rsa   works
ec  ec    works
ec  ed   fails

ed  *    all fail

ed25519 does not work for certificates

greenpdx, 2023-09-29:
I have https_request working with root CA P-256, intermediate cert P-256, and leaf cert P-256
only using crt bundle works
using cacert_buf and global ca store still fail

greenpdx, 2023-09-30:
I am going to open a new issue about not supporting ed25519

closed 2023-09-30 by greenpdx