espressif/esp-idf #12312: esp_wifi_set_country() for openthread border router

mmt-10, 2023-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Hi,

I would like to add `esp_wifi_set_country()` to `examples/openthread/ot_br`.

So, I added below Lines between L113 to L114 on `esp_ot_br.c`.
Is it OK?
If you have any advice, please let me know.

[https://github.com/espressif/esp-idf/blob/e088c3766ba440e72268b458a68f27b6e7d63986/examples/openthread/ot_br/main/esp_ot_br.c#L113](https://github.com/espressif/esp-idf/blob/e088c3766ba440e72268b458a68f27b6e7d63986/examples/openthread/ot_br/main/esp_ot_br.c#L113)

```cpp
#if CONFIG_EXAMPLE_CONNECT_WIFI
static wifi_country_t wifi_country = {.cc = "US", .schan = 1, .nchan = 11};
esp_wifi_set_country(&wifi_country);
#if CONFIG_OPENTHREAD_BR_AUTO_START
```

### What is the actual behavior?

none

### Steps to reproduce.

1. Step
2. Step
3. Step
...

zwx1995esp, 2023-10-12:
Sorry for the delay, actually this API is wifi-related, I will ask someone from wifi group to confirm that if your using is correct. (And this issue is not a bug, so could you please remove the label `type: bug`?)

zwx1995esp, 2023-10-12:
Just confirmed from the WiFi group, `esp_wifi_set_country` should be called after wifi_init, so you need to call this function after example_connect().

mmt-10, 2023-10-12:
Thank you for your reply!
I understand it.
And, How do I remove `type: bug` label?
It is attached automatically.

closed 2023-10-12 by mmt-10