espressif/esp-idf #12314: Compile error when build examples/system/light_sleep with CONFIG_ESP_CONSOLE_UART_DEFAULT=n

AxelLin, 2023-09-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-439-gcb174b0fe1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

build ok.

### What is the actual behavior?

Build error if hitting below settings:

idf.py set-target esp32c6

Using below config:
 # CONFIG_ESP_CONSOLE_UART_DEFAULT is not set
 CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
 # CONFIG_ESP_CONSOLE_UART_CUSTOM is not set
 # CONFIG_ESP_CONSOLE_NONE is not set
 CONFIG_ESP_CONSOLE_SECONDARY_NONE=y

 -c /home/axel/esp/esp-idf/components/esp_hw_support/sleep_modes.c
/home/axel/esp/esp-idf/components/esp_hw_support/sleep_modes.c: In function 'light_sleep_uart_prepare':
/home/axel/esp/esp-idf/components/esp_hw_support/sleep_modes.c:152:61: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
  152 | #define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/esp_hw_support/sleep_modes.c:478:113: note: in expansion of macro 'UART_FLUSH_US_PER_CHAR'
  478 |             (sleep_duration < (int64_t)((UART_LL_FIFO_DEF_LEN - uart_ll_get_txfifo_len(CONSOLE_UART_DEV) + 1) * UART_FLUSH_US_PER_CHAR) + SLEEP_UART_FLUSH_DONE_TO_SLEEP_US)) {
      |                                                                                                                 ^~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/esp_hw_support/sleep_modes.c:152:61: note: each undeclared identifier is reported only once for each function it appears in
  152 | #define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/esp_hw_support/sleep_modes.c:478:113: note: in expansion of macro 'UART_FLUSH_US_PER_CHAR'
  478 |             (sleep_duration < (int64_t)((UART_LL_FIFO_DEF_LEN - uart_ll_get_txfifo_len(CONSOLE_UART_DEV) + 1) * UART_FLUSH_US_PER_CHAR) + SLEEP_UART_FLUSH_DONE_TO_SLEEP_US)) {
      |

Also note, got below error if build with idf.py set-target esp32c3:

/home/axel/esp/esp-idf/examples/system/light_sleep/main/uart_wakeup.c: In function 'uart_initialization':
/home/axel/esp/esp-idf/examples/system/light_sleep/main/uart_wakeup.c:100:23: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
  100 |         .baud_rate  = CONFIG_ESP_CONSOLE_UART_BAUDRATE,
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                       CONFIG_ESP_CONSOLE_UART_NUM

### Steps to reproduce.

build examples/system/light_sleep with below settings:

 # CONFIG_ESP_CONSOLE_UART_DEFAULT is not set
 CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
 # CONFIG_ESP_CONSOLE_UART_CUSTOM is not set
 # CONFIG_ESP_CONSOLE_NONE is not set
 CONFIG_ESP_CONSOLE_SECONDARY_NONE=y

igrr, 2023-09-27:
Thanks for opening the issue! We have merged the fix for it yesterday, should appear on Github soon. Merge commit ID will be 8545f8595f2591e1d172e127e13d2eb0f6145149.

AxelLin, 2023-09-27:
BTW, I hit another build error when try to compile with esp-idf-lib.
/home/axel/esp/esp-idf-dev/esp-idf-lib/components/mhz19b/mhz19b.c:64:23: error: 'UART_SCLK_APB' undeclared (first use in this function); did you mean 'UART_SCLK_RTC'?
   64 |         .source_clk = UART_SCLK_APB,
      |                       ^~~~~~~~~~~~~
      |                       UART_SCLK_RTC

I'm not sure if this is esp-idf-lib issue or esp-idf issue.
(UART_SCLK_APB does not exist for esp32c6 target.)

igrr, 2023-09-27:
The reference to UART_SCLK_APB is in esp-idf-lib, so yes, i think it's an issue there:
https://github.com/UncleRus/esp-idf-lib/blob/15c4da67ef7ec713d9ab6d87c54c4445742620f8/components/mhz19b/mhz19b.c#L63-L65. I think it can be changed to UART_SCLK_DEFAULT.

marciogranzotto, 2023-10-05:
@igrr I don't think this was uploaded to github yet, correct? Any ETA? or can you explain how can I fix it locally so I can continue developing?

AxelLin, 2023-10-13:
FYI, I also got below warnings when CONFIG_ESP_CONSOLE_UART_DEFAULT=n with 5.1 branch.
```
Building C object esp-idf/console/CMakeFiles/__idf_console.dir/esp_console_repl.c.obj/home/axel/esp/esp-idf/components/console/esp_console_repl.c:479:13: warning: 'esp_console_repl_task' defined but not used [-Wunused-function]
  479 | static void esp_console_repl_task(void *args)
      |             ^~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/console/esp_console_repl.c:375:18: warning: 'esp_console_common_init' defined but not used [-Wunused-function]
  375 | static esp_err_t esp_console_common_init(size_t max_cmdline_length, esp_console_repl_com_t *repl_com)
      |                  ^~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/console/esp_console_repl.c:354:18: warning: 'esp_console_setup_history' defined but not used [-Wunused-function]
  354 | static esp_err_t esp_console_setup_history(const char *history_path, uint32_t max_history_len, esp_console_repl_com_t *repl_com)
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/console/esp_console_repl.c:329:18: warning: 'esp_console_setup_prompt' defined but not used [-Wunused-function]
  329 | static esp_err_t esp_console_setup_prompt(const char *prompt, esp_console_repl_com_t *repl_com)
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~

```

AxelLin, 2023-11-02:
> Thanks for opening the issue! We have merged the fix for it yesterday, should appear on Github soon. Merge commit ID will be [8545f85](https://github.com/espressif/esp-idf/commit/8545f8595f2591e1d172e127e13d2eb0f6145149). Will backport to release/v5.1 as well.

Just remind v5.1 branch still needs fix.

AxelLin, 2023-11-22:
> Just remind v5.1 branch still needs fix.

@igrr 
Recent v5.1 update does not include this fix, I'm wondering if it's missed?
I hope a compile fix can be applied asap.

igrr, 2023-11-22:
@AxelLin You are right, sorry, I couldn't get it merged yet due to some of the tests failing... I was looking at this yesterday, hope to get this resolved soon.

closed 2023-12-06 by igrr