espressif/esp-idf #12317: Different Wi-Fi scanning behavior on ESP-IDF 4.4.5 and ESP-IDF 5.1.1

VNovytskyi, 2023-09-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a project in which the ESP32 must be connected to some access point and periodically scanning the other access points. It means that ESP32 works in STA mode and sometimes uses a scan procedure. When scanning, it is very important that the microcontroller remains connected to the access point. To achieve this goal, I use breaks between channel scans so that packets over Wi-Fi have time to be transmitted. This algorithm works fine on ESP-IDF 4.4.5. Some time ago it became necessary to upgrade to version ESP-IDF 5.1.1 (I need the specific Ethernet chip support). On the new version, my scanning algorithm stopped working correctly. On the new version my scanning algorithm stopped working correctly because when the scanning function is called, the event of disconnection from the access point immediately occurs. This behaviour was not present in the version ESP-IDF 4.4.5.

I use ESP32 rev 0 for testing. I have created a test code that reproduces the problem (this code is like an example/wifi/getting_started + scan):
```c
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

#define EXAMPLE_ESP_WIFI_SSID       CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS       CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_ESP_MAXIMUM_RETRY   CONFIG_ESP_MAXIMUM_RETRY

#if CONFIG_ESP_WPA3_SAE_PWE_HUNT_AND_PECK
#define ESP_WIFI_SAE_MODE WPA3_SAE_PWE_HUNT_AND_PECK
#define EXAMPLE_H2E_IDENTIFIER ""
#elif CONFIG_ESP_WPA3_SAE_PWE_HASH_TO_ELEMENT
#define ESP_WIFI_SAE_MODE WPA3_SAE_PWE_HASH_TO_ELEMENT
#define EXAMPLE_H2E_IDENTIFIER CONFIG_ESP_WIFI_PW_ID
#elif CONFIG_ESP_WPA3_SAE_PWE_BOTH
#define ESP_WIFI_SAE_MODE WPA3_SAE_PWE_BOTH
#define EXAMPLE_H2E_IDENTIFIER CONFIG_ESP_WIFI_PW_ID
#endif
#if CONFIG_ESP_WIFI_AUTH_OPEN
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_OPEN
#elif CONFIG_ESP_WIFI_AUTH_WEP
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WEP
#elif CONFIG_ESP_WIFI_AUTH_WPA_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WPA_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA2_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WPA2_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA_WPA2_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WPA_WPA2_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA3_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WPA3_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA2_WPA3_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WPA2_WPA3_PSK
#elif CONFIG_ESP_WIFI_AUTH_WAPI_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WAPI_PSK
#endif

static EventGroupHandle_t s_wifi_event_group;

#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT    BIT1

static const char *TAG = "wifi station";

static int s_retry_num = 0;
static bool connect = true;

static void event_handler(void* arg, esp_event_base_t event_base, int32_t event_id, void* event_data) {
  if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
    esp_wifi_connect();
[... 345 lines cut ...]
I (2071) wifi:state: run -> init (0)
I (2071) wifi:pm stop, total sleep time: 737923 us / 1138729 us

I (2071) wifi:<ba-del>idx:0, tid:0
I (2071) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (2081) wifi station: Event: WIFI_EVENT_STA_DISCONNECTED
I (2251) wifi station: Scan channel 2
I (2751) wifi station: Scan channel 3
I (3251) wifi station: Scan channel 4
I (3751) wifi station: Scan channel 5
I (4251) wifi station: Scan channel 6
I (4751) wifi station: Scan channel 7
I (5251) wifi station: Scan channel 8
I (5751) wifi station: Scan channel 9
I (6251) wifi station: Scan channel 10
I (6751) wifi station: Scan channel 11
I (7251) wifi station: Scan channel 12
I (7751) wifi station: Scan channel 13
I (8151) wifi station: Firmware End

Done
```
As you can see on ESP-IDF 5.1.1 the log line presence `I (2081) wifi station: Event: WIFI_EVENT_STA_DISCONNECTED` before start scanning and on ESP-IDF 4.4.5 is absent. 

If someone can explain why this happens I would be very grateful!

jgujarathi, 2023-10-17:
Hi @VNovytskyi , 
Thank you for reaching out. This issue has been noticed and will be fixed in an upcoming release.

AxelLin, 2023-11-05:
> Hi @VNovytskyi , Thank you for reaching out. This issue has been noticed and will be fixed in an upcoming release.

@jgujarathi How is the status now? Seems recent esp-idf update does not include the fix?

VNovytskyi, 2023-11-11:
@jgujarathi 
There is no new version of the ESP-IDF on 11.11.2023. I am using the ESP-IDF 5.0.4 to avoid the wrong behaviour with the Wi-Fi module. ESP-IDF 5.0.4 and 5.1.1 is very similar and you can use ESP-IDF 5.0.4 while waiting for ESP-IDF 5.1 fix.

AxelLin, 2023-11-16:
> Hi @VNovytskyi , Thank you for reaching out. This issue has been noticed and will be fixed in an upcoming release.

@jgujarathi I found v5.1.2 is already tagged. Is this fixed in v5.1.2?

AxelLin, 2023-11-26:
My question was for @jgujarathi since he said "This issue has been noticed and will be fixed in an upcoming release." in https://github.com/espressif/esp-idf/issues/12317#issuecomment-1766136403 .

@jgujarathi Any update?

AxelLin, 2023-12-10:
> Hi @AxelLin , @VNovytskyi , This issue has been fixed on v5.1.2. Please check.

@jgujarathi Could you share the commit of the fix and close the issue?

jgujarathi, 2024-03-04:
Hi @AxelLin , 
The commit is 3d056fd748909b74e7f301965ddaab94e0431ef8 for the 5.1 IDF version. 
Thanks!

Sherry616, 2024-04-08:
Thanks for reporting, changes have been merged. We will close this issue and feel free to reopen if have more updates.

closed 2024-04-08 by Sherry616