espressif/esp-idf #12318: nimBLE built with `-O2` = array-bounds warnings

KaeLL, 2023-09-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Describe: v5.2-dev-2934-g3b748a6cb7
Relevant nimBLE defconfig
```
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_BT_NIMBLE_MAX_CONNECTIONS=1
CONFIG_BT_NIMBLE_MAX_BONDS=0
CONFIG_BT_NIMBLE_MAX_CCCDS=0
CONFIG_BT_NIMBLE_SECURITY_ENABLE=n
CONFIG_BT_NIMBLE_GATT_MAX_PROCS=0
CONFIG_BT_NIMBLE_HS_FLOW_CTRL=n
```

<details>
<summary>Build log</summary>

```
[815/1139] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_store_util.c.obj
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_store_util.c: In function 'ble_store_util_delete_oldest_peer':
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_store_util.c:243:37: warning: array subscript 0 is outside array bounds of 'ble_addr_t[0]' [-Warray-bounds]
  243 |     rc = ble_store_util_delete_peer(&peer_id_addrs[0]);
      |                                     ^~~~~~~~~~~~~~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_store_util.c:228:16: note: while referencing 'peer_id_addrs'
  228 |     ble_addr_t peer_id_addrs[MYNEWT_VAL(BLE_STORE_MAX_BONDS)];
      |                ^~~~~~~~~~~~~
[863/1139] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_resolv.c.obj
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_resolv.c: In function 'ble_hs_resolv_list_rmv':
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_resolv.c:608:17: warning: array subscript 2 is above array bounds of 'struct ble_hs_resolv_entry[1]' [-Warray-bounds]
  608 |                 &g_ble_hs_resolv_list[position + 1],
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_resolv.c:40:35: note: while referencing 'g_ble_hs_resolv_list'
   40 | static struct ble_hs_resolv_entry g_ble_hs_resolv_list[BLE_RESOLV_LIST_SIZE];
      |                                   ^~~~~~~~~~~~~~~~~~~~
[874/1139] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/store/ram/src/ble_store_ram.c.obj
In file included from /esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/hci_common.h:23,
                 from /esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h:30,
                 from /esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/src/ble_store_ram.c:36:
In function 'ble_addr_cmp',
    inlined from 'ble_store_ram_find_cccd' at /esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/src/ble_store_ram.c:313:17:
/esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/ble.h:303:18: warning: array subscript [0, 306783378] is outside array bounds of 'struct ble_store_value_cccd[0]' [-Warray-bounds]
  303 |     type_diff = a->type - b->type;
      |                 ~^~~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/src/ble_store_ram.c: In function 'ble_store_ram_find_cccd':
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/src/ble_store_ram.c:48:5: note: object 'ble_store_ram_cccds' of size 0
   48 |     ble_store_ram_cccds[MYNEWT_VAL(BLE_STORE_MAX_CCCDS)];
      |     ^~~~~~~~~~~~~~~~~~~
In function 'ble_addr_cmp',
    inlined from 'ble_addr_cmp' at /esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/ble.h:299:19,
    inlined from 'ble_store_ram_find_cccd' at /esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/src/ble_store_ram.c:313:17:
/esp-idf/components/bt/host/nimble/nimble/nimble/include/nimble/ble.h:308:20: warning: array subscript [0, 306783378] is outside array bounds of 'struct ble_store_value_cccd[0]' [-Warray-bounds]
  308 |     return memcmp(a->val, b->val, sizeof(a->val));
      |                   ~^~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/src/ble_store_ram.c: In function 'ble_store_ram_find_cccd':
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/src/ble_store_ram.c:48:5: note: object 'ble_store_ram_cccds' of size 0
   48 |     ble_store_ram_cccds[MYNEWT_VAL(BLE_STORE_MAX_CCCDS)];
      |     ^~~~~~~~~~~~~~~~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/src/ble_store_ram.c:319:21: warning: array subscript [0, 134217727] is outside array bounds of 'struct ble_store_value_cccd[0]' [-Warray-bounds]
  319 |             if (cccd->chr_val_handle != key->chr_val_handle) {
      |                     ^~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/src/ble_store_ram.c:48:5: note: object 'ble_store_ram_cccds' of size 0
   48 |     ble_store_ram_cccds[MYNEWT_VAL(BLE_STORE_MAX_CCCDS)];
      |     ^~~~~~~~~~~~~~~~~~~
In function 'ble_store_ram_read_cccd',
[... 115 lines cut ...]
      |     ~~~~~~~~~~~~~~~~~~~~~~^~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c: In function 'ble_store_config_write':
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:38:5: note: while referencing 'ble_store_config_cccds'
   38 |     ble_store_config_cccds[MYNEWT_VAL(BLE_STORE_MAX_CCCDS)];
      |     ^~~~~~~~~~~~~~~~~~~~~~
In function 'ble_store_config_write_peer_sec',
    inlined from 'ble_store_config_write' at /esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:594:14:
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:302:31: warning: array subscript idx is outside array bounds of 'struct ble_store_value_sec[0]' [-Warray-bounds]
  302 |     ble_store_config_peer_secs[idx] = *value_sec;
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c: In function 'ble_store_config_write':
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:34:5: note: while referencing 'ble_store_config_peer_secs'
   34 |     ble_store_config_peer_secs[MYNEWT_VAL(BLE_STORE_MAX_BONDS)];
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In function 'ble_store_config_write_our_sec',
    inlined from 'ble_store_config_write' at /esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:598:14:
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:171:30: warning: array subscript idx is outside array bounds of 'struct ble_store_value_sec[0]' [-Warray-bounds]
  171 |     ble_store_config_our_secs[idx] = *value_sec;
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c: In function 'ble_store_config_write':
/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:30:5: note: while referencing 'ble_store_config_our_secs'
   30 |     ble_store_config_our_secs[MYNEWT_VAL(BLE_STORE_MAX_BONDS)];
```

</details>

rahult-github, 2023-09-28:
Hello @KaeLL ,

I tried at the same commit and it compiles cleanly. Attached  log with verbose build output for reference.  [compile_o2_log.txt](https://github.com/espressif/esp-idf/files/12745689/compile_o2_log.txt).

Since you observe the issue at your end, can you please help
1. Double confirm you are on clean commit. You can do a git pull/ git submodule update to be sure .
2. Can you please share your complete sdkconfig file .
3. Also please mention how have you enabled the -O2 flag. One way of doing it is via menuconfig by setting the COMPILER_OPTIMIZATION_PERF option. But in case you have done any other way, then please share the same. I can try to same to reproduce the issue and fix it.

AxelLin, 2023-09-28:
Hi @KaeLL 
What is your build target SoC?

KaeLL, 2023-09-28:
@rahult-github 
1.
<details>
<summary>$ git status && echo '' && git describe --tags && echo '' && git submodule status</summary>

```
On branch master
Your branch is behind 'origin/master' by 131 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

nothing to commit, working tree clean

v5.2-dev-2934-g3b748a6cb7

 d037ec89546fad14b5c4d5456c2e23a71e554966 components/bootloader/subproject/components/micro-ecc/micro-ecc (v1.0)
 ffa102a7565dde12510b03d36b616ff9749fe1ad components/bt/controller/lib_esp32 (heads/master-12-gffa102a)
 9da8ad3ebd7932ea6f92578fb1337c7df5000ac7 components/bt/controller/lib_esp32c2/esp32c2-bt-lib (remotes/origin/HEAD)
 e9ad3d704f1034310de8f747d503ea5443df6b67 components/bt/controller/lib_esp32c3_family (remotes/origin/HEAD)
 97499655f3148ee273baadb6b48bbb168207d90b components/bt/controller/lib_esp32c6/esp32c6-bt-lib (heads/main-13-g9749965)
 ff664688f4fedb16c7eafc5bc88a4dacbc3e8ba5 components/bt/controller/lib_esp32h2/esp32h2-bt-lib (remotes/origin/HEAD)
 acdf522a21518a69b3582ff4a4e5b67512a13b32 components/bt/host/nimble/nimble (remotes/origin/nimble-1.5.0-idf)
 eeecc49ce8af123cf8ad40efdb9673e37b56230f components/cmock/CMock (v2.5.2-2-geeecc49)
 ad378ec993cde453b0c93439aa4dbcd317970ccf components/esp_coex/lib (remotes/origin/HEAD)
 95c370089907f74805eac72975b0e85c62e175ed components/esp_phy/lib (remotes/origin/HEAD)
 b74ee0160632978f0e36763f55be0ea6ee18f0c8 components/esp_wifi/lib (heads/master-48-gb74ee016)
 8c9cd0517adf99e363812e9a295dfe3898fdd345 components/heap/tlsf (heads/idf)
 102b03c8095de8a337c293f79dce189be63186f3 components/ieee802154/lib (remotes/origin/HEAD)
 cb8693b058ba302f4829ec6d03f609ac6f848546 components/json/cJSON (v1.7.15-18-gcb8693b)
 7896c6cad020d17a986f7e850f603e084e319328 components/lwip/lwip (remotes/origin/2.1.3-esp)
 727a037da6f9a62a9d3d5def5bd21a2bc9c621c6 components/mbedtls/mbedtls (mbedtls-3.4.1-8-g727a037da)
 05b347643f6679cc60a50a9664b86a24ebf3ae10 components/mqtt/esp-mqtt (ESP-MQTT_FOR_IDF_4.x-223-g05b3476)
 64c77643c92b76fff36bd13211f6b821c4beb8ef components/openthread/lib (remotes/origin/HEAD)
 4ae691476f07164dee3d12d304a9eb0b4f0ea0a3 components/openthread/openthread (thread-reference-20230119-574-g4ae691476)
 abc67a11c6db271bedbb9f58be85d6f4e2ea8389 components/protobuf-c/protobuf-c (v1.4.1)
 0dbb3f71c5f6fae3747a9d935372773762baf852 components/spiffs/spiffs (0.2-255-g0dbb3f7)
 7d2bf62b7e6afaf38153041a9d53c21aeeca9a25 components/unity/unity (v2.4.3-51-g7d2bf62)
```

</details>
2. Is the entire thing really necessary?
<details>
<summary>defconfig</summary>

```
# This file was generated using idf.py save-defconfig. It can be edited manually.
# Espressif IoT Development Framework (ESP-IDF) 5.2.0 Project Minimal Configuration
#
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF=y
CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS=y
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_AFTER_NORESET=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions_ota.csv"
CONFIG_PARTITION_TABLE_OFFSET=0xC000
CONFIG_COMPILER_OPTIMIZATION_PERF=y
CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_SILENT=y
CONFIG_COMPILER_WARN_WRITE_STRINGS=y
CONFIG_BT_ENABLED=y
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_BT_NIMBLE_MAX_CONNECTIONS=1
CONFIG_BT_NIMBLE_MAX_BONDS=0
CONFIG_BT_NIMBLE_MAX_CCCDS=0
CONFIG_BT_NIMBLE_SECURITY_ENABLE=n
CONFIG_BT_NIMBLE_SVC_GAP_DEVICE_NAME="AR"
CONFIG_BT_NIMBLE_GATT_MAX_PROCS=0
[... 48 lines cut ...]
CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y
CONFIG_FREERTOS_CORETIMER_1=y
CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS=y
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU0=y
CONFIG_LWIP_SNTP_MAX_SERVERS=4
CONFIG_MBEDTLS_SSL_VARIABLE_BUFFER_LENGTH=y
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN=y
CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y
CONFIG_MBEDTLS_HKDF_C=y
CONFIG_MQTT_TRANSPORT_WEBSOCKET=n
CONFIG_MQTT_SKIP_PUBLISH_IF_DISCONNECTED=y
CONFIG_MQTT_REPORT_DELETED_MESSAGES=y
CONFIG_NVS_ASSERT_ERROR_CHECK=y
CONFIG_SPI_FLASH_ENABLE_ENCRYPTED_READ_WRITE=n
CONFIG_ULP_COPROC_ENABLED=y
CONFIG_ULP_COPROC_RESERVE_MEM=128
CONFIG_WIFI_PROV_BLE_BONDING=y
CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION=y
```

</details>
3. Through menuconfig, as shown in the defconfig above.

@AxelLin Vanilla ESP32

rahult-github, 2023-09-29:
Hi @KaeLL  , Please help check if attached patch works for you. The patch is to be applied in $IDF_PATH/components/bt/host/nimble/nimble path. 
[fix_compilation_issue.txt](https://github.com/espressif/esp-idf/files/12761741/fix_compilation_issue.txt)

KaeLL, 2023-09-29:
@rahult-github It does. Merge that.
Now, on a lightly related note, the same defconfig results in another warning
```
/esp-idf/components/espcoredump/src/core_dump_common.c:109:14: warning: unused variable 'bytes_free' [-Wunused-variable]
  109 |     uint32_t bytes_free = esp_core_dump_free_stack_space(s_coredump_stack);
      |              ^~~~~~~~~~
```
related to another component. Whether it needs filing a new issue or not, I'll leave it up to you. It's just that given how simple it is to fix it, it was worth noting.

closed 2023-10-25 by espressif-bot