espressif/esp-idf #12319: Building merged binary command issue

krupis, 2023-09-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I am trying to build a one-file binary that contains the bootloader, partition, app and all other binaries in a single file.

I have read the official documentation
https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#merge-binaries-for-flashing-merge-bin
He uses a simple command:
```
esptool.py --chip esp32 merge_bin -o merged-flash.bin "@flash_args"
```

In my case, instead of using esptool.py, I must use full esptool and python paths (I think this has something to do with the way I have installed esp-idf). I have installed it via the vscode esp-idf extension setup.

So I cd in to my project build dir and run the following command with full python and esptool paths:

### What is the actual behavior?

After executing the following command in my project build directory:
```
C:\Users\kapeika.lu\Desktop\WORK\myproject\myproject2\myproject-fw\build> C:\Users\kapeika.lu\.espressif5\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\kapeika.lu\esp5\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 merge_bin -o flash_image.bin @flash_args
```
I am getting an error message that I am missing some arguments (logs attached to **Build or Installation Logs** section).

### Steps to reproduce.

1. cd into build directory
2. Run a command ```esptool.py --chip ESP32 merge_bin -o merged-flash.bin @flash_args``` 

Keep in mind that you may need to use full paths for python and esptool.py

### Build or installation Logs.

```plain
usage: esptool merge_bin [-h] --output OUTPUT [--format {r,a,w}] [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}] [--flash_mode {keep,qio,qout,dio,dout}]
                         [--flash_size {keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}] [--spi-connection SPI_CONNECTION] [--target-offset TARGET_OFFSET]
                         [--fill-flash-size {256KB,512KB,1MB,2MB,4MB,8MB,16MB,32MB,64MB,128MB}]
                         <address> <filename> [<address> <filename> ...]
esptool merge_bin: error: the following arguments are required: <address> <filename>
```

dobairoland, 2023-10-02:
```
C:\Users\kapeika.lu\Desktop\WORK\myproject\myproject2\myproject-fw\build> C:\Users\kapeika.lu\.espressif5\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\kapeika.lu\esp5\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 merge_bin -o flash_image.bin @flash_args
```

It looks like this is the working directory: `C:\Users\kapeika.lu\Desktop\WORK\myproject\myproject2\myproject-fw\build`.

Have you created there a `flash_args` file? I mean `flash_args` and not `flash_args.txt` or with other file extentions.
Can you show its content?

krupis, 2023-10-02:
@dobairoland 
Hello. Yes the flash args file exist in the build directory.
The contents of this file are as following:
```
--flash_mode dio --flash_freq 40m --flash_size detect
0x1000 bootloader/bootloader.bin
0x10000 myproject.bin
0x8000 partition_table/partition-table.bin
0xd000 ota_data_initial.bin
0xf000 phy_init_data.bin
```