espressif/esp-idf #12320: Improve the ESP-IDF sniffer sample to inclue radio tap header , and named pipe usage

emaayan, 2023-09-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

the sniffer example in the esp-idf examples can be improved by adding a radio tap header that can also display various data from the rx_ctrl struct in wireshark 
for example: 
it can be added  this way: 
```
#define IT_PRESENT 0b00000000000001000000100001100000
**typedef struct // https://www.radiotap.org/fields/
{    
    int8_t signal;                         // 5 https://www.radiotap.org/fields/Antenna%20signal.html
    int8_t noise;                          //  6 https://www.radiotap.org/fields/Antenna%20noise.html
    uint8_t antenna;                       // 11 https://www.radiotap.org/fields/Antenna.html
    uint8_t pad_for_channel;
    r_tapdata_channel_t r_tapdata_channel; // 18 https://www.radiotap.org/fields/XChannel
    // EVERY CHANGE IN THIS STRUCUCTURE MUST BE REFLECTED IN IT_PRESET BIT FIELD
} __attribute__((packed)) r_tapdata_t;**

size_t hex_to_byte_array(const char *hexArray, size_t hexLength, uint8_t byteArray[]);

typedef struct pcap_rec
{
    pcap_rec_hdr_t pcap_rec_hdr;
    **ieee80211_radiotap_header_t ieee80211_radiotap_header;**
    r_tapdata_t r_tapdata;
    uint8_t buf[MAX_LENGTH];
} __attribute__((packed)) pcap_rec_t;
```

and then the method that actually writs the data can do this
i don't enough about either the radio tap header files and and the rx_ctrl strcut to do a full mapping, 
but this is the gist of thing.s 

```
pcap_rec_t capture_create_pcap_record(wifi_promiscuous_pkt_t *pkt)
{
    wifi_pkt_rx_ctrl_t ctrl = pkt->rx_ctrl;    
    uint8_t *payload = pkt->payload;

    uint32_t sig_packetLength = ctrl.sig_len;
    uint32_t pack_len = sig_packetLength + ieee80211_radiotap_header.it_len;
    pcap_rec_hdr_t pcap_rec_hdr = {.ts_sec =ctrl.timestamp / 1000000U, .ts_usec = ctrl.timestamp % 1000000U, .incl_len = pack_len > pcap_hdr.snaplen ? pcap_hdr.snaplen : pack_len, .orig_len = pack_len};
    pcap_rec_t pcap_rec = {.pcap_rec_hdr = pcap_rec_hdr, .ieee80211_radiotap_header = ieee80211_radiotap_header, .buf = {}};
    memcpy(pcap_rec.buf, payload, pcap_rec_hdr.incl_len);

    pcap_rec.r_tapdata.noise = ctrl.noise_floor;
    pcap_rec.r_tapdata.signal = ctrl.rssi;
    pcap_rec.r_tapdata.antenna = ctrl.ant;
    pcap_rec.r_tapdata.r_tapdata_channel.channel = ctrl.channel;
    return pcap_rec;
}
```

the result would be something like this:
the radiotap header v0, is the actuall data being written
while the Radio information are fields wireshark generates based on the previous section. 

[image]

additioanlly wireshark can accept connect directly to tcp sockets and read the same packet type
so all esp32 needs to do is setup a tcp server that will start sending data as soon as it gets a socket 
wireshark can be configure using TCP@<ip> with the port being 19000 as default so doing wireshark -k -i tcp@<ip> would get wireshark to directly start reading capture packets from esp

zhangyanjiaoesp, 2024-04-01:
@emaayan Thanks for your suggestion, we will consider to add it, but the priority is not high.

archef2000, 2024-06-19:
What is the pcap_rec_t type and how are you populating the flags field?

emaayan, 2024-06-27:
> What is the pcap_rec_t type and how are you populating the flags field?
That's my representation of how wireshark record would look (based on pcap format) alongside the wifi headers 
The flags portion is the tricky bit and i was hoping for some help from espressif because it involves taking data from the incoming wifi traffic and adapting it to wireshark and i don't know enough about esp to fill in all the details
You can check out my project :
https://github.com/emaayan/WifiSniffer/blob/main/Esp32Sniffer/components/wifi_sniffer/include/capture_lib.h