espressif/esp-idf #12322: subprocess.CalledProcessError: Command '['/home/franz/.espressif/python_env/idf4.2_py3.11_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/franz/esp-idf-v4.2.4/requirements.txt']' returned non-zero exit status 1.

franz-ms-muc, 2023-09-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v4.2.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected to run ./install.sh all smooth

### What is the actual behavior?

Instead it brings Error see below: 

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain
franz@franz-Nitro-AN517-55:~/esp-idf-v4.2.4$ ./install.sh all
Detecting the Python interpreter
Checking "python" ...
Python 3.11.4
"python" has been detected
Installing ESP-IDF tools
Selected targets are: esp32h2, esp32s2, esp32c2, esp32c3, esp32, esp32s3
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20211220 (already installed)
Installing Python environment and packages
Python 3.11.4
pip 23.0.1 from /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages/pip (python 3.11)
Installing Python packages from /home/franz/esp-idf-v4.2.4/requirements.txt
Ignoring cffi: markers 'python_version < "3.6"' don't match your environment
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 4)) (66.1.1)
Requirement already satisfied: click>=5.0 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 8)) (8.1.7)
Requirement already satisfied: pyserial>=3.0 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 9)) (3.5)
Requirement already satisfied: future>=0.15.2 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 10)) (0.18.3)
Requirement already satisfied: cryptography<35,>=2.1.4 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 11)) (3.4.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 14)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 15)) (0.30)
Requirement already satisfied: idf-component-manager~=1.0 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 16)) (1.4.0)
Collecting gdbgui==0.13.2.0
  Using cached gdbgui-0.13.2.0-py3-none-any.whl (878 kB)
Requirement already satisfied: pygdbmi<=0.9.0.2 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 21)) (0.9.0.2)
[... 339 lines cut ...]
        File "/tmp/pip-build-env-ujrr9fti/overlay/lib/python3.11/site-packages/Cython/Build/Dependencies.py", line 1301, in cythonize_one
          raise CompileError(None, pyx_file)
      Cython.Compiler.Errors.CompileError: src/gevent/libev/corecext.pyx
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Traceback (most recent call last):
  File "/home/franz/esp-idf-v4.2.4/tools/idf_tools.py", line 1693, in <module>
    main(sys.argv[1:])
  File "/home/franz/esp-idf-v4.2.4/tools/idf_tools.py", line 1689, in main
    action_func(args)
  File "/home/franz/esp-idf-v4.2.4/tools/idf_tools.py", line 1412, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr, env=env_copy)
  File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/franz/.espressif/python_env/idf4.2_py3.11_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/franz/esp-idf-v4.2.4/requirements.txt']' returned non-zero exit status 1.
franz@franz-Nitro-AN517-55:~/esp-idf-v4.2.4$
```

franz-ms-muc, 2023-09-28:
try to do a Setup in Eclipse: 

```
Installing tools...
This can take a while. Please be patient.
Executing /usr/bin/python3 /home/franz/esp-idf-v4.2.4/tools/idf_tools.py install all
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.16.4 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20211220 (already installed)
Skipping ninja@1.10.0 (already installed)
OK

Executing /usr/bin/python3 /home/franz/esp-idf-v4.2.4/tools/idf_tools.py install-python-env
Python 3.11.4
pip 23.0.1 from /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages/pip (python 3.11)
Installing Python packages from /home/franz/esp-idf-v4.2.4/requirements.txt
Ignoring cffi: markers 'python_version < "3.6"' don't match your environment
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 4)) (66.1.1)
Requirement already satisfied: click>=5.0 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 8)) (8.1.7)
Requirement already satisfied: pyserial>=3.0 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 9)) (3.5)
Requirement already satisfied: future>=0.15.2 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 10)) (0.18.3)
Requirement already satisfied: cryptography<35,>=2.1.4 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 11)) (3.4.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 14)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 15)) (0.30)
Requirement already satisfied: idf-component-manager~=1.0 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 16)) (1.4.0)
Collecting gdbgui==0.13.2.0
  Using cached gdbgui-0.13.2.0-py3-none-any.whl (878 kB)
Requirement already satisfied: pygdbmi<=0.9.0.2 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 21)) (0.9.0.2)
Collecting python-socketio<5
  Using cached python_socketio-4.6.1-py2.py3-none-any.whl (51 kB)
Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 27)) (1.5.4)
Requirement already satisfied: bitstring>=3.1.6 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 28)) (3.1.9)
Requirement already satisfied: ecdsa>=0.16.0 in /home/franz/.espressif/python_env/idf4.2_py3.11_env/lib/python3.11/site-packages (from -r /home/franz/esp-idf-v4.2.4/requirements.txt (line 29)) (0.18.0)
Collecting Flask<1.0,>=0.12.2
  Using cached Flask-0.12.5-py2.py3-none-any.whl (81 kB)
Collecting Flask-Compress<2.0,>=1.4.0
  Using cached Flask_Compress-1.14-py3-none-any.whl (8.4 kB)
Collecting Flask-SocketIO<3.0,>=2.9
  Using cached Flask_SocketIO-2.9.6-py2.py3-none-any.whl (16 kB)
Collecting gevent<2.0,>=1.2.2
  Using cached gevent-1.5.0.tar.gz (5.3 MB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'error'
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [322 lines of output]
      Compiling src/gevent/resolver/cares.pyx because it changed.
      [1/1] Cythonizing src/gevent/resolver/cares.pyx
      warning: src/gevent/resolver/cares.pyx:32:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
      warning: src/gevent/resolver/cares.pyx:33:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
      Compiling src/gevent/libev/corecext.pyx because it changed.
      [1/1] Cythonizing src/gevent/libev/corecext.pyx
      warning: src/gevent/libev/corecext.pyx:325:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See ht [... cut ...]
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Executing /home/franz/.espressif/python_env/idf4.2_py3.11_env/bin/python -m pip list
Copying OpenOCD Rules
Copying File: /home/franz/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/contrib/60-openocd.rules to destination: /etc/udev/rules.d/60-openocd.rules
Unable to copy rules for OpenOCD to system directory, try running the eclipse with sudo command
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.
```

franz-ms-muc, 2023-09-29:
not important anymore.
could install 4.2.5 and do my Work there. 

see https://github.com/espressif/esp-idf/issues/12321

franz-ms-muc, 2023-09-29:
@espressif-bot please put this issue to a very low priority.

dobairoland, 2023-10-02:
The workaround has been added to 4.2.5: https://github.com/espressif/esp-idf/commit/9e9cbd541b1a46524b3ab250ba82bb544091fd08

closed 2023-10-02 by dobairoland