espressif/esp-idf #12325: Countdown GPTimer problem

KaeLL, 2023-09-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Describe: v5.2-dev-2934-g3b748a6cb7

The following code works, and it's an attempt at testing one-shot functionality of the GPTimer.
```C
#include "driver/gpio.h"
#include "driver/gptimer.h"
#include "esp_attr.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "soc/gpio_struct.h"

gptimer_handle_t gpt = NULL;

uint32_t ledp[] = { 0, 1, 4, 5, 16, 17, 20, 21 };
uint8_t idx = 0, idx1 = 7;

bool IRAM_ATTR gpt_cb( gptimer_handle_t timer, const gptimer_alarm_event_data_t *edata, void *user_ctx )
{
	assert( !gptimer_stop( gpt ) );
	GPIO.out_w1tc = ( uint32_t )ledp[ idx1 ];
	GPIO.out_w1ts = ( uint32_t )ledp[ idx ];

	idx1 = idx;
	idx = ( idx + 1 ) % ( sizeof ledp / sizeof *ledp );

	return false;
}

void gpt_run( uint16_t msec )
{
	ESP_ERROR_CHECK( gptimer_register_event_callbacks( gpt, &( const gptimer_event_callbacks_t ) { gpt_cb }, NULL ) );
	ESP_ERROR_CHECK( gptimer_enable( gpt ) );
	ESP_ERROR_CHECK( gptimer_set_alarm_action( gpt, &( const gptimer_alarm_config_t ) { 40000 * msec } ) );
	ESP_ERROR_CHECK( gptimer_start( gpt ) );
}

void gpt_init( void )
{
	ESP_ERROR_CHECK( gptimer_new_timer(
		&( const gptimer_config_t ) {
			.clk_src = GPTIMER_CLK_SRC_DEFAULT,
			.direction = GPTIMER_COUNT_UP,
			.intr_priority = 3,
			.resolution_hz = 40000000 },
		&gpt ) );
}

void app_main( void )
{
	ESP_ERROR_CHECK( gpio_config(
		&( const gpio_config_t ) { .mode = GPIO_MODE_INPUT_OUTPUT, .pin_bit_mask = BIT64( 0 ) | BIT64( 2 ) | BIT64( 4 ) } ) );
	gpt_init();
	gpt_run( 1000 );
	vTaskDelay( pdMS_TO_TICKS( 1100 ) );
	ESP_ERROR_CHECK( gptimer_disable( gpt ) );
	gpt_run( 2000 );
	vTaskDelay( pdMS_TO_TICKS( 2100 ) );
	ESP_ERROR_CHECK( gptimer_disable( gpt ) );
	gpt_run( 3000 );
	vTaskDelay( pdMS_TO_TICKS( 3100 ) );
	ESP_ERROR_CHECK( gptimer_disable( gpt ) );
	gpt_run( 4000 );
	vTaskDelay( pdMS_TO_TICKS( 4100 ) );
	ESP_ERROR_CHECK( gptimer_disable( gpt ) );
}
```
When attempting to switch to the countdown timer, the assert inside the ISR triggers saying the `timer is not running`. Naturally, I assumed the driver must've already stopped the timer, so I just commented out the assert. That results in the first `gptimer_disable` failing saying the `timer not in enable state`, which is weird, since it seems the timer is neither in run state nor in enable state. So next step is test if the driver is put in init state after triggering the ISR, which doesn't make much sense, but alas, I ain't come this far. So I comment out the `gptimer_disable`'s, and proceed to test it out, and the result is `gptimer_enable` failing saying `timer not in init state`.
What gives? Either there's an issue with the countdown functionality and this is an actual issue ticket, or I can't make sense of the new APIs enough to use it, in which case this is a request for improvement of the `timer_group/gptimer` example by adding and example of how to use the countdown timer.

KaeLL, 2023-09-29:
While we're at it, what's the ideal way of unconditionally starting/restarting the timer?
```C
gptimer_set_raw_count( gpt, 40000 * msec );
gptimer_start( gpt );
```
From what I understand, the way above should handle both running and a stopped timer, but it'll always throw the `timer is not enabled yet` error in the case of the former, and it would always throw a `timer is not running` if I called `gptimer_stop` before `gptimer_start`.

KaeLL, 2023-09-29:
Just realized perhaps I should've 
set raw count to 40000 * msec and alarm to 0. If so, the question in the second comment still remains.

suda-morris, 2023-10-07:
@KaeLL Thanks for raising this question. It was also discussed in our team before.

> gptimer_start() should handle both a running and a stopped timer

Starting a timer that was already started by other threads/tasks/ISR was considered an "invalid state". This sounds a little bit strict. But it can help expose some potential issues in the user's application code.

You can also bypass the `ESP_ERR_INVALID_STATE` error returned by `gptimer_start` in the application code.

KaeLL, 2023-10-09:
@suda-morris Yes, but that doesn't solve the problem of the error message showing up without it being an actual error, and bypassing esp_err types can lead to issues if the same error is raised by more than one location within the function, either now or eventually.

closed 2025-02-26 by espressif-bot