espressif/esp-idf #12326: about Partition Table

franz-ms-muc, 2023-09-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

.

KaeLL, 2023-09-29:
> The “Offset” field for each partition is empty. The gen_esp32part.py tool fills in each blank offset, starting after the partition table and making sure each partition is aligned correctly.

adokitkat, 2024-11-05:
To expand upon previous answer, the preference is up to you, but I think it is easier to just leave the offset blank as it is automatically calculated - e.g. if you set your partition size to 1M and your program exceeds his value, you only need to make the partition bigger and you don't have to meddle and manually also update all other offsets in the csv. If you for some reason need to have a partition on certain offset, then you can use it.

closed 2024-11-05 by adokitkat