espressif/esp-idf #12327: C3 partitions, bootloader and everything else

Snifi0, 2023-09-29:
Hey, sorry for asking, where could I get the bootloader partitions and whatever else there is to make it work with c, I accidentally uploaded esp32 bootloader and all that. Or like a 0x0? Not sure how to say it but I'd hope you understand.

KonstantinKondrashov, 2023-10-03:
Hi @Snifi0!
Sorry, I am not quite sure that I got the question.
After the `idf.py build` command you should be able to see the log like this:
```
python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m \
0x1000 build/bootloader/bootloader.bin \
0x8000 build/partition_table/partition-table.bin \
0x10000 build/hello_world.bin
```
It shows at what offset in flash memory a certain binary file will be written. (note that for C3 chip bootloader.bin file goes to 0x0 address, it is defined by BOOTLOADER_OFFSET_IN_FLASH).

If you need APIs to read something from a partition you can use any from this https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/partition.html. Also, you may find interesting an interaction with already written partitions using [parttool.py](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#command-line-interface)

Does this answer your question?

Snifi0, 2023-10-07:
> Hi @Snifi0! Sorry, I am not quite sure that I got the question. After the `idf.py build` command you should be able to see the log like this:
>
> ```
> python -m esptool --chip esp32 -b 460800 --before [...]

thanks! yes it does.

closed 2023-10-07 by Snifi0