espressif/esp-idf #12328: Issue to download multiple files with esp_http_client : esp-tls: [sock=59] select() timeout HTTP_CLIENT: Connection failed, sock < 0

0x0fe, 2023-09-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Espressif SoC revision.

ESP32-D0WDR2-V3-V3 (revision v3.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32D0WDR2

### Power Supply used.

USB

### What is the expected behavior?

The expected behavior is to be able to download multiples files one after the other.

### What is the actual behavior?

The actual behavior is connection error after the first file. 
This test opens a task which will attempt to downlaod a file and write it to the SD_MMC via  ring buffer with a second task, a semaphore ensures synchronisation.  Once one file is downlaoded, it will start the next file etc. Here for simplicity it download the same file multiple times, in th real application it would be different files obviously. 

### Steps to reproduce.

This test application is used in PIO, hence the setup/loop function, however, the example comes from IDF and can be run in IDF .

```cpp
#include "esp_system.h"
#include "esp_event.h"
#include "esp_check.h"
#include "esp_crt_bundle.h"

#include "FS.h"
#include "SD_MMC.h"
#include <WiFi.h>

#define SD_D0                   2
#define SD_CLK                  14
#define SD_CMD                  15

static const char *TAG = "THROUGHPUT";

typedef struct {
    FILE *f_out;
    size_t buffer_size;
    RingbufHandle_t rb;
    SemaphoreHandle_t start;
    SemaphoreHandle_t done;
    size_t bytes_downloaded;
[... 271 lines cut ...]
E (56368) THROUGHPUT: HTTP_EVENT_ERROR
HTTP request failed: ESP_ERR_HTTP_CONNECT
Min. free heap: 2052043 bytes
E (61364) THROUGHPUT: Content length is 0
E (61380) esp-tls: [sock=59] select() timeout
E (61380) TRANSPORT_BASE: Failed to open a new connection: 32774
E (61380) HTTP_CLIENT: Connection failed, sock < 0
E (61383) THROUGHPUT: HTTP_EVENT_ERROR
HTTP request failed: ESP_ERR_HTTP_CONNECT
Min. free heap: 2051983 bytes
E (66379) THROUGHPUT: Content length is 0
E (66395) esp-tls: [sock=59] select() timeout
E (66395) TRANSPORT_BASE: Failed to open a new connection: 32774
E (66395) HTTP_CLIENT: Connection failed, sock < 0
E (66398) THROUGHPUT: HTTP_EVENT_ERROR
HTTP request failed: ESP_ERR_HTTP_CONNECT
Min. free heap: 2051879 bytes
E (71394) THROUGHPUT: Content length is 0
E (71410) esp-tls: [sock=59] select() timeout
E (71410) TRANSPORT_BASE: Failed to open a new connection: 32774
E (71410) HTTP_CLIENT: Connection failed, sock < 0
E (71413) THROUGHPUT: HTTP_EVENT_ERROR
HTTP request failed: ESP_ERR_HTTP_CONNECT
Min. free heap: 2051803 bytes
```

0x0fe, 2023-10-02:
So, i wrote a simpelr implementation, without ringbuffer and semaphore, and i still get the same error.
I also suspect the 1024KB/s counted on the the first round is not true, probably that the connection times out or something (5 second default timeout, 5MB : 1024KB/s). 

```
static void (*s_on_write_data)(const uint8_t *buffer, int size) = NULL;
static const uint32_t MAX_BLOCK_SIZE = 4096;//1536;
FILE* file;
esp_http_client_handle_t client=NULL;

static esp_err_t _http_event_handle(esp_http_client_event_t *evt){

  switch (evt->event_id)
  {
    case HTTP_EVENT_ERROR:
    ESP_LOGI(TAG, "HTTP_EVENT_ERROR");
    break;
    case HTTP_EVENT_ON_CONNECTED:
    //ESP_LOGI(TAG, "HTTP_EVENT_ON_CONNECTED");
    break;
    case HTTP_EVENT_HEADER_SENT:
    //ESP_LOGI(TAG, "HTTP_EVENT_HEADER_SENT");
    break;
    case HTTP_EVENT_ON_HEADER:
    //ESP_LOGI(TAG, "HTTP_EVENT_ON_HEADER");
    //printf("%.*s\n", evt->data_len, (char*)evt->data);
    break;
    case HTTP_EVENT_ON_DATA:
    //ESP_LOGI(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
    if (!esp_http_client_is_chunked_response(evt->client))
    {
      if (esp_http_client_get_status_code(evt->client) == 200)
      {
        if (s_on_write_data)
        {
          if (evt->data_len < MAX_BLOCK_SIZE){
            ((uint8_t *)evt->data)[evt->data_len] = 0;
          }
          s_on_write_data((const uint8_t *)evt->data, evt->data_len);
        }
      }
      //printf("%.*s\n", evt->data_len, (char*)evt->data);
    }
    break;
    case HTTP_EVENT_ON_FINISH:
    printf("HTTP_EVENT_ON_FINISH\n");
    break;
    case HTTP_EVENT_DISCONNECTED:
    printf("HTTP_EVENT_DISCONNECTED\n");
    break;
  }
  return ESP_OK;
}
void on_write_data(const uint8_t *buffer, int size){

 write(fileno(file), buffer, size);
}
bool http_get_file_perform(const char *link, void (*cb)(const uint8_t *buffer, int size)){

 s_on_write_data = cb;
 String fullpath = String("/sdcard/test.mp3");
 file = fopen(fullpath.c_str(), "w+");
 if (file == NULL)
 {
    printf("Error opening file\n");
    return 1;
 }

 auto start_ = millis();
 if(client==NULL){
  esp_http_client_config_t config = {
    .url = link,
    .event_handler = _http_event_handle,
    .buffer_size = MAX_BLOCK_SIZE,
    .skip_cert_common_name_check=1,
    .keep_alive_enable=0,      
  };
  client = esp_http_client_init(&config);
  esp_http_client_set_method(client, HTTP_METHOD_GET);
 } 

 esp_err_t err = esp_http_client_perform(client);
 if (err == ESP_OK)
 {
  printf("Status = %d, content_length = %d\n",
         esp_http_client_get_status_code(client),
         esp_http_client_get_content_length(client));
  const size_t TOTAL_SIZE = esp_http_client_get_content_length(client);
  auto end_ = millis();
  size_t time_ = (end_ - start_) / 1000;
  String speed_ = String((TOTAL_SIZE / time_) / 1024);
  printf("Speed %u KB/s\n", (TOTAL_SIZE / time_) / 1024);
  printf("Min. free heap: %" PRIu32 " bytes\n", esp_get_minimum_free_heap_size());
 }
 else
 {
  ESP_LOGE(TAG, "Failed to perform HTTP GET request from %s err %d", link, esp_http_client_get_status_code(client));
  delay(500);
  esp_http_client_close(client);
  esp_http_client_cleanup(client);  
  client=NULL;
 }
 fclose(file);
 //esp_http_client_close(client);
 //esp_http_client_cleanup(client);
 return err == ESP_OK;
}
```

I also dont understand why i get two DISCONNECTED events.

```
ready
..
Wifi connected
HTTP_EVENT_ON_FINISH
Status = 200, content_length = 5242880
Speed 1024 KB/s
Min. free heap: 2161455 bytes
E (15125) HTTP_CLIENT: Failed to perform HTTP GET request from http://cachefly.cachefly.net/5mb.test err -1
HTTP_EVENT_DISCONNECTED
HTTP_EVENT_DISCONNECTED
E (21632) esp-tls: [sock=59] select() timeout
E (21632) TRANSPORT_BASE: Failed to open a new connection: 32774
E (21632) HTTP_CLIENT: Connection failed, sock < 0
E (21635) HTTP_CLIENT: Failed to perform HTTP GET request from http://cachefly.cachefly.net/5mb.test err 0
HTTP_EVENT_DISCONNECTED
HTTP_EVENT_DISCONNECTED
E (28148) esp-tls: [sock=59] select() timeout
E (28148) TRANSPORT_BASE: Failed to open a new connection: 32774
E (28148) HTTP_CLIENT: Connection failed, sock < 0
E (28151) HTTP_CLIENT: Failed to perform HTTP GET request from http://cachefly.cachefly.net/5mb.test err 0
HTTP_EVENT_DISCONNECTED
HTTP_EVENT_DISCONNECTED
E (34664) esp-tls: [sock=59] select() timeout
E (34664) TRANSPORT_BASE: Failed to open a new connection: 32774
E (34664) HTTP_CLIENT: Connection failed, sock < 0
E (34667) HTTP_CLIENT: Failed to perform HTTP GET request from http://cachefly.cachefly.net/5mb.test err 0
HTTP_EVENT_DISCONNECTED
HTTP_EVENT_DISCONNECTED
E (41180) esp-tls: [sock=59] select() timeout
E (41180) TRANSPORT_BASE: Failed to open a new connection: 32774
E (41180) HTTP_CLIENT: Connection failed, sock < 0
E (41183) HTTP_CLIENT: Failed to perform HTTP GET request from http://cachefly.cachefly.net/5mb.test err 0
HTTP_EVENT_DISCONNECTED
HTTP_EVENT_DISCONNECTED
```

0x0fe, 2023-10-03:
so, after further investigation and tests, i think i found the settings in cause :

my settings : 

```
CONFIG_LWIP_MAX_ACTIVE_TCP=32
CONFIG_LWIP_MAX_LISTENING_TCP=16
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65534
CONFIG_LWIP_TCP_WND_DEFAULT=65534
CONFIG_LWIP_TCP_RECVMBOX_SIZE=64

```

and with these settings, i can get it to work normally, but throughput  has dropped of course

```
CONFIG_LWIP_MAX_ACTIVE_TCP=16
CONFIG_LWIP_MAX_LISTENING_TCP=16
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=5744
CONFIG_LWIP_TCP_WND_DEFAULT=5744
CONFIG_LWIP_TCP_RECVMBOX_SIZE=32
```

```
Min. free heap: 2206563 bytes
HTTP_EVENT_ON_FINISH
Status = 200, content_length = 5242880
Speed 72 KB/s
HTTP_EVENT_DISCONNECTED
client_cleanup 0
Min. free heap: 2206223 bytes
HTTP_EVENT_ON_FINISH
Status = 200, content_length = 5242880
Speed 65 KB/s
HTTP_EVENT_DISCONNECTED
client_cleanup 0
Min. free heap: 2206223 bytes
HTTP_EVENT_ON_FINISH
Status = 200, content_length = 5242880
Speed 68 KB/s
HTTP_EVENT_DISCONNECTED
client_cleanup 0
Min. free heap: 2206223 bytes
HTTP_EVENT_ON_FINISH
Status = 200, content_length = 5242880
Speed 65 KB/s
HTTP_EVENT_DISCONNECTED
client_cleanup 0
Min. free heap: 2206223 bytes
HTTP_EVENT_ON_FINISH
Status = 200, content_length = 5242880
Speed 74 KB/s
HTTP_EVENT_DISCONNECTED
client_cleanup 0
Min. free heap: 2206139 bytes
HTTP_EVENT_ON_FINISH
Status = 200, content_length = 5242880
Speed 73 KB/s
HTTP_EVENT_DISCONNECTED
client_cleanup 0
Min. free heap: 2206139 bytes
HTTP_EVENT_ON_FINISH
Status = 200, content_length = 5242880
Speed 76 KB/s
HTTP_EVENT_DISCONNECTED
client_cleanup 0
Min. free heap: 2206111 bytes
```

hmalpani, 2023-10-23:
Hello @0x0fe 

I tried the code you shared above with some modifications to reproduce this issue with the configs you used. I am still unable to reproduce this issue.

This is my modified code:
```
bool http_get_file_perform()
{
    s_on_write_data = NULL;
    char local_response_buffer[1048] = {0};
    // auto start_ = millis();
    if (client == NULL) {
        esp_http_client_config_t config = {
            .url = "https://192.168.20.194:8000/temp.txt",
            .event_handler = _http_event_handle,
            .buffer_size = MAX_BLOCK_SIZE,
            .skip_cert_common_name_check=1,
            .cert_pem = howsmyssl_com_root_cert_pem_start,
            .keep_alive_enable=0,
            .user_data = local_response_buffer,
        };
        client = esp_http_client_init(&config);
        esp_http_client_set_method(client, HTTP_METHOD_GET);
    } 

    esp_err_t err = esp_http_client_perform(client);
    if (err == ESP_OK) {
        printf("Status = %d, content_length = %d\n",
                esp_http_client_get_status_code(client),
                esp_http_client_get_content_length(client));
        const size_t TOTAL_SIZE = esp_http_client_get_content_length(client);
        // auto end_ = millis();
        // size_t time_ = (end_ - start_) / 1000;
        // String speed_ = String((TOTAL_SIZE / time_) / 1024);
        // printf("Speed %u KB/s\n", (TOTAL_SIZE / time_) / 1024);
        printf("Min. free heap: %" PRIu32 " bytes\n", esp_get_minimum_free_heap_size());
    } else {
        ESP_LOGE(TAG, "Failed to perform HTTP GET request, err %d", esp_http_client_get_errno(client));
        // delay(500);
        esp_http_client_close(client);
        esp_http_client_cleanup(client);  
        client=NULL;
    }
    return err == ESP_OK;
}

Can you share a minimal application with sdkconfig which can help reproduce this issue?

Thanks.
```

0x0fe, 2023-10-23:
Hello, can you elaborate ? are you able to run the code with sucessfull download and decent download speed? (1M or better).
Or is there an error? 
Regarding this issue i have pinpointed the main parameter to be

CONFIG_LWIP_TCP_SND_BUF_DEFAULT=32768
CONFIG_LWIP_TCP_WND_DEFAULT=32768

IF set too high, it will download sucessfully once and then crash with essentially TCPIP stack error.
IF set lower it will work stable with reduced datarate, currently i am set at the middle and i get consistent rate around 300KB/s (byte, not bit) 
Which is not too bad. 
I attach my SDKConfig.
 
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/13079394/sdkconfig.zip)

hmalpani, 2023-11-20:
Hello @0x0fe 
Apologies for the delayed reply.

Yes I was able to run the code with successful download. Have you checked with the code snippet I shared above at your side. I didn't get any errors when I tested with that code. Can you please try it out and let me know the results.