espressif/esp-idf #12329: error: comparison between two arrays is deprecated in C++20 [-Werror=array-compare]

franz-ms-muc, 2023-09-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.1.1

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

```
TEST_CASE("string", "[string]")
{
	char q[] = "GeeksforGeeks";
	setString("CF-A", "string", q);
	char r[] = "Hello";
    TEST_ASSERT_NOT_EQUAL(q, r);
    ESP_LOGI(TAG, "before: %s %s", q, r);
    getString("CF-A", "string", "unknown", r, 33U);
    ESP_LOGI(TAG, "after: %s %s", q, r);
    TEST_ASSERT_EQUAL_STRING(q, r);
}
```

### What is the actual behavior?

v

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
In file included from /home/franz/esp-idf-v5.1.1/components/unity/unity/src/unity.h:16,
                 from /home/franz/git/zas/IHMKratosCAN/Application/components/Settings/test/test_settings.cpp:13:
/home/franz/git/zas/IHMKratosCAN/Application/components/Settings/test/test_settings.cpp: In function 'void test_func_48()':
/home/franz/esp-idf-v5.1.1/components/unity/unity/src/unity.h:134:130: error: comparison between two arrays is deprecated in C++20 [-Werror=array-compare]
  134 | #define TEST_ASSERT_NOT_EQUAL(expected, actual)                                                    UNITY_TEST_ASSERT(((expected) !=  (actual)), __LINE__, " Expected Not-Equal")
      |                                                                                                                       ~~~~~~~~~~~^~~~~~~~~~~~
/home/franz/esp-idf-v5.1.1/components/unity/unity/src/unity_internals.h:710:102: note: in definition of macro 'UNITY_TEST_ASSERT'
  710 | #define UNITY_TEST_ASSERT(condition, line, message)                                              if (condition) {} else {UNITY_TEST_FAIL((UNITY_LINE_TYPE)(line), (message));}
      |                                                                                                      ^~~~~~~~~
/home/franz/git/zas/IHMKratosCAN/Application/components/Settings/test/test_settings.cpp:53:5: note: in expansion of macro 'TEST_ASSERT_NOT_EQUAL'
   53 |     TEST_ASSERT_NOT_EQUAL(q, r);
      |     ^~~~~~~~~~~~~~~~~~~~~
/home/franz/esp-idf-v5.1.1/components/unity/unity/src/unity.h:134:130: note: use unary '+' which decays operands to pointers or '&'view_convert_expr' not supported by dump_decl<declaration error>[0] != &'view_convert_expr' not supported by dump_decl<declaration error>[0]' to compare the addresses
  134 | #define TEST_ASSERT_NOT_EQUAL(expected, actual)                                                    UNITY_TEST_ASSERT(((expected) !=  (actual)), __LINE__, " Expected Not-Equal")
      |                                                                                                                       ~~~~~~~~~~~^~~~~~~~~~~~
/home/franz/esp-idf-v5.1.1/components/unity/unity/src/unity_internals.h:710:102: note: in definition of macro 'UNITY_TEST_ASSERT'
  710 | #define UNITY_TEST_ASSERT(condition, line, message)                                              if (condition) {} else {UNITY_TEST_FAIL((UNITY_LINE_TYPE)(line), (message));}
      |                                                                                                      ^~~~~~~~~
/home/franz/git/zas/IHMKratosCAN/Application/components/Settings/test/test_settings.cpp:53:5: note: in expansion of macro 'TEST_ASSERT_NOT_EQUAL'
   53 |     TEST_ASSERT_NOT_EQUAL(q, r);
      |     ^~~~~~~~~~~~~~~~~~~~~
cc1plus: some warnings being treated as errors
ninja: build stopped: subcommand failed.
```

zikalino, 2023-10-02:
@franz-ms-muc 

Macro

```c++
TEST_ASSERT_NOT_EQUAL(q, r);
```
expands to generic **!=** comparison:

```c++
UNITY_TEST_ASSERT(((expected) !=  (actual)), __LINE__, " Expected Not-Equal")
```
And this is why the compiler is comiling.

You can override this and compare two string pointers by casting char[] to char*:

```c++
TEST_CASE("string", "[string]")
{
    char q[] = "GeeksforGeeks";
    setString("CF-A", "string", q);
    char r[] = "Hello";
    TEST_ASSERT_NOT_EQUAL((char*)q, (char*)r);
    ESP_LOGI(TAG, "before: %s %s", q, r);
    getString("CF-A", "string", "unknown", r, 33U);
    ESP_LOGI(TAG, "after: %s %s", q, r);
    TEST_ASSERT_EQUAL_STRING(q, r);
}
```

KonstantinKondrashov, 2023-10-11:
If you have a look at the description of [TEST_ASSERT_NOT_EQUAL()](https://github.com/ThrowTheSwitch/Unity/blob/7d2bf62b7e6afaf38153041a9d53c21aeeca9a25/src/unity.h#L127-L134) here you can see that it compares integers only. You can not give arrays into it, this is why you got this build error.
Using TEST_ASSERT_EQUAL_STRING is correct here.

Closing it for now, feel free to reopen it if there are any questions. Thanks.

closed 2023-10-11 by KonstantinKondrashov