espressif/esp-idf #12333: 当使用clang为esp32构建app时，bootloader大小超过0x7000（默认配置）

PickaxeHit, 2023-09-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1,also in master

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

能正常构建完成，不会出现超出大小限制的情况

### What is the actual behavior?

使用clang时：
```
root@localhost:~/main# idf.py build
Executing action: all (aliases: build)
Running cmake in directory /root/main/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/root/.espressif/python_env/idf5.1_py3.11_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /root/main"...
-- IDF_TARGET is not set, guessed 'esp32' from sdkconfig '/root/main/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.39.2")
-- The C compiler identification is Clang 15.0.0
-- The CXX compiler identification is Clang 15.0.0
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /root/.espressif/tools/esp-clang/15.0.0-23786128ae/esp-clang/bin/clang
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /root/.espressif/tools/esp-clang/15.0.0-23786128ae/esp-clang/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /root/.espressif/tools/esp-clang/15.0.0-23786128ae/esp-clang/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /root/main/sdkconfig
Compiler supported targets:   Registered Targets:
    aarch64    - AArch64 (little endian)
    aarch64_32 - AArch64 (little endian ILP32)
    aarch64_be - AArch64 (big endian)
    arm64      - ARM64 (little endian)
    arm64_32   - ARM64 (little endian ILP32)
    riscv32    - 32-bit RISC-V
    riscv64    - 64-bit RISC-V
    xtensa     - Xtensa 32

CMake Warning at /root/esp/esp-idf/components/esp_common/project_include.cmake:14 (message):
  Building ESP-IDF with clang is an experimental feature and is not yet
  officially supported.
Call Stack (most recent call first):
  /root/esp/esp-idf/tools/cmake/build.cmake:398 (include)
  /root/esp/esp-idf/tools/cmake/build.cmake:620 (__build_process_project_includes)
[... 215 lines cut ...]
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /root/main/build/bootloader/bootloader.bin
[104/104] cd /root/main/build/bootloade...ot/main/build/bootloader/bootloader.binBootloader binary size 0x6810 bytes. 0x7f0 bytes (7%) free.
[872/886] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.objFAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
/root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.1.1\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DUNITY_INCLUDE_CONFIG_H -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/root/main/build/config -I/root/main/main -I/root/esp/esp-idf/components/newlib/platform_include -I/root/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/root/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/root/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/root/esp/esp-idf/components/freertos/esp_additions/include -I/root/esp/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/root/esp/esp-idf/components/esp_hw_support/include -I/root/esp/esp-idf/components/esp_hw_support/include/soc -I/root/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/root/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/root/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/root/esp/esp-idf/components/heap/include -I/root/esp/esp-idf/components/log/include -I/root/esp/esp-idf/components/soc/include -I/root/esp/esp-idf/components/soc/esp32 -I/root/esp/esp-idf/components/soc/esp32/include -I/root/esp/esp-idf/components/hal/esp32/include -I/root/esp/esp-idf/components/hal/include -I/root/esp/esp-idf/components/hal/platform_port/include -I/root/esp/esp-idf/components/esp_rom/include -I/root/esp/esp-idf/components/esp_rom/include/esp32 -I/root/esp/esp-idf/components/esp_rom/esp32 -I/root/esp/esp-idf/components/esp_common/include -I/root/esp/esp-idf/components/esp_system/include -I/root/esp/esp-idf/components/esp_system/port/soc -I/root/esp/esp-idf/components/esp_system/port/include/private -I/root/esp/esp-idf/components/xtensa/include -I/root/esp/esp-idf/components/xtensa/esp32/include -I/root/esp/esp-idf/components/lwip/include -I/root/esp/esp-idf/components/lwip/include/apps -I/root/esp/esp-idf/components/lwip/include/apps/sntp -I/root/esp/esp-idf/components/lwip/lwip/src/include -I/root/esp/esp-idf/components/lwip/port/include -I/root/esp/esp-idf/components/lwip/port/freertos/include -I/root/esp/esp-idf/components/lwip/port/esp32xx/include -I/root/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/root/esp/esp-idf/components/esp_ringbuf/include -I/root/esp/esp-idf/components/efuse/include -I/root/esp/esp-idf/components/efuse/esp32/include -I/root/esp/esp-idf/components/driver/include -I/root/esp/esp-idf/components/driver/deprecated -I/root/esp/esp-idf/components/driver/analog_comparator/include -I/root/esp/esp-idf/components/driver/dac/include -I/root/esp/esp-idf/components/driver/gpio/include -I/root/esp/esp-idf/components/driver/gptimer/include -I/root/esp/esp-idf/components/driver/i2c/include -I/root/esp/esp-idf/components/driver/i2s/include -I/root/esp/esp-idf/components/driver/ledc/include -I/root/esp/esp-idf/components/driver/mcpwm/include -I/root/esp/esp-idf/components/driver/parlio/include -I/root/esp/esp-idf/components/driver/pcnt/include -I/root/esp/esp-idf/components/driver/rmt/include -I/root/esp/esp-idf/components/driver/sdio_slave/include -I/root/esp/esp-idf/components/driver/sdmmc/include -I/root/esp/esp-idf/components/driver/sigma_delta/include -I/root/esp/esp-idf/components/driver/spi/include -I/root/esp/esp-idf/components/driver/temperature_sensor/include -I/root/esp/esp-idf/components/driver/touch_sensor/include -I/root/esp/esp-idf/components/driver/twai/include -I/root/esp/esp-idf/components/driver/uart/include -I/root/esp/esp-idf/components/driver/usb_serial_jtag/include -I/root/esp/esp-idf/components/driver/touch_sensor/esp32/include -I/root/esp/esp-idf/components/esp_pm/include -I/root/esp/esp-idf/components/mbedtls/port/include -I/root/esp/esp-idf/components/mbedtls/mbedtls/include -I/root/esp/esp-idf/components/mbedtls/mbedtls/library -I/root/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/root/esp/esp-idf/components/esp_app_format/include -I/root/esp/esp-idf/components/bootloader_support/include -I/root/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/root/esp/esp-idf/components/esp_partition/include -I/root/esp/esp-idf/components/app_update/include -I/root/esp/esp-idf/components/esp_mm/include -I/root/esp/esp-idf/components/spi_flash/include -I/root/esp/esp-idf/components/pthread/include -I/root/esp/esp-idf/components/esp_timer/include -I/root/esp/esp-idf/components/app_trace/include -I/root/esp/esp-idf/components/esp_event/include -I/root/esp/esp-idf/components/nvs_flash/include -I/root/esp/esp-idf/components/esp_phy/include -I/root/esp/esp-idf/components/esp_phy/esp32/include -I/root/esp/esp-idf/components/vfs/include -I/root/esp/esp-idf/components/esp_netif/include -I/root/esp/esp-idf/components/wpa_supplicant/include -I/root/esp/esp-idf/components/wpa_supplicant/port/include -I/root/esp/esp-idf/components/wpa_supplicant/esp_supplicant/include -I/root/esp/esp-idf/components/esp_coex/include -I/root/esp/esp-idf/components/esp_wifi/include -I/root/esp/esp-idf/components/esp_wifi/wifi_apps/include -I/root/esp/esp-idf/components/unity/include -I/root/esp/esp-id [... cut ...]
/root/main/main/main.c:6:10: fatal error: uniconv.h: No such file or directory
    6 | #include "uniconv.h"
      |          ^~~~~~~~~~~
compilation terminated.
[877/886] Building C object esp-idf/wif...wifi_provisioning.dir/src/manager.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /root/main/build/log/idf_py_stderr_output_6325 and /root/main/build/log/idf_py_stdout_output_6325
```

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### More Information.

在master和v5.1.1都测试过了

PickaxeHit, 2023-10-15:
Hello? Is anyone here?

igrr, 2023-10-15:
> Error: Bootloader binary size 0x7290 bytes is too large for partition table offset 0x8000. Bootloader binary can be maximum 0x7000 (28672) bytes unless the partition table offset is increased in the Partition Table section of the project configuration menu.

Have you tried increasing the partition table offset in menuconfig, as the error message suggests?

Clang-based toolchain generates larger binaries, so yes, with the default settings the bootloader ends up being too large for the default partition table offset. You have to either reduce the bootloader size (for example, by reducing the bootloader log level) or move the partition table to a larger offset.

PickaxeHit, 2023-10-15:
Set to 0x9000 worked. But if the target is esp32s3, there will be no error.
Will this issue be solved in the future?

closed 2023-10-21 by PickaxeHit