espressif/esp-idf #12334: A2DP not working with Car Stereo

divyansh, 2023-10-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Latest version installed with fresh Arduino installation

### Espressif SoC revision.

VID: 0x1c04 PID 0xEA60

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP wroom 32

### Power Supply used.

USB

### What is the expected behavior?

As the A2DP example ESP32_A2DP_example_sender_8bit works perfectly fine with BT speakers which are not having any pin.
But Car stereo's which we call as Android Headup Display have a default pin "0000".
I am unable to connect the ESP32_A2DP_example_sender_8bit example project to the Android Head up Display.
  My setup function and isValid funciton is as mentioned below rest is unchanged.

`bool isValid(const char* ssid, esp_bd_addr_t address, int rssi){
   Serial.print("available SSID: ");
   Serial.println(ssid);
   return true;
}

void setup() {
  Serial.begin(115200);

  a2dp_source.set_ssid_callback(isValid);
  a2dp_source.set_pin_code("0000",ESP_BT_PIN_TYPE_VARIABLE);
  a2dp_source.set_ssp_enabled(1);
  a2dp_source.set_auto_reconnect(false);
  a2dp_source.start(get_data_frames);  

  a2dp_source.set_volume(30);`

### What is the actual behavior?

Here I am explaining just what happens when we connect the Android Headup Display ( AHUD )with our Android Phone.
When the search for new bluetooth device is started, and the device (AHUD) is clicked upon a pop comes for entering password.
I am guessing this problem is coming.
In ESP32, I am unable to change the program to work with AHUD as it throws a pin pop on Android device whereas in ESP32 i am unaware of this function.

I have enable SSP in the code and set the pin but still it deosn;t connect.

### Steps to reproduce.

1.  I used A2DP example provided in Arduino Iibrary bt_music_sender_ssid_callback
2. Then I added following code to the setup function.
 `  a2dp_source.set_pin_code("0000",ESP_BT_PIN_TYPE_VARIABLE);
  a2dp_source.set_ssp_enabled(1);`
4. The code works on Bluetooth speakers and Android Auto as these devices doesn't have any pin.
5. But Car Audio systems which we call as Android Headup display , , generally have default password "0000"
6. The ESP-32 is not able to connect to the Android Head Display .
...

### Debug Logs.

```plain
[ 11144][D][BluetoothA2DPSource.cpp:458] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 0
[ 11144][I][BluetoothA2DPSource.cpp:385] filter_inquiry_scan_result(): [BT_AV] Scanned device: 00:58:76:c7:18:07
[ 11151][I][BluetoothA2DPSource.cpp:391] filter_inquiry_scan_result(): [BT_AV] --Class of Device: 0x240404
[ 11161][I][BluetoothA2DPSource.cpp:395] filter_inquiry_scan_result(): [BT_AV] --RSSI: -95
[ 11169][I][BluetoothA2DPSource.cpp:411] filter_inquiry_scan_result(): [BT_AV] --Compatiblity: Compatible
[ 11178][I][BluetoothA2DPSource.cpp:413] filter_inquiry_scan_result(): [BT_AV] --Name: CarKit_blink
[ 11187][I][BluetoothA2DPSource.cpp:433] filter_inquiry_scan_result(): [BT_AV] --Result: Target device found
[ 11196][D][BluetoothA2DPCommon.cpp:177] set_last_connection(): [BT_AV] set_last_connection: 00:58:76:c7:18:07
[ 11206][D][BluetoothA2DPCommon.cpp:181] set_last_connection(): [BT_AV] no change!
[ 11213][I][BluetoothA2DPSource.cpp:437] filter_inquiry_scan_result(): [BT_AV] Cancel device discovery ...
[ 11224][D][BluetoothA2DPSource.cpp:458] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 1
[ 11231][I][BluetoothA2DPSource.cpp:468] bt_app_gap_callback(): [BT_AV] Device discovery stopped.
[ 11240][I][BluetoothA2DPSource.cpp:469] bt_app_gap_callback(): [BT_AV] a2dp connecting to peer: CarKit_blink
[ 11249][W][BluetoothA2DPCommon.cpp:41] connect_to(): [BT_AV] connect_to to 00:58:76:c7:18:07
[ 11258][I][BluetoothA2DPCommon.cpp:277] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[ 11370][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[ 11371][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x0
[ 11379][I][BluetoothA2DPSource.cpp:631] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state 4, evt 0x0
[ 11388][D][BluetoothA2DPSource.cpp:599] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
[ 11397][D][BluetoothA2DPSource.cpp:607] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Connecting
[ 11411][D][BluetoothA2DPSource.cpp:696] bt_app_av_state_connecting(): [BT_AV] bt_app_av_state_connecting evt 0
[ 16497][D][BluetoothA2DPSource.cpp:458] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 16
[ 16498][I][BluetoothA2DPSource.cpp:526] bt_app_gap_callback(): [BT_AV] event: 16
[ 16502][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[ 16513][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x0
[ 16522][I][BluetoothA2DPSource.cpp:631] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state 4, evt 0x0
[ 16531][D][BluetoothA2DPSource.cpp:599] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
```

### More Information.

1.  I used A2DP example provided in Arduino Iibrary bt_music_sender_ssid_callback
2. Then I added following code to the setup function.
 `  a2dp_source.set_pin_code("0000",ESP_BT_PIN_TYPE_VARIABLE);
  a2dp_source.set_ssp_enabled(1);`
4. The code works on Bluetooth speakers and Android Auto as these devices doesn't have any pin.
5. But Car Audio systems which we call as Android Headup display , , generally have default password "0000"
6. The ESP-32 is not able to connect to the Android Head Display .

divyansh, 2023-10-01:
Hi,
I searched online that ESP-IDF on arduino may not work as expected hence I installed ESP-32 IDF latest version on ubuntu.
Following error is coming when i am trying to connect the device using ESP-32 IDF using the default program of A2DP_source

`I (663) main_task: Calling app_main()
I (683) BTDM_INIT: BT controller compile version [153f120]
I (683) BTDM_INIT: Bluetooth MAC: cc:db:a7:14:c7:d6
I (683) phy_init: phy_version 4771,450c73b,Aug 16 2023,11:03:10
I (1363) BT_AV: Starting device discovery...
W (1363) BT_BTC: btc_avrc_tg_set_rn_supported_evt failed: AVRC TG not yet initialized

W (1373) BT_BTC: A2DP Enable with AVRC
W (1373) BT_AVRC: For AVRCP Target Cateory 1 and 3, SDP record shall contain additional protocol list
I (1393) BT_AV: bt_app_av_sm_hdlr state: 1, event: 0x4
I (1393) BT_AV: event: 10
I (1403) BT_AV: event: 10
I (1403) BT_AV: event: 10
I (1403) main_task: Returned from app_main()
I (1413) BT_AV: Discovery started.
I (5613) BT_AV: Scanned device: 00:58:76:c7:18:07
I (5613) BT_AV: --Class of Device: 0x240404
I (5613) BT_AV: --RSSI: -75
I (6023) BT_AV: Scanned device: 00:58:76:c7:18:07
I (6023) BT_AV: --Class of Device: 0x240404
I (6023) BT_AV: --RSSI: -76
I (6223) BT_AV: Scanned device: 00:58:76:c7:18:07
I (6223) BT_AV: --Class of Device: 0x240404
I (6223) BT_AV: --RSSI: -77
`

BetterJincheng, 2023-10-08:
@divyansh Thank you for your feedback. These warnings do exist for certain reasons, but theoretically they will not affect the establishment of the connection. Is it still not possible to connect to your device Using ESP-IDF A2DP_SOURCE?

divyansh, 2023-10-12:
Hi,
Yes I am finding difficulty in connecting the device.

BetterJincheng, 2023-11-03:
@divyansh Sorry for late reply. Please attach the whole logs on ESP32 side. Thanks!

closed 2025-03-05 by Alvin1Zhang