espressif/esp-idf #12335: OpenThread - Example for simple end device

baylf2000, 2023-10-01:
### Is your feature request related to a problem?

There are no examples currently for thread end devices, like a light or switch.

### Describe the solution you'd like.

It would be very helpful to have some simple examples that might be used for Thread lights, switches, sensors etc.

chshu, 2023-10-24:
Thread is a network layer protocol, just like Wi-Fi. An application is required to build a Thread products (light, switch, ...).

Matter is the currently the most popular application runs over Thread, please check the light example in esp-matter here: https://github.com/espressif/esp-matter/tree/main/examples/light

If you want to implement some other applicaitons, you can also use the TCP/UDP APIs directly, refer to the udp example: https://github.com/espressif/esp-thread-br/blob/main/components/esp_ot_cli_extension/src/esp_ot_udp_socket.c

chshu, 2023-11-10:
@baylf2000 Please reopen if any follow up questions.

closed 2023-11-10 by chshu