espressif/esp-idf #12336: USB HOST bInterval value check too strict

ddosoff, 2023-10-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3065-g272b4091f1

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

T-Display Amoled

### Power Supply used.

USB

### What is the expected behavior?

I'm using ESP32S3 as USB host controller to connect CDC ACM device based on the STM32 chip.

I was playing with cdc_acm_host example to check if it can open my device. And it works with my linux CDC ACM driver without any issues.

### What is the actual behavior?

After enabling more debug info in the cdc_acm_host example I found that esp says:

**bInterval value (255) of Interrupt pipe exceeds max supported limit.**

After googling about [bInterval](https://microchip.my.site.com/s/article/What-is-bInterval) I found actually no problem if it's more than 32, because it should be truncated to 32.
 
Also I found it's really truncated in esp-idf code later. So I disabled this error check and my CDC ACM device was opened succesfully.

### Steps to reproduce.

1. Run cdc_acm_host example
2. Connect CDC ACM device with high bInterval
3. See error.

### Debug Logs.

```plain
bInterval value (255) of Interrupt pipe exceeds max supported limit.
```

gabryelreyes, 2023-10-05:
I have successfully reproduced this error and came to the same conclusion. 
The interval can go up to 255, but the FRAME_LIST_LEN shall then be capped at 32.

Sadly I don't have the knowledge to propose an adequate alternative/fix to this issue.

roma-jam, 2023-10-06:
Hi everyone, 

@ddosoff Thank you so much for reporting that! 

The FRAME_LIST_LEN is a hardware restriction. If the packet interval were too high, this could lead to missing packets.

Otherwise, it's true that for most cases, the bInterval value presented in a descriptor can be 255, but the device can still work without a problem with that hardware restriction as the polling interval is still 32 frames for FS devices. 

I’ll make some verification and return back ASAP.

esp32beans, 2023-11-13:
The following boards fail due to notification interrupt endpoint bInterval > 32.

* Arduino Uno R3
* Arduino Leonardo
* Arduino Mega2560 R3

tore-espressif, 2023-11-20:
Hi all, thanks for the reports. The fix was merged internally, it should show up in GitHub in a matter of few days

closed 2023-11-27 by espressif-bot