espressif/esp-idf #12342: LCD display corruption caused by Wi-Fi activities

GaryChan-work, 2023-10-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

We found that the LCD display screen may be corrupted (see link below) if Wi-Fi is connected and has some activities (e.g. communicating with host, performing OTA update, etc). If Wi-Fi is turned off, the display works fine. Besides, the LCD display buffer is placed in the PSRAM (Octal mode@80MHz). If we temporarily change it to internal RAM, it seems the problem can be avoided.

[https://drive.google.com/file/d/1J46mps4F4HuG6tB0qJbvmQMnisdy4Jf3/view?usp=sharing](url)

Here is the IDF information:
IDF version: v4.4.4 (patched with commit "3cec3a00260")
LVGL version: v7.11
Display Buffer: PSRAM (Octal Mode@80MHz)
ESP LCD I/F: i80 bus, 8-bit bus width, 16-bit RGB, 10MHz pixel CLK

Can anyone help to see how to fix this issue or know the root cause?
Thanks for your help!

Gary

suda-morris, 2023-10-07:
It's due to a hardware limitation in the MEM_SPI bandwidth. We recommend you to enable these default Kconfig options: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/i80_controller/sdkconfig.defaults.esp32s3

Also can you see the problem on the master branch?

You may also get some hints in this FAQ (although different LCD model, but same root cause): https://docs.espressif.com/projects/espressif-esp-faq/en/latest/software-framework/peripherals/lcd.html#does-esp32-s3-require-an-external-psram-to-use-the-rgb-screen

puddletowntom, 2024-02-14:
Hi, 

I have a similar issue. I made changes to the config as you mention here but it didnt work.  Did you make any progress with this? Is there a way of switching PSRAM off for Wi-Fi while having PSRAM on for the LCD. 

The similar issue i raised can be found here,

https://github.com/espressif/esp-idf/issues/13169

fabiengagne, 2026-09-16:
I have noted a similar issue with the display, unrelated to WiFi.

I am documenting a display problem and the workaround we validated on a Waveshare ESP32-S3-Touch-LCD-5 Rev1.2. The failure looked at first like an RGB timing, power, touch, or ribbon-cable issue, but it was strongly correlated with application traffic to octal PSRAM.

### Environment

- Waveshare ESP32-S3-Touch-LCD-5, PCB Rev1.2
- ESP32-S3 revision 0.2, 16 MiB flash, 8 MiB octal PSRAM at 80 MHz
- 800 x 480 RGB565 panel
- ESP-IDF 6.0.2
- LVGL 9.5.0
- Espressif `esp_lvgl_adapter` 0.5.3
- 16 MHz LCD pixel clock
- one full RGB565 framebuffer in PSRAM
- internal DMA bounce buffer covering 20 display lines

### Symptoms

After a cold boot, the splash screen was initially stable. Near the transition to the first interactive screen, the image started moving or became vertically shifted. Corruption could persist indefinitely, but sometimes stopped after a touch event or after an LCD sleep/wake cycle. A touch in a scrollable region could also trigger corruption in a diagnostic build.

The behavior occurred on USB, a battery supply, and a 3S LiPo supply. It did not disappear when we temporarily disabled external UART initialization, the shared I2C devices, or the screen-sleep feature. The original demonstration firmware also had intermittent startup behavior, although it was not an exact reproducer of our application failure.

The fact that touching the screen sometimes repaired it was especially misleading. Touch was not the cause; it changed task scheduling or display activity enough to alter the outcome.

### Reproducer

We added a hardware diagnostic that allocates 1 MiB in PSRAM, writes a known 32-bit pattern, and verifies it while the RGB display continues scanning. The operator recorded whether the display remained correct after each run.

| PSRAM access policy | Correct | Glitched | Approximate throughput |
| --- | ---: | ---: | ---: |
| Continuous | 7/10 | 3/10 | 15.1–16.3 MiB/s |
| 16 KiB, no pause | 10/10 | 0/10 | 12.5–12.8 MiB/s |
| 8 KiB, no pause | 7/10 | 3/10 | 12.0–12.5 MiB/s |
| 4 KiB, no pause | 8/10 | 2/10 | 11.9–12.6 MiB/s |
| 2 KiB, no pause | 9/10 | 1/10 | 11.6–12.4 MiB/s |
| 4 KiB plus `taskYIELD()` | 5/10 | 5/10 | 11.6–12.3 MiB/s |
| 2 KiB plus `taskYIELD()` | 1/10 | 9/10 | 11.1–12.1 MiB/s |
| 8 KiB plus 1 ms delay | 25/25 | 0/25 | 5.6–5.9 MiB/s |
| 4 KiB plus 1 ms delay | 25/25 | 0/25 | 3.5–3.6 MiB/s |
| 3 KiB plus 1 ms delay | 25/25 | 0/25 | 2.7–2.8 MiB/s |
| 2 KiB plus 1 ms delay | 25/25 | 0/25 | 1.88–1.91 MiB/s |
| 1 KiB plus 1 ms delay | 25/25 | 0/25 | 0.97–0.98 MiB/s |

The non-monotonic results without a blocking delay indicate a scheduling-sensitive failure. `taskYIELD()` was not sufficient because it did not guarantee that the display-feeding task got useful execution time.

### Fix used in our application

We made two complementary changes.

First, the LVGL draw buffer was moved from PSRAM to internal SRAM and limited to 10 lines. The full RGB framebuffer remains in PSRAM:

```c
esp_lv_adapter_display_config_t display_config =
    ESP_LV_ADAPTER_DISPLAY_RGB_DEFAULT_CONFIG(panel, NULL, 800, 480, rotation);
display_config.profile.buffer_height = 10;
display_config.profile.use_psram = false;
display_config.tear_avoid_mode = ESP_LV_ADAPTER_TEAR_AVOID_MODE_NONE;
```

Second, application-controlled operations that scan, copy, encode, decode, or transform large PSRAM regions use a shared policy:

- serialize participating bulk jobs with a mutex;
- process 4 KiB per chunk;
- call `vTaskDelay(pdMS_TO_TICKS(1))` between chunks.

We selected 4 KiB as a conservative midpoint below the largest chunk that passed 25 consecutive trials. This retained about 3.5 MiB/s in the combined write-and-verify test.

We also reduced the RGB driver from three framebuffers to one. Automatic LCD restart at VSYNC is disabled in the final configuration because it can mask a failure but does not reserve enough PSRAM bandwidth to prevent it.

After moving the LVGL draw buffer to internal SRAM and pacing explicit bulk PSRAM work, repeated cold boots, navigation, scrolling, and diagnostic runs no longer reproduced the corruption.

### Interpretation

Our working explanation is PSRAM bandwidth or arbitration starvation: the RGB driver cannot refill its internal DMA bounce buffer before the panel needs the next pixels. This is supported by the controlled PSRAM-load experiment, but we did not probe the buses with a logic analyzer.

The result is from one Rev1.2 board and visual pass/fail observations. It should not be treated as a universal maximum throughput for every ESP32-S3 RGB board. Pixel clock, framebuffer count, bounce-buffer size, PSRAM mode, task placement, and the particular memory operation will all affect the limit.

I would be interested to know whether others have observed the same vertical shift on this board, and whether another ESP-IDF RGB configuration provides a more direct way to reserve PSRAM bandwidth for the LCD path.

Let me know if this was useful.

### Supporting Data
[  [waveshare-psram-bandwidth.md](https://github.com/user-attachments/files/32308615/waveshare-psram-bandwidth.md) ](url)