espressif/esp-idf #12343: Mesh Internal communication : 2 nodes are becoming root node  in self organizing network.

amitabhskrishworks, 2023-10-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v 5.0

### Espressif SoC revision.

ESP 32 (chip revision 3.1)

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP 32 One

### Power Supply used.

USB

### What is the expected behavior?

One node becoming root node in esp wifi mesh.

### What is the actual behavior?

2 nodes are becoming root node .

### Steps to reproduce.

1. Flash 2 devices with mesh internal communication.
2. Powered both device at same time .
3. They both started searching for network and then both of them become root node.

### Debug Logs.

```plain
Device 1 :
I (838) wifi_init: tcp mss: 1440
I (838) wifi_init: WiFi IRAM OP enabled
I (838) wifi_init: WiFi RX IRAM OP enabled
I (848) phy_init: phy_version 4771,450c73b,Aug 16 2023,11:03:10
I (958) wifi:mode : sta (08:3a:8d:80:44:4c) + softAP (08:3a:8d:80:44:4d)
I (958) wifi:enable tsf
I (968) wifi:Total power save buffer number: 8
I (968) wifi:Init max length of beacon: 752/752
I (968) wifi:Init max length of beacon: 752/752
I (978) mesh: <nvs>read layer:1
I (978) mesh: <nvs>read assoc:1
I (978) wifi:Set ps type: 1

I (3158) wifi:mode : sta (08:3a:8d:80:44:4c)
I (3168) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x1, need_scan_router:0x0, look_for_nwk_count:1
I (3168) mesh_main: <MESH_EVENT_MESH_STARTED>ID:11:12:13:14:15:16
I (3168) mesh_main: mesh starts successfully, heap:154300, root not fixed<0>(tree), ps:1

[... 355 lines cut ...]
I (80938) wifi:I (80928) mesh: [TXQ]<max:128>up(0, be:0), down(0, be:0), mgmt:0, xon(req:0, rsp:0), bcast:0, wnd(0, parent:00:00:00:00:00:00)
<ba-add>idx:0 (ifx:0, c0:74:ad:cd:6a:49), tid:0, ssn:0, winSize:64
I (80948) mesh: [RXQ]<max:128 = cfg:128 + extra:0>self:0, <max:128 = cfg:128 + extra:0>tods:0 
I (80938) mesh_main: <MESH_EVENT_ROOT_ADDRESS>root address:08:b6:1f:ff:b1:81
I (80928) mesh_main: The Total Nodes in Network are 1
I (80978) mesh_main: This node is Root Node
I (80978) mesh_main: This is routing table : 08:b6:1f:ff:b1:80 and the current layer is 1
I (80988) mesh_main: layer:1, rtableSize:1, ROOT
I (81308) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (82928) esp_netif_handlers: sta ip: 172.16.0.54, mask: 255.255.255.0, gw: 172.16.0.1
I (82928) mesh_main: <IP_EVENT_STA_GOT_IP>IP:172.16.0.54
I (85988) mesh_main: The Total Nodes in Network are 1
I (85988) mesh_main: This node is Root Node
I (85988) mesh_main: This is routing table : 08:b6:1f:ff:b1:80 and the current layer is 1     
I (85998) mesh_main: layer:1, rtableSize:1, ROOT
I (90998) mesh_main: The Total Nodes in Network are 1
I (90998) mesh_main: This node is Root Node
I (90998) mesh_main: This is routing table : 08:b6:1f:ff:b1:80 and the current layer is 1     
I (90998) mesh_main: layer:1, rtableSize:1, ROOT
I (96008) mesh_main: The Total Nodes in Network are 1
I (96008) mesh_main: This node is Root Node
I (96008) mesh_main: This is routing table : 08:b6:1f:ff:b1:80 and the current layer is 1     
I (96008) mesh_main: layer:1, rtableSize:1, ROOT
I (101018) mesh_main: The Total Nodes in Network are 1
```

zhangyanjiaoesp, 2023-10-17:
@amitabhskrishworks 
Have you called the API `esp_mesh_allow_root_conflicts(true)` ?

amitabhskrishworks, 2023-10-17:
No i have not used this option .

amitabhskrishworks, 2023-10-17:
> @amitabhskrishworks Have you called the API `esp_mesh_allow_root_conflicts(true)` ?

No

zhangyanjiaoesp, 2023-10-17:
when the device2 becomes root, the device1 still be root? The log shows that there is 1 node in the network.

> I (101018) mesh_main: The Total Nodes in Network are 1

amitabhskrishworks, 2023-10-20:
Yes 
The device 1 shows that it is root node and device 2 shows that it is root node

amitabhskrishworks, 2023-10-20:
> Collaborator

2 mesh network get formed with same Mesh ID

zhangyanjiaoesp, 2023-10-23:
@amitabhskrishworks Did you test just with the `internal_communication` without any changes? If yes, I will test it locally to reproduce this issue.

amitabhskrishworks, 2023-10-25:
Yes i did not made any changes except adding some logs

zhangyanjiaoesp, 2023-10-26:
@amitabhskrishworks  I have tested locally, and I didn't reproduce your issue. And I noticed that your rssi from the router is -83/-85, can you get your devices closer to the router and test it again?

amitabhskrishworks, 2023-11-02:
I tested it and got the same result but this time with lesser frequency.

amitabhskrishworks, 2023-11-06:
This is my full code which makes 2 root nodes when 2 nodes are switched ON simultaneously
I have node used the code relaed to mesh_light

#include <string.h>
#include <inttypes.h>
#include "esp_wifi.h"
#include "esp_mac.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_mesh.h"
#include "nvs_flash.h"
#include <stdio.h>
#include <stdlib.h>

#define RX_SIZE          (1500)
#define TX_SIZE          (1460)

static const char *MESH_TAG = "mesh_main";
static const uint8_t MESH_ID[6] = { 0x11, 0x12, 0x13, 0x14, 0x15, 0x16};
static uint8_t tx_buf[TX_SIZE] = { 0, };
static uint8_t rx_buf[RX_SIZE] = { 0, };
static bool is_running = true;
static bool is_mesh_connected = false;
static mesh_addr_t mesh_parent_addr;
static int mesh_layer = -1;
static esp_netif_t *netif_sta = NULL;

void esp_mesh_p2p_tx_main(void *arg)
{
    ESP_LOGI(MESH_TAG, " esp_mesh_p2p_tx_main tasks is about to start ");
    int i = 0, route_table_size = 0, j = 0;
    //int send_count = 0;
    char *str="UPSTREAM DATA BY Shubham Kanaujiya";
    bool is_running = true;
    esp_err_t err;
    mesh_addr_t route_table[CONFIG_MESH_ROUTE_TABLE_SIZE];
    mesh_addr_t route_tables[CONFIG_MESH_ROUTE_TABLE_SIZE];
    mesh_data_t data;
    data.data = (uint8_t *) str;
    data.data = tx_buf;
    data.size = sizeof(tx_buf);
    data.proto = MESH_PROTO_BIN;
    data.tos = MESH_TOS_P2P;
    
    while (is_running)
    {
        ESP_LOGI(MESH_TAG, "The Total Nodes in Network are %d",esp_mesh_get_total_node_num());
        esp_mesh_get_routing_table((mesh_addr_t *)&route_tables, CONFIG_MESH_ROUTE_TABLE_SIZE * 6, &route_table_size);
        if (esp_mesh_is_root())
        {
            ESP_LOGI(MESH_TAG, "This node is Root Node");
            for (i = 0; i < route_table_size; i++)
            {
                ESP_LOGI(MESH_TAG, "This is routing table : " MACSTR " and the current layer is %d", MAC2STR(route_tables[i].addr),
                esp_mesh_get_layer());
            }
        }
        else
        {
            ESP_LOGI(MESH_TAG, "This node is Non-Root node ");
            for (i = 0; i < route_table_size; i++)
            {
                ESP_LOGI(MESH_TAG, "This is routing table : " MACSTR " and the current layer is %d", MAC2STR(route_tables[i].addr),
                esp_mesh_get_layer());
            }
        }
        
        if (esp_mesh_is_root())
        {
            ESP_LOGI(MESH_TAG, "layer:%d, rtableSize:%d, %s", mesh_layer, esp_mesh_get_routing_table_size(),
[... 305 lines cut ...]
    cfg.channel = CONFIG_MESH_CHANNEL;
    cfg.router.ssid_len = strlen(CONFIG_MESH_ROUTER_SSID);
    memcpy((uint8_t *) &cfg.router.ssid, CONFIG_MESH_ROUTER_SSID, cfg.router.ssid_len);
    memcpy((uint8_t *) &cfg.router.password, CONFIG_MESH_ROUTER_PASSWD,
           strlen(CONFIG_MESH_ROUTER_PASSWD));
    /* mesh softAP */
    ESP_ERROR_CHECK(esp_mesh_set_ap_authmode(CONFIG_MESH_AP_AUTHMODE));
    cfg.mesh_ap.max_connection = CONFIG_MESH_AP_CONNECTIONS;
    cfg.mesh_ap.nonmesh_max_connection = CONFIG_MESH_NON_MESH_AP_CONNECTIONS;
    memcpy((uint8_t *) &cfg.mesh_ap.password, CONFIG_MESH_AP_PASSWD,
           strlen(CONFIG_MESH_AP_PASSWD));
    ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
    /* mesh start */
    ESP_ERROR_CHECK(esp_mesh_start());
#ifdef CONFIG_MESH_ENABLE_PS
    /* set the device active duty cycle. (default:10, MESH_PS_DEVICE_DUTY_REQUEST) */
    ESP_ERROR_CHECK(esp_mesh_set_active_duty_cycle(CONFIG_MESH_PS_DEV_DUTY, CONFIG_MESH_PS_DEV_DUTY_TYPE));
    /* set the network active duty cycle. (default:10, -1, MESH_PS_NETWORK_DUTY_APPLIED_ENTIRE) */
    ESP_ERROR_CHECK(esp_mesh_set_network_duty_cycle(CONFIG_MESH_PS_NWK_DUTY, CONFIG_MESH_PS_NWK_DUTY_DURATION, CONFIG_MESH_PS_NWK_DUTY_RULE));
#endif
    ESP_LOGI(MESH_TAG, "mesh starts successfully, heap:%" PRId32 ", %s<%d>%s, ps:%d\n",  esp_get_minimum_free_heap_size(),
             esp_mesh_is_root_fixed() ? "root fixed" : "root not fixed",
             esp_mesh_get_topology(), esp_mesh_get_topology() ? "(chain)":"(tree)", esp_mesh_is_ps_enabled());

}

zhangyanjiaoesp, 2023-11-06:
I just run the above code can reproduce the problem, right?

amitabhskrishworks, 2023-11-06:
yes 
Please note that if you will switch nodes one after another then this problem doesn't occurs.

zhangyanjiaoesp, 2023-11-07:
please provide the IDF branch commit id you used.

amitabhskrishworks, 2023-11-07:
ESP IDF version - v5.0.4
Is IDF branch commit id, other thing ?

zhangyanjiaoesp, 2023-11-07:
> ESP IDF version - v5.0.4 Is IDF branch commit id, other thing ?

OK, so the commit id is https://github.com/espressif/esp-idf/commit/8fbf4ba6058bcf736317d8a7aa75d0578563c38b, I will test on it.

zhangyanjiaoesp, 2023-11-07:
Have tested based on your code for 20 times, still can't reproduce your issue. Can you provide your `sdkconfig` file?

amitabhskrishworks, 2023-11-10:
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=20
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=9
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_RTC_SLOW_CLOCK_SUPPORT_8MD256=y
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_IDCACHE_PER_CORE=y
CONFIG_SOC_MMU_LINEAR_ADDRESS_REGION_NUM=5
CONFIG_SOC_CPU_CORES_NUM=2
CONFIG_SOC_CPU_INTR_NUM=32
CONFIG_SOC_CPU_HAS_FPU=y
CONFIG_SOC_CPU_BREAKPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINT_SIZE=64
CONFIG_SOC_DAC_PERIPH_NUM=2
CONFIG_SOC_DAC_RESOLUTION=8
CONFIG_SOC_GPIO_PORT=1
CONFIG_SOC_GPIO_PIN_COUNT=40
CONFIG_SOC_GPIO_VALID_GPIO_MASK=0xFFFFFFFFFF
CONFIG_SOC_GPIO_VALID_DIGITAL_IO_PAD_MASK=0xEF0FEA
CONFIG_SOC_I2C_NUM=2
[... 1613 lines cut ...]
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_HRT=y
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

amitabhskrishworks, 2023-11-17:
[sdk_config.txt](https://github.com/espressif/esp-idf/files/13391583/sdk_config.txt)

zhangyanjiaoesp, 2023-11-20:
@amitabhskrishworks  Do you use the ESP32 device with PSRAM?

zhangyanjiaoesp, 2023-11-23:
I'm not sure, I will test on the PSRAM devices with the same code and sdkconfig with yours and feedback to you.

amitabhskrishworks, 2023-12-15:
@zhangyanjiaoesp  Thanks for identifying the cause of issue. My router configurations are:
SSID:	EWS-2ND F
Protocol:	Wi-Fi 4 (802.11n)
Security type:	WPA3-Personal
Manufacturer:	Intel Corporation
Description:	Intel(R) Wireless-AC 9560 160MHz
Driver version:	22.0.1.1
Network band:	2.4 GHz
Network channel:	3
Link speed (Receive/Transmit):	144/144 (Mbps)

kdm2223, 2024-01-12:
We also encountered a situation with two root nodes and are hoping for a good solution.

zhangyanjiaoesp, 2024-04-10:
@amitabhskrishworks Sorry to reply you so late. I was held up by something else. I checked the mesh code and found that the default configuration of `esp_mesh_allow_root_conflicts()` is true, so the conflict cannot be resolved when there are two roots in the test. After I call `esp_mesh_allow_root_conflicts(false)` to disable it, the roots conflict can be solved.

Sherry616, 2024-07-17:
Hi @amitabhskrishworks, could you please share your latest updates on this issue? Thanks.

closed 2024-07-31 by Sherry616