espressif/esp-idf #12345: Compilation fails with OpenThread "Enable Joiner" config enabled

baylf2000, 2023-10-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-439-gcb174b0fe1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Expect successful compilation with this option enabled.

### What is the actual behavior?

Compile fails with the errors below involving various `mbedtls_ssl` components.

### Steps to reproduce.

With `esp32c6` selected as the device:

1. Run `idf.py menuconfig` and go to `Component Config - OpenThread` and select `Enable Joiner`.
2. Run `idf.y build`

### Build or installation Logs.

```plain
lf@lf-virtual-machine:~/Documents/esp_projects/light_switch$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/lf/Documents/esp_projects/light_switch/build
Executing "ninja all"...
[0/1] Re-running CMake...-- Component directory /home/lf/esp/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lf/esp/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lf/esp/esp-matter/examples/common/relinker does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32c6
Skipping optional dependency: espressif/cmake_utilities
Processing 13 dependencies:
[13/13] idf (5.1.1)-- Found Python3: /home/lf/esp/esp-tools/python_env/idf5.1_py3.11_env/bin/python3 (found version "3.11.4") found components: Interpreter 
Skipping optional dependency: espressif/cmake_utilities
-- Project sdkconfig file /home/lf/Documents/esp_projects/light_switch/sdkconfig
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at /home/lf/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:795, /home/lf/esp/esp-matter/components/esp_matter/Kconfig:41) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at /home/lf/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:795, /home/lf/esp/esp-matter/components/esp_matter/Kconfig:41) is defined with a prompt outside the choice
Loading defaults file /home/lf/Documents/esp_projects/light_switch/sdkconfig.defaults...
Loading defaults file /home/lf/Documents/esp_projects/light_switch/sdkconfig.defaults.esp32c6...
/home/lf/Documents/esp_projects/light_switch/sdkconfig.defaults.esp32c6:10 CONFIG_BT_NIMBLE_HCI_EVT_BUF_SIZE was replaced with CONFIG_BT_NIMBLE_TRANSPORT_EVT_SIZE
-- App "light_switch" version: 1.0
-- Adding linker script /home/lf/Documents/esp_projects/light_switch/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/lf/esp/esp-idf/components/esp_system/ld/esp32c6/sections.ld.in
-- Adding linker script /home/lf/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.ld
-- Adding linker script /home/lf/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.api.ld
-- Adding linker script /home/lf/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.rvfp.ld
-- Adding linker script /home/lf/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.newlib.ld
-- Adding linker script /home/lf/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.version.ld
-- Adding linker script /home/lf/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.phy.ld
-- Adding linker script /home/lf/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.coexist.ld
-- Adding linker script /home/lf/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.net80211.ld
[... 33 lines cut ...]
/home/lf/esp/esp-idf/components/openthread/openthread/src/core/meshcop/dtls.cpp: In member function 'ot::Error ot::MeshCoP::Dtls::Setup(bool)':
/home/lf/esp/esp-idf/components/openthread/openthread/src/core/meshcop/dtls.cpp:312:5: error: 'mbedtls_ssl_conf_handshake_timeout' was not declared in this scope; did you mean 'mbedtls_ssl_conf_read_timeout'?
  312 |     mbedtls_ssl_conf_handshake_timeout(&mConf, 8000, 60000);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     mbedtls_ssl_conf_read_timeout
/home/lf/esp/esp-idf/components/openthread/openthread/src/core/meshcop/dtls.cpp:321:9: error: 'mbedtls_ssl_conf_dtls_cookies' was not declared in this scope; did you mean 'mbedtls_ssl_conf_dn_hints'?
  321 |         mbedtls_ssl_conf_dtls_cookies(&mConf, mbedtls_ssl_cookie_write, mbedtls_ssl_cookie_check, &mCookieCtx);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         mbedtls_ssl_conf_dn_hints
/home/lf/esp/esp-idf/components/openthread/openthread/src/core/meshcop/dtls.cpp:333:16: error: 'mbedtls_ssl_set_hs_ecjpake_password' was not declared in this scope
  333 |         rval = mbedtls_ssl_set_hs_ecjpake_password(&mSsl, mPsk, mPskLength);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/lf/esp/esp-idf/components/openthread/openthread/src/core/meshcop/dtls.cpp: In member function 'ot::Error ot::MeshCoP::Dtls::SetClientId(const uint8_t*, uint8_t)':
/home/lf/esp/esp-idf/components/openthread/openthread/src/core/meshcop/dtls.cpp:558:16: error: 'mbedtls_ssl_set_client_transport_id' was not declared in this scope; did you mean 'mbedtls_ssl_conf_transport'?
  558 |     int rval = mbedtls_ssl_set_client_transport_id(&mSsl, aClientId, aLength);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                mbedtls_ssl_conf_transport
/home/lf/esp/esp-idf/components/openthread/openthread/src/core/meshcop/dtls.cpp: In member function 'void ot::MeshCoP::Dtls::Process()':
/home/lf/esp/esp-idf/components/openthread/openthread/src/core/meshcop/dtls.cpp:927:17: error: 'mbedtls_ssl_set_hs_ecjpake_password' was not declared in this scope
  927 |                 mbedtls_ssl_set_hs_ecjpake_password(&mSsl, mPsk, mPskLength);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[242/823] Building CXX object esp-idf/openthread/CMakeFiles/__idf_openthread.dir/openthread/src/core/net/netif.cpp.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/lf/Documents/esp_projects/light_switch/build/log/idf_py_stderr_output_105569 and /home/lf/Documents/esp_projects/light_switch/build/log/idf_py_stdout_output_105569
lf@lf-virtual-machine:~/Documents/esp_projects/light_switch$
```

mozzbozz, 2023-10-07:
Can confirm. Same hardware (esp32-c6), basically same IDF version (ESP-IDF v5.1.1-dirty). Coming from a clean config from the in my case light-example (not light switch like above), activating OpenThread and following to that the "Enable Joiner" lets the build process fail with the error messages above. Disabling that "Enable Joiner" option again, leads to the build process successfully working again.

mozzbozz, 2023-10-07:
I tried a bit of stuff. Couldn't get it working, however activating the following CONFIG flags seems to be getting us one step further?

- `CONFIG_MBEDTLS_SSL_PROTO_DTLS` (Component config -> mbedTLS -> Support DTLS protocol (all versions))
- `CONFIG_MBEDTLS_ECJPAKE_C` (Component config -> mbedTLS -> Elliptic curve J-PAKE)
- `CONFIG_MBEDTLS_KEY_EXCHANGE_ECJPAKE` (Component config -> mbedTLS -> TLS Key Exchange Methods -> Enable ECJPAKE based ciphersuite modes)

Seems to clear the errors from, just to break down at the linker with tons of messages like the following:

```
meshcop/dtls.cpp:875: undefined reference to `mbedtls_ssl_read'
```

No idea if the CONFIG flags from above are doing the right things or if it just leads in a wrong direction. Yikes.

baylf2000, 2023-10-16:
Bump. Any update from Espressif on this bug, or if there are any workarounds?

baylf2000, 2023-10-20:
Seems this issue has been fixed since this was logged. I see there has been a few updates pushed. It would have been nice for the maintainers to acknowledge something in this ticket. Maybe the already knew of the issue and just fixed it.

closed 2023-10-20 by baylf2000

chshu, 2023-10-27:
@mozzbozz @baylf2000 The Thread Matter device utilizes the Matter commissioning process to join to the Thread network, doesn't need the Thread Joiner feature. We didn't fix anything recently, the default configurations in the esp-matter examples should work as is.

If you want to try out the Thread Joiner feature, it's necessary to enable specific mbedtls options. You can refer to the configuration in the ot_cli example, available at this link: https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_cli/sdkconfig.defaults#L18, which could support the Joiner feature.

BartJozwicki, 2024-07-11:
Enabling joiner causes the same issue, I am trying to use openThread and join network ESP-IDF v5.2.2 but cannot build project: 

[image]

platini76, 2024-09-18:
I did not activate joiner but I had the same result

arrlee, 2025-03-11:
I am using esp-idf 5.2.5. I had the same problem when creating a new idf project (idf.py create-project <name>) and then enabling Joiner + OpenThread. Setting below fixed it me:

MBEDTLS_SSL_PROTO_DTLS=y # Component config -> mbedTLS -> Support DTLS protocol (all versions)
MBEDTLS_ECJPAKE_C=y # Component config -> mbedTLS -> Elliptic curve J-PAKE
MBEDTLS_KEY_EXCHANGE_ECJPAKE=y   # Component config -> mbedTLS -> TLS Key Exchange Methods -> Enable ECJPAKE based ciphersuite modes