espressif/esp-idf #12346: how to send/receive broadcast via wifi

m986883511, 2023-10-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to esp32 communicate with my pc, but I dont known the ip one the esp32 everytime, so I should use broadcast.
I learn the exmaples in project, check udp_server, it use ip.
I found l2tap can send broadcast, but it need a rj45 interface, I config my meneconfig (example_connect) to wifi, but it run error.

```shell
I (261) cpu_start: App cpu up.
I (300) cpu_start: Pro cpu start user code
I (300) cpu_start: cpu freq: 160000000 Hz
I (301) cpu_start: Application information:
I (304) cpu_start: Project name:     l2tap_example
I (309) cpu_start: App version:      1
I (313) cpu_start: Compile time:     Oct  3 2023 19:11:56
I (319) cpu_start: ELF file SHA256:  b90bc170d...
I (325) cpu_start: ESP-IDF:          v5.2-dev-2317-g3247253671-dirty
I (332) cpu_start: Min chip rev:     v0.0
I (337) cpu_start: Max chip rev:     v0.99
I (341) cpu_start: Chip rev:         v0.1
I (346) heap_init: Initializing. RAM available for dynamic allocation:
I (353) heap_init: At 3FCA0E50 len 000488C0 (290 KiB): DRAM
I (360) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (366) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (372) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (380) spi_flash: detected chip: gd
I (383) spi_flash: flash io: dio
W (387) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (400) sleep: Configure to isolate all GPIO pins in sleep state
I (407) sleep: Enable automatic switching of GPIO sleep configuration
I (414) app_start: Starting scheduler on CPU0
I (419) app_start: Starting scheduler on CPU1
I (419) main_task: Started on CPU0
I (429) main_task: Calling app_main()
I (429) l2tap_example: esp_vfs_l2tap_intf_register start
I (439) l2tap_example: esp_vfs_l2tap_intf_register end
I (449) l2tap_example: nvs_flash_init start
I (459) l2tap_example: nvs_flash_init end
I (459) l2tap_example: nvs_flash_init start
I (459) l2tap_example: nvs_flash_init end
I (469) l2tap_example: esp_event_loop_create_default start
I (469) l2tap_example: esp_event_loop_create_default end
I (479) l2tap_example: example_connect start
E (589) w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
E (589) w5500.mac: emac_w5500_init(715): vefiry chip ID failed
I (589) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (599) esp_eth: esp_eth_driver_install(228): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0x10a (ESP_ERR_INVALID_VERSION) at 0x4200a89a
0x4200a89a: eth_start at /home/ubuntu/esp/esp-idf/examples/common_components/protocol_examples_common/eth_connect.c:153 (discriminator 1)

file: "/IDF/examples/common_components/protocol_examples_common/eth_connect.c" line 153
func: eth_start
expression: esp_eth_driver_install(&config, &s_eth_handle)

abort() was called at PC 0x4037dad3 on core 0
0x4037dad3: _esp_error_check_failed at /home/ubuntu/esp/esp-idf/components/esp_system/esp_err.c:50

Backtrace: 0x40375ada:0x3fca4840 0x4037dadd:0x3fca4860 0x403852aa:0x3fca4880 0x4037dad3:0x3fca48f0 0x4200a89a:0x3fca4920 0x4200a9d4:0x3fca4a30 0x42009c3f:0x3fca4a50 0x42009975:0x3fca4a70 0x42085c4f:0x3fca4ac0 0x4038052d:0x3fca4af0
0x40375ada: panic_abort at /home/ubuntu/esp/esp-idf/components/esp_system/panic.c:452

0x4037dadd: esp_system_abort at /home/ubuntu/esp/esp-idf/components/esp_system/port/esp_system_chip.c:93

0x403852aa: abort at /home/ubuntu/esp/esp-idf/components/newlib/abort.c:38

0x4037dad3: _esp_error_check_failed at /home/ubuntu/esp/esp-idf/components/esp_system/esp_err.c:50

0x4200a89a: eth_start at /home/ubuntu/esp/esp-idf/examples/common_components/protocol_examples_common/eth_connect.c:153 (discriminator 1)

0x4200a9d4: example_ethernet_connect at /home/ubuntu/esp/esp-idf/examples/common_components/protocol_examples_common/eth_connect.c:230

0x42009c3f: example_connect at /home/ubuntu/esp/esp-idf/examples/common_components/protocol_examples_common/connect.c:87

0x42009975: app_main at /home/ubuntu/project/protocols/l2tap/main/l2tap_main.c:261 (discriminator 13)

0x42085c4f: main_task at /home/ubuntu/esp/esp-idf/components/freertos/app_startup.c:217 (discriminator 13)

0x4038052d: vPortTaskWrapper at /home/ubuntu/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

so, how to send/receive broadcast via wifi?

nopnop2002, 2023-10-03:
Using mDNS, you can find out the other party's IP address using the host name or service name.

https://github.com/nopnop2002/esp-idf-mdns

hansw123, 2024-05-17:
@m986883511 
sorry for late reply
you can use udp example  and change  ip address  to255.255.255.255 or network segment address+255

Sherry616, 2024-07-24:
Hi @m986883511, could you please share the latest updates on this issue? Thanks.

m986883511, 2024-07-24:
> Hi @m986883511, could you please share the latest updates on this issue? Thanks.

I will test in few days

m986883511, 2024-08-11:
yes, I test ok

closed 2024-08-11 by m986883511