espressif/esp-idf #12347: esp32: esp_lcd_panel_draw_bitmap breaks if used from a core different from the one that setup the SPI and lcd panel for st7789

greenaddress, 2023-10-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-D0WDQ6-V3 (revision v3.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 T-Display

### Power Supply used.

USB

### What is the expected behavior?

Expected behavior is to be able to use the esp_lcd library with SPI devices such as st7789 from tasks that run on different core from the one that setup the panel/spi.

### What is the actual behavior?

If I setup the panel in main and start a task on app core then the task and display freezes after a few seconds.

If I setup the panel in app core and then run in app core it works.

If I setup the panel in pro core and then run in pro core it works.

If I setup the panel in a app core and start a task on pro core then the task and display freezes after a few seconds.

If I don't pin the task, it will work until the task runs on the wrong core randomly.

### Steps to reproduce.

1. Setup display in main and start a task pinned on the same core that paints the display in an infinite loop of a different color.
2. Verify the display paints constantly.

4. Change the task to be pinned to the other core
5. Verify the display stops painting after no time

### Debug Logs.

```plain
Nothing in the logs
```

### More Information.

This problem does not seem to happen with tft loboris library.

The SPI IRAM interrupt handler for tx and SPI master has been turned off.

greenaddress, 2023-10-03:
Disable in menuconfig under drivers the SPI iram interrupt handler for tx and master

```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h"
#include <inttypes.h>
#include <stdio.h>
#include <string.h>

#include "esp_timer.h"
#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_lcd_panel_io.h>
#include <esp_lcd_panel_ops.h>
#include <esp_lcd_panel_vendor.h>
#include <esp_random.h>
#include <freertos/semphr.h>

#define DISP_PIN_NUM_MISO 2
#define DISP_PIN_NUM_MOSI 19
#define DISP_PIN_NUM_CLK 18
#define DISP_PIN_NUM_CS 5
#define DISP_PIN_NUM_DC 16
#define DISP_PIN_NUM_TCS -1
#define DISP_PIN_NUM_RST 23
#define DISP_PIN_NUM_BCKL 4

#define DISP_X_GAP 40
#define DISP_Y_GAP 53
#define DISP_WIDTH 240
#define DISP_HEIGHT 135
#define LCD_PIXEL_CLOCK_HZ 10000000

#define PIXEL_SIZE uint16_t

static esp_lcd_panel_handle_t panel_handle = NULL;
static SemaphoreHandle_t esp_lcd_mutex = NULL;
static DMA_ATTR PIXEL_SIZE display_buffer[DISP_WIDTH];

void prepare_display(void)
{
    spi_bus_config_t buscfg = {
        .sclk_io_num = DISP_PIN_NUM_CLK,
        .mosi_io_num = DISP_PIN_NUM_MOSI,
        .miso_io_num = DISP_PIN_NUM_MISO,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = DISP_WIDTH * sizeof(uint16_t),
    };

    ESP_ERROR_CHECK(spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO));

    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_spi_config_t io_config = {
        .dc_gpio_num = DISP_PIN_NUM_DC,
        .cs_gpio_num = DISP_PIN_NUM_CS,
        .pclk_hz = LCD_PIXEL_CLOCK_HZ,
        .lcd_cmd_bits = 8,
        .lcd_param_bits = 8,
        .spi_mode = 0,
        .trans_queue_depth = 10,
    };

    ESP_ERROR_CHECK(esp_lcd_new_panel_io_spi((esp_lcd_spi_bus_handle_t)SPI2_HOST, &io_config, &io_handle));

    esp_lcd_panel_dev_config_t panel_config = {
        .reset_gpio_num = DISP_PIN_NUM_RST,
        .rgb_endian = LCD_RGB_ENDIAN_RGB,
        .bits_per_pixel = sizeof(PIXEL_SIZE) * 8,
    };

    ESP_ERROR_CHECK(esp_lcd_new_panel_st7789(io_handle, &panel_config, &panel_handle));
    gpio_config_t bk_gpio_config = { .mode = GPIO_MODE_OUTPUT, .pin_bit_mask = 1ULL << DISP_PIN_NUM_BCKL };
    ESP_ERROR_CHECK(gpio_config(&bk_gpio_config));
    ESP_ERROR_CHECK(gpio_set_level(DISP_PIN_NUM_BCKL, 0));
    ESP_ERROR_CHECK(esp_lcd_panel_reset(panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_init(panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_disp_on_off(panel_handle, true));
    ESP_ERROR_CHECK(esp_lcd_panel_swap_xy(panel_handle, true));
    ESP_ERROR_CHECK(esp_lcd_panel_mirror(panel_handle, true, false));
    ESP_ERROR_CHECK(esp_lcd_panel_invert_color(panel_handle, true));
    ESP_ERROR_CHECK(esp_lcd_panel_set_gap(panel_handle, DISP_X_GAP, DISP_Y_GAP));
    ESP_ERROR_CHECK(gpio_set_level(DISP_PIN_NUM_BCKL, 1));
}

static void othercore(void* ignore)
{
    while (1) {
        const int x = esp_random() % (DISP_WIDTH - 1);
        const int xend = x + (esp_random() % (DISP_WIDTH - x)) + 1;
        const int y = esp_random() % (DISP_HEIGHT - 1);
        const int yend = y + (esp_random() % (DISP_HEIGHT - y)) + 1;

        if (xSemaphoreTake(esp_lcd_mutex, 500 / portTICK_PERIOD_MS) == pdTRUE) {
            esp_fill_random(display_buffer, sizeof(PIXEL_SIZE));
            for (int i = 1; i < xend - x; ++i) {
                display_buffer[i] = display_buffer[0];
            }
            for (int i = 0; i < yend - y; ++i) {
                ESP_ERROR_CHECK(esp_lcd_panel_draw_bitmap(panel_handle, x, y + i, xend, y + i + 1, display_buffer));
            }
            xSemaphoreGive(esp_lcd_mutex);
        } else {
            vTaskDelay(20 / portTICK_PERIOD_MS);
        }
    }
}

void app_main(void)
{
    /* mutex can be removed if using the display from just one task */
    esp_lcd_mutex = xSemaphoreCreateMutex();
    prepare_display();
    /* switch this to 0 for all to work, 1 for all to break */
    int core = 0;
    BaseType_t retval = xTaskCreatePinnedToCore(&othercore, "appcpu", 5 * 1024, NULL, tskIDLE_PRIORITY + 1, NULL, core);
    assert(retval == pdPASS);

    vTaskSuspend(NULL);
}
```

this code shows quickly the problem (tested on T-Display ESP32 from lilygo)

suda-morris, 2023-10-08:
@greenaddress  Thanks for providing the code. I updated it a little bit (mainly changed the GPIO number to make it work on my ST7789).

```c
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h"
#include <inttypes.h>
#include <stdio.h>
#include <string.h>

#include "esp_timer.h"
#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_lcd_panel_io.h>
#include <esp_lcd_panel_ops.h>
#include <esp_lcd_panel_vendor.h>
#include <esp_random.h>
#include <freertos/semphr.h>

#define TEST_LCD_H_RES          240
#define TEST_LCD_V_RES          280

#define TEST_IMG_SIZE (200 * 200 * sizeof(uint16_t))

#define DISP_PIN_NUM_MOSI 4
#define DISP_PIN_NUM_CLK 2
#define DISP_PIN_NUM_CS 0
#define DISP_PIN_NUM_DC 19
#define DISP_PIN_NUM_RST 5
#define DISP_PIN_NUM_BCKL 18

static esp_lcd_panel_handle_t panel_handle = NULL;

void prepare_display(void)
{
    gpio_config_t bk_gpio_config = {
        .mode = GPIO_MODE_OUTPUT,
        .pin_bit_mask = 1ULL << DISP_PIN_NUM_BCKL
    };
    ESP_ERROR_CHECK(gpio_config(&bk_gpio_config));
    ESP_ERROR_CHECK(gpio_set_level(DISP_PIN_NUM_BCKL, 1));

    spi_bus_config_t buscfg = {
        .sclk_io_num = DISP_PIN_NUM_CLK,
        .mosi_io_num = DISP_PIN_NUM_MOSI,
        .miso_io_num = -1,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 100 * 100 * sizeof(uint16_t),
    };
    ESP_ERROR_CHECK(spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO));

    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_spi_config_t io_config = {
        .dc_gpio_num = DISP_PIN_NUM_DC,
        .cs_gpio_num = DISP_PIN_NUM_CS,
        .pclk_hz = 10 * 1000 * 1000,
        .lcd_cmd_bits = 8,
        .lcd_param_bits = 8,
        .spi_mode = 0,
        .trans_queue_depth = 10,
    };

    ESP_ERROR_CHECK(esp_lcd_new_panel_io_spi((esp_lcd_spi_bus_handle_t)SPI2_HOST, &io_config, &io_handle));

    esp_lcd_panel_dev_config_t panel_config = {
        .reset_gpio_num = DISP_PIN_NUM_RST,
        .rgb_ele_order = LCD_RGB_ELEMENT_ORDER_RGB,
        .bits_per_pixel = 16,
    };
    ESP_ERROR_CHECK(esp_lcd_new_panel_st7789(io_handle, &panel_config, &panel_handle));

    ESP_ERROR_CHECK(esp_lcd_panel_reset(panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_init(panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_invert_color(panel_handle, true));
    ESP_ERROR_CHECK(esp_lcd_panel_set_gap(panel_handle, 0, 20));
    ESP_ERROR_CHECK(esp_lcd_panel_disp_on_off(panel_handle, true));

    ESP_ERROR_CHECK(gpio_set_level(DISP_PIN_NUM_BCKL, 0));
}

static void othercore(void *ignore)
{
    uint8_t *img = heap_caps_malloc(TEST_IMG_SIZE, MALLOC_CAP_DMA);
    while (1) {
        uint8_t color_byte = esp_random() & 0xFF;
        int x_start = esp_random() % (TEST_LCD_H_RES - 200);
        int y_start = esp_random() % (TEST_LCD_V_RES - 200);
        memset(img, color_byte, TEST_IMG_SIZE);
        esp_lcd_panel_draw_bitmap(panel_handle, x_start, y_start, x_start + 200, y_start + 200, img);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void app_main(void)
{
    prepare_display();
    int core = 1;
    xTaskCreatePinnedToCore(&othercore, "appcpu", 5 * 1024, NULL, tskIDLE_PRIORITY + 1, NULL, core);
}
```

But I can see the LCD is working as expected no mater the `core` is `0` or `1`.

larryli, 2023-10-08:
> @greenaddress  Thanks for providing the code. I updated it a little bit (mainly changed the GPIO number to make it work on my ST7789).
> 
> But I can see the LCD is working as expected no mater the `core` is `0` or `1`.

Maybe you remove the semaphore related code.

greenaddress, 2023-10-11:
@suda-morris your code does not compile for v5.1.1 (.rgb_ele_order was a new addition not in v5.1.1), did you test the code I provided in v5.1.1? I did not test master for this bug but in v5.1.1 it is present