espressif/esp-idf #12349: Startup SHA-256 com check always fails on ESP32-S3-PICO-1

aircable, 2023-10-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master 5.2.2

### Espressif SoC revision.

Chip is ESP32-S3-PICO-1 (LGA56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-PICO-1 plain board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Application check succeeds during boot

### What is the actual behavior?

SHA-256 comparison failed:
Calculated: 83cf18d959ac1e2acd83fb02df7a5e2399da79caf5c439b8b0ff21b42aee52a9
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...

### Steps to reproduce.

Any application started on that chip does that.

### Debug Logs.

```plain
full boot log:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x29 (SPI_FAST_FLASH_BOOT)
Wake count 1
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x4ec
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad8
load:0x403cc700,len:0x2bac
SHA-256 comparison failed:
Calculated: 83cf18d959ac1e2acd83fb02df7a5e2399da79caf5c439b8b0ff21b42aee52a9
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c987c
```

### More Information.

Also, is there a way to disable the app boot check to make it boot faster?
It fails and boots anyway.

aircable, 2023-10-03:
BOOTLOADER_SKIP_VALIDATE_ALWAYS set to true has no effect.

KonstantinKondrashov, 2023-10-11:
Hi @aircable!
Probably you use this option [CONFIG_ESPTOOLPY_HEADER_FLASHSIZE_UPDATE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esptoolpy-header-flashsize-update). If it is enabled then an SHA256 digest is no longer appended to the image when updating the binary header with the flash size during flashing. Disable it to fix this issue (the log that you see produces the ROM (1st) bootloader).

> Also, is there a way to disable the app boot check to make it boot faster? It fails and boots anyway.

Have you already used the [BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/speed.html?highlight=bootloader_skip_validate_in_deep_sleep#improving-startup-time)? I see that you use deep sleep - it skips image validation only after deep sleep.

aircable, 2023-10-11:
You are so very right. Thank you so much for letting me know. And it even explains it in the comments for this option.
Sorry I missed it.
I was trying to get the fastest boot from deep sleep and I disabled all validations during boot.
You may close the thread.

closed 2023-10-11 by KonstantinKondrashov