espressif/esp-idf #12351: GPIO floating in light sleep

eme-bennytang, 2023-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I am trying out the lightsleep for my ESP32S3.
I set a GPIO to High before going to sleep, but I found that when it enter sleep, the GPIO go down immediately.
After some digging, seems the GPIO will be set to High Impedance after sleep?

Is there any way I can hold some IO to its original state (state before sleep) while sleeping?

Thanks

KonstantinKondrashov, 2023-10-04:
Hi @eme-bennytang!
You may take a look at this chapter in TRM [6.7 RTC IO MUX for Low Power and Analog Input/Output](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf).

You can try to use [gpio_hold_en() gpio_hold_dis()]( https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/gpio.html?highlight=gpio#_CPPv412gpio_hold_en10gpio_num_t).

```c
gpio_hold_dis(num_pin);
...

gpio_set_level(num_pin, 0);
gpio_hold_en(num_pin);
esp_sleep_enable_timer_wakeup(30 * 1000); // 30 sec sleep
esp_light_sleep_start();
gpio_hold_dis(num_pin);

```

esp-wzh, 2023-10-09:
@eme-bennytang If you are using autolight sleep controlled by esp_pm, It is recommended to use `gpio_sleep_sel_dis` to disable automatic switching of gpio mode when entering sleep, because if `PM_SLP_DISABLE_GPIO` or `ESP_SLEEP_GPIO_RESET_WORKAROUND` is enabled, the system will configure all gpio sleep modes to high impedance during [startup](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/sleep_gpio.c#L195C21-L195C21).

refer:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/kconfig.html#config-pm-slp-disable-gpio

esp-wzh, 2024-02-22:
If no more issues, this thread can be closed.

closed 2024-12-17 by Alvin1Zhang