espressif/esp-idf #12356: ESP32H2 deep_sleep and deep_sleep_wake_stub example fails to compile

absoluteabutaj, 2023-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello there, 

When tried to compile the deep_sleep and deep_sleep_wake_stub example target for ESP32H2, it fails at linking, 
```
${PATH_TO_ESP}/.espressif-v5.1.1/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/esp_hw_support/libesp_hw_support.a(sleep_modes.c.obj): in function `esp_deep_sleep_start':
${PATH_TO_ESP}/esp/esp-idf-v5.1.1/components/esp_hw_support/sleep_modes.c:790: undefined reference to `esp_get_deep_sleep_wake_stub'
${PATH_TO_ESP}/.espressif-v5.1.1/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: {PATH_TO_ESP}/esp/esp-idf-v5.1.1/components/esp_hw_support/sleep_modes.c:877: undefined reference to `esp_wake_deep_sleep'
${PATH_TO_ESP}/.espressif-v5.1.1/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: {PATH_TO_ESP}/esp/esp-idf-v5.1.1/components/esp_hw_support/sleep_modes.c:877: undefined reference to `esp_wake_deep_sleep'
${PATH_TO_ESP}/.espressif-v5.1.1/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: {PATH_TO_ESP}/esp/esp-idf-v5.1.1/components/esp_hw_support/sleep_modes.c:877: undefined reference to `esp_set_deep_sleep_wake_stub'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the {PATH_TO_ESP}/esp/esp-idf-v5.1.1/examples/system/deep_sleep/build/log/idf_py_stderr_output_133785 and {PATH_TO_ESP}/esp/esp-idf-v5.1.1/examples/system/deep_sleep/build/log/idf_py_stdout_output_133785
```
Tried light_sleep example and it works as described.

Note: For ESP32C6 target it works as described as well.

KonstantinKondrashov, 2023-10-05:
Hi @absoluteabutaj!
I have tried to build the following examples for master (v5.2-dev-3220-gaa3b9df), they are built without any errors.
- esp-idf/examples/system/deep_sleep_wake_stub
- esp-idf/examples/system/deep_sleep

Please check your IDF version to which chip it supports https://github.com/espressif/esp-idf#esp-idf-release-and-soc-compatibility. Probably you need to update your IDF barch to get it.

absoluteabutaj, 2023-10-05:
Hey @KonstantinKondrashov - Thanks for sharing.

Switching to the `master` branch did the trick. Would you mind sharing the diff between the `tag v5.1.1` and the `master` branch if possible, specific to this issue?

KonstantinKondrashov, 2023-10-05:
@absoluteabutaj
For your barch `tag v5.1.1`, it is not supported, please check the page - https://github.com/espressif/esp-idf/tree/v5.1.1/examples/system/deep_sleep, there is no ESP32-H2 mark (see the picture below). However [the release/v5.1 breach](https://github.com/espressif/esp-idf/tree/release/v5.1/examples/system/deep_sleep) has it. You can try switching to this batch then.

[image]

absoluteabutaj, 2023-10-05:
@KonstantinKondrashov - Thanks for clarifying this.

closed 2023-10-05 by absoluteabutaj