espressif/esp-idf #12357: Esp32s3 LCD_ Camera cannot trigger a DMA interrupt

planeboy817, 2023-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Esp32s3 LCD_ Camera cannot trigger a DMA interrupt
In the case of esp32s3 camera, it was not possible to trigger the DMA interrupt and LCD_ Camera peripheral's vs pin interrupt, look for ll_ Cam. h did not find any errors and cannot find answers to related issues online. May I ask if it is due to the lack of flash and PSRAM versions of the chip in ESP32s3, which do not have DMA functionality, or other bugs （idf    version 5.1）

`static esp_err_t ll_cam_dma_init(cam_obj_t *cam)
{
    for (int x = (SOC_GDMA_PAIRS_PER_GROUP - 1); x >= 0; x--) {
        if (GDMA.channel[x].in.link.addr == 0x0) {
            cam->dma_num = x;
            ESP_LOGI(TAG, "DMA Channel=%d", cam->dma_num);
            break;
        }
        if (x == 0) {
            cam_deinit();
            ESP_LOGE(TAG, "Can't found available GDMA channel");
			return ESP_FAIL;
        }
    }

    if (REG_GET_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_DMA_CLK_EN) == 0) {
        REG_CLR_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_DMA_CLK_EN);
        REG_SET_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_DMA_CLK_EN);
        REG_SET_BIT(SYSTEM_PERIP_RST_EN1_REG, SYSTEM_DMA_RST);
        REG_CLR_BIT(SYSTEM_PERIP_RST_EN1_REG, SYSTEM_DMA_RST);
    }

    GDMA.channel[cam->dma_num].in.int_clr.val = ~0;
    GDMA.channel[cam->dma_num].in.int_ena.val = 0;

    GDMA.channel[cam->dma_num].in.conf0.val = 0;
    GDMA.channel[cam->dma_num].in.conf0.in_rst = 1;
    GDMA.channel[cam->dma_num].in.conf0.in_rst = 0;

    //internal SRAM only
    if (!cam->psram_mode) {
        GDMA.channel[cam->dma_num].in.conf0.indscr_burst_en = 1;
        GDMA.channel[cam->dma_num].in.conf0.in_data_burst_en = 1;
    }

    GDMA.channel[cam->dma_num].in.conf1.in_check_owner = 0;
    // GDMA.channel[cam->dma_num].in.conf1.in_ext_mem_bk_size = 2;

    GDMA.channel[cam->dma_num].in.peri_sel.sel = 5;
    //GDMA.channel[cam->dma_num].in.pri.rx_pri = 1;//rx prio 0-15
    //GDMA.channel[cam->dma_num].in.sram_size.in_size = 6;//This register is used to configure the size of L2 Tx FIFO for Rx channel. 0:16 bytes, 1:24 bytes, 2:32 bytes, 3: 40 bytes, 4: 48 bytes, 5:56 bytes, 6: 64 bytes, 7: 72 bytes, 8: 80 bytes.
    //GDMA.channel[cam->dma_num].in.wight.rx_weight = 7;//The weight of Rx channel 0-15
    return ESP_OK;
}
’

’esp_err_t ll_cam_config(cam_obj_t *cam, const camera_config_t *config)
{
    esp_err_t ret = ESP_OK;
    if (REG_GET_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_LCD_CAM_CLK_EN) == 0) {
        REG_CLR_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_LCD_CAM_CLK_EN);
        REG_SET_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_LCD_CAM_CLK_EN);
        REG_SET_BIT(SYSTEM_PERIP_RST_EN1_REG, SYSTEM_LCD_CAM_RST);
        REG_CLR_BIT(SYSTEM_PERIP_RST_EN1_REG, SYSTEM_LCD_CAM_RST);
        // printf("地址：%d\n",DR_REG_SYSTEM_BASE);
    }
    /*lcd摄像头的一些寄存器初始化操作 dma传输和dma中断的设置*/
    printf("配置摄像头\n");
    // lcd_cam_lc_reg_date.val=33566752;
    LCD_CAM.cam_ctrl.val = 0;

    LCD_CAM.cam_ctrl.cam_clkm_div_b = 0;
    LCD_CAM.cam_ctrl.cam_clkm_div_a = 0;
    LCD_CAM.cam_ctrl.cam_clkm_div_num = 160000000 / config->xclk_freq_hz;
    LCD_CAM.cam_ctrl.cam_clk_sel = 3;//Select Camera module source clock. 0: no clock. 1: APLL. 2: CLK160. 3: no clock.

    LCD_CAM.cam_ctrl.cam_stop_en = 0;
    LCD_CAM.cam_ctrl.cam_vsync_filter_thres = 4; // Filter by LCD_CAM clock
    LCD_CAM.cam_ctrl.cam_update = 0;
    LCD_CAM.cam_ctrl.cam_byte_order = cam->swap_data;
    LCD_CAM.cam_ctrl.cam_bit_order = 0;
    LCD_CAM.cam_ctrl.cam_line_int_en = 0;
    LCD_CAM.cam_ctrl.cam_vs_eof_en = 0; //1: CAM_VSYNC to generate in_suc_eof. 0: in_suc_eof is controlled by reg_cam_rec_data_cyclelen

    LCD_CAM.cam_ctrl1.val = 0;
    LCD_CAM.cam_ctrl1.cam_rec_data_bytelen = LCD_CAM_DMA_NODE_BUFFER_MAX_SIZE - 1; // Cannot be assigned to 0, and it is easy to overflow
    LCD_CAM.cam_ctrl1.cam_line_int_num = 0; // The number of hsyncs that generate hs interrupts
    LCD_CAM.cam_ctrl1.cam_clk_inv = 0;
    LCD_CAM.cam_ctrl1.cam_vsync_filter_en = 1;
    LCD_CAM.cam_ctrl1.cam_2byte_en = 0;
    LCD_CAM.cam_ctrl1.cam_de_inv = 0;
    LCD_CAM.cam_ctrl1.cam_hsync_inv = 0;
    LCD_CAM.cam_ctrl1.cam_vsync_inv = 0;
    LCD_CAM.cam_ctrl1.cam_vh_de_mode_en = 0;

    LCD_CAM.cam_rgb_yuv.val = 0;

#if CONFIG_CAMERA_CONVERTER_ENABLED
    if (config->conv_mode) {
        ret = ll_cam_converter_config(cam, config);
        if(ret != ESP_OK) {
            return ret;
        }
    }
#endif

    LCD_CAM.cam_ctrl.cam_update = 1;
    LCD_CAM.cam_ctrl1.cam_start = 1;

    ret = ll_cam_dma_init(cam);

    return ret;
}’
’void ll_cam_vsync_intr_enable(cam_obj_t *cam, bool en)
{
    LCD_CAM.lc_dma_int_clr.cam_vsync_int_clr = 1;
    if (en) {
        LCD_CAM.lc_dma_int_ena.cam_vsync_int_ena = 1;
        //xiamianjiachulaide
        LCD_CAM.lc_dma_int_ena.lcd_vsync_int_ena = 1;
         printf("中断使能\n");
    } else {
         #**printf("中断失能\n");**
        LCD_CAM.lc_dma_int_ena.cam_vsync_int_ena = 0;
    }
}`

suda-morris, 2023-10-08:
The camera driver is located in the [esp32-camera library](https://github.com/espressif/esp32-camera), you may want to report any issues and questions there.

suda-morris, 2023-10-08:
> May I ask if it is due to the lack of flash and PSRAM versions of the chip in ESP32s3

Have you enabled the `psram_mode`? Where did you allocate the memory of the framebuffer? SRAM or PSRAM?

planeboy817, 2023-10-08:
> > May I ask if it is due to the lack of flash and PSRAM versions of the chip in ESP32s3
> 
> Have you enabled the `psram_mode`? Where did you allocate the memory of the framebuffer? SRAM or PSRAM？
Yes, I have turned on psram mode and set the rate of 80mhz and allocate cache in psram, but I read the esp32s3 document and it seems that dma is first put into internal ram through hub1 and then out of internal ram through hub2, is it because of lack of internal ram, so dma cannot be used

yes，