espressif/esp-idf #12358: Async HTTP client fails after reading 15709 bytes from the response

smadurange, 2023-10-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.1

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 ESP-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

The HTTP client should download and read 48000 bytes into the buffer.

### What is the actual behavior?

When `is_async` is set to true, the download always stops after 15709 bytes with the error `transport_base: esp_tls_conn_read error, errno=No more processes`. I tried increasing the timeout to a very large number (500000 ms), but the result is the same.

### Steps to reproduce.

main.c:

```
#include <freertos/FreeRTOS.h>
#include <freertos/event_groups.h>
#include <freertos/task.h>
#include <freertos/queue.h>

#include <esp_log.h>
#include <esp_netif.h>
#include <esp_system.h>
#include <esp_wifi.h>
#include <esp_tls.h>
#include <esp_http_client.h>

#include <nvs_flash.h>
#include <sys/param.h>
#include <driver/gpio.h>

#define MAXLEN 48000

#define WIFI_SSID       CONFIG_ESP_WIFI_SSID
#define WIFI_PASSWORD   CONFIG_ESP_WIFI_PASSWORD
#define WIFI_MAX_RETRY  CONFIG_ESP_MAXIMUM_RETRY

#define WIFI_CONNECTED_BIT  BIT0
[... 248 lines cut ...]
E (12354) transport_base: esp_tls_conn_read error, errno=No more processes
D (12454) HTTP_CLIENT: data_process=15709, content_length=48000
E (12564) transport_base: esp_tls_conn_read error, errno=No more processes
D (12664) HTTP_CLIENT: data_process=15709, content_length=48000
E (12874) transport_base: esp_tls_conn_read error, errno=No more processes
D (12974) HTTP_CLIENT: data_process=15709, content_length=48000
D (14894) transport_base: remain data in cache, need to read again
...
D (14894) HTTP_CLIENT: data_process=15709, content_length=48000
V (19914) transport_base: poll_read: select - Timeout before any socket was ready!
D (20014) HTTP_CLIENT: data_process=15709, content_length=48000
V (25024) transport_base: poll_read: select - Timeout before any socket was ready!
D (25124) HTTP_CLIENT: data_process=15709, content_length=48000
D (87554) HTTP_CLIENT: data_process=15709, content_length=48000
D (88944) HTTP_CLIENT: data_process=15709, content_length=48000
D (88944) HTTP_CLIENT: Read finish or server requests close
I (88944) app: HTTP_EVENT_ON_FINISH
D (88944) event: no handlers have been registered for event ESP_HTTP_CLIENT_EVENT:5 posted to loop 0x3ffbb1d8
I (88954) app: HTTPS Status = 200, content_length = 48000
I (88964) main_task: Returned from app_main()
```

### More Information.

If I set `is_async` to `false` in the exact same code above, it works (downloads the full 48000-byte file).

hmalpani, 2023-10-26:
Hello @smadurange 
Apologies for the delayed response. I have created a patch to fix this issue. Can you please try with the following patch and let me know if it solves the issue for you.

Thanks!

[IDFGH_11191.txt](https://github.com/espressif/esp-idf/files/13176627/IDFGH_11191.txt)

hmalpani, 2023-11-06:
Hello @smadurange 
Any updates on this issue? Does the patch I shared works for you?

smadurange, 2023-11-09:
@hmalpani I do not have access to the chip at the moment. I will test this next week and get back to you.

smadurange, 2023-11-20:
@hmalpani I tried with the patch, now I get the full 48KB file. However, the `HTTP_EVENT_ON_DATA` event doesn't seem to stop firing after downloading 48KB. I continue to see the log line `I (19828) app: HTTP response size = 48000 bytes`. I'm guessing the `rc != ESP_ERR_HTTP_EAGAIN` condition doesn't evaluate to true in the following `for` loop in the main function:

```
for(;;) {
    rc = esp_http_client_perform(http_client);
    if (rc != ESP_ERR_HTTP_EAGAIN)
        break;
    vTaskDelay((TickType_t) 100 / portTICK_PERIOD_MS);
}
```

hmalpani, 2023-11-20:
Hello @smadurange 

I think the file you are downloading have size greater than 48KB. In that case, client will continue to download the file. If you want to download only 48KB, you can use [partial download](https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/esp_http_client_example.c#L727). Or you can handle the 48KB downloaded chunk in the event handler. Hope this helps.

smadurange, 2023-11-20:
> Hello @smadurange
>
> I think the file you are downloading have size greater than 48KB. In that case, client will continue to download the file. If you want to download only 48KB, you can use [partial [...]

Yes, you are right. Used the wrong file. So, this patch fixes my original issue. Thanks!

closed 2023-11-27 by espressif-bot