espressif/esp-idf #12359: esp_deep_sleep_start hangs if sleep request is rejected

macksal, 2023-10-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom

### Power Supply used.

External 3.3V

### What is the expected behavior?

esp_deep_sleep_start() should put the chip into deep sleep. If the sleep request fails for any reason, it should either return or software-reset the chip.

### What is the actual behavior?

If a deep sleep request fails, the system logs `sleep: Deep sleep request is rejected` and hangs for at least 10 seconds, before resetting with a Timer Group 0 Watchdog reset. (This may be config-dependent). The culprit seems to be this while loop: https://github.com/espressif/esp-idf/blob/e088c3766ba440e72268b458a68f27b6e7d63986/components/esp_hw_support/sleep_modes.c#L883

The loop is entered even if the sleep request fails. 

### Steps to reproduce.

1. Call `esp_sleep_enable_ext0_wakeup(16, 0)`
2. Hold GPIO16 low
3. Call `esp_deep_sleep_start()`

### Debug Logs.

```plain
W (5550) sleep: Deep sleep request is rejected
```
After 10+ second delay, the chip resets and the next line logged is:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037f391
0x4037f391: .check_finish_cmd at --snip--/esp/esp-idf/components/esp_system/port/arch/xtensa/esp_ipc_isr_routines.S:28

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1384
load:0x403c9700,len:0x4
load:0x403c9704,len:0xec0
load:0x403cc700,len:0x3440
entry 0x403c9970
I (208) cpu_start: Multicore app
```
```

### More Information.

This seems to be fixed by 28aec9594a9931cf5995968706968a2a74d0a3b0, which doesn't appear to be released anywhere. That commit makes `esp_deep_sleep_start` actually return if the sleep request fails, instead of locking the CPU in a while loop. This would be acceptable to me.

It also seems reasonable to do a software reset if the sleep request fails because the wakeup condition was already met.

Yardie-, 2023-10-06:
Yes this caused me lots of confusion until I saw this report. Thinking the watchdog reset was because of my code.

KonstantinKondrashov, 2023-10-06:
Hi! Sorry for the inconvenience. I will help with this issue. Now on master, we have two APIs that can run deep sleep (note that __attribute__((__noreturn__)) was removed):
[void esp_deep_sleep_start(void)
void esp_deep_sleep(uint64_t time_in_us)](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/include/esp_sleep.h#L487)
Using these APIs it was expected that it would go to sleep anyway, but adding a reject error makes it a break change.

Right now the behavior is the following:
- master:
  - if deepsleep is rejected then it just returns.

- v5.1:
  - if deepsleep is rejected then it goes to loop and chip is reset by "Timer Group 0 Watchdog" (time is defined in Kconfig).

Our plan was to update these APIs to return `esp_err_t`, and try to avoid breaking changes.

macksal, 2023-10-06:
@KonstantinKondrashov thanks for the info. 2 follow up questions:

- For clarity, which KConfig option defines the TG0 watchdog timeout? Is TG0 the timer source for the _task watchdog_, the RTC watchdog or something else?

- Do you know when this API change will have stable release? My product goes into production within the month. I have OTA capability, but prefer to fix this before release.

closed 2023-10-25 by espressif-bot