espressif/esp-idf #12360: ESP32 S3 TinyUSB problems with USB unplugging detection

SverchokGreen, 2023-10-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.1.1

### Espressif SoC revision.

ESP32 S3

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

I'm using an example tusb_msc from esp-idf-v5.1.1 and using spiflash to store files. My device is expected to be Self-Powered Device.
I did everything as described in this documentation in the section [Self-Powered Device](https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32s2/api-reference/peripherals/usb_device.html)

const tinyusb_config_t tusb_cfg = {
	.device_descriptor = &descriptor_config,
	.string_descriptor = string_desc_arr,
	.string_descriptor_count = sizeof(string_desc_arr) / sizeof(string_desc_arr[0]),
	.external_phy = false,
	.configuration_descriptor = desc_configuration,
	.self_powered = true,
	.vbus_monitor_io = GPIO_NUM_1
};

I expect that when I disconnect the device from USB I will have access to files on the SPI flash. But when I disconnect the device from the PC, the tud_umount_cb() function is not called and tinyusb_msc_storage_in_use_by_usb_host() returns true. Function tud_umount_cb() call only when I  umount device from PC, if remove device without umount from PC there is no event to determine whether the device is connected or not.

### What is the actual behavior?

I expected that using the vbus_monitor_io would generate a connection and disconnection event on the USB bus.

### Steps to reproduce.

Flash tusb_msc example and remove from PC without manual umount.

tore-espressif, 2023-10-23:
hi @SlySquirrel could you please share your HW connection of VBUS to ESP32-S3?

SlySquirrel, 2023-10-23:
Hi, I use hw design similar as https://github.com/wuxx/nanoESP32-S3 with battery supply and VBUS connection
[image]
VBUS connect only R35(20 kOm)/R36(41.2 KOm). When USB connect on IO1 about 3.3V. C24 not present.

tore-espressif, 2023-10-23:
Thanks, I'll try to reproduce the problem with your HW setup and let you know ;)

silverchris, 2024-01-12:
I am also having this issue.
ESP-IDF v5.1.2

If I read the GPIO with gpio_get_level, it returns the expected value with usb connected and disconnected.

tore-espressif, 2024-01-12:
Hi, I'm sorry for the extraordinary slow responses here. I'll get to it ASAP

tore-espressif, 2024-01-12:
Hello @silverchris and @SlySquirrel thank you for your patience.
Indeed, there is a regression in our tinyusb version >= 0.15, we will fix soon.

In the meanwhile, you can use tinyusb v0.14. It contains the same features for MSC device as v0.15.

You can do this be requiring specific version in your `main/idf_component.yml` like this:
``` yml
## IDF Component Manager Manifest File
dependencies:
  espressif/esp_tinyusb: "^1.4.2"
  espressif/tinyusb: '0.14.2'
```

Also please make sure that the pin you are using for VBUS monitoring is not used by any other peripheral. Here's a nice summary for ESP32-S3 https://www.luisllamas.es/en/which-pins-can-i-use-on-esp32-s3/

silverchris, 2024-01-13:
@tore-espressif downgrading to  0.14.2 fixed it for me. Thank you!

diplfranzhoepfinger, 2024-01-22:
Hi, 

Question: 

when the USB Storage is exposed to the Host, the Storage is no more accessible by the Application ?

tore-espressif, 2024-01-22:
> so a "Shared" Access is per Standard not possible right ?

This is not per (USB) Standard but per differences how the MSC host and the application access the memory partition.

The MSC host accesses _memory sectors_ while the application accesses _files_ (implemented in FATFS):

in general it is impossible to ensure file consistency when the disk is mounted both to the app and to the host (Unless both the host and the device mount the FS read-only)
The problem is that if the MSC host updates the FAT, then the FAT FS code running on the device won't "know" that something on disk has changed, and won't "reload" the FAT from disk.

igrr, 2024-01-22:
Just to add, MTP class tries to address this problem. Here is the issue about supporting it in tinyusb: https://github.com/hathach/tinyusb/issues/1079

diplfranzhoepfinger, 2024-01-22:
@igrr very cool

franz-zasso, 2024-01-27:
https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/device/usb_msc_wireless_disk

roma-jam, 2024-02-14:
Hello everyone, 
Thank you for your patience.

> Indeed, there is a regression in our tinyusb version >= 0.15, we will fix soon.

New version is available: [espressif/tinyusb^0.15.0~5](https://components.espressif.com/components/espressif/tinyusb)
Please update the manifest file accordingly.

closed 2024-03-11 by roma-jam