espressif/esp-idf #12361: HMAC cannot set the key value to efuse.

nikhil-robinson, 2023-10-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to run the HMAC message  encryption  but not able to save the key to the efuse. not getting  ESP_OK in return.

```c
#include "esp_efuse.h"
#include "esp_log.h"
#include "esp_err.h"
#include "esp_random.h"
#include "esp_hmac.h"
#include "esp32s3/rom/efuse.h"

void app_main()
{
    int ets_status = ets_efuse_write_key(ETS_EFUSE_BLOCK_KEY4,
                        ETS_EFUSE_KEY_PURPOSE_HMAC_UP,
                        key_data, sizeof(key_data));
    
    switch (ets_status)
    {
    case ESP_OK:
        ESP_LOGI("EN","Successful");
        break;
    case ESP_ERR_INVALID_ARG:
        ESP_LOGE("EN","Error in the passed arguments");
        break;
    case ESP_ERR_INVALID_STATE:
        ESP_LOGE("EN","Error in efuses state, unused block not found");
        break;
    case ESP_ERR_EFUSE_REPEATED_PROG:
        ESP_LOGE("EN","Error repeated programming of programmed bits is strictly forbidden");
        break;
    case ESP_ERR_CODING:
        ESP_LOGE("EN","Error range of data does not match the coding scheme.");
        break;
    default:
        ESP_LOGE("EN","%s",esp_err_to_name(ets_status));
        break;
    }

}

```

##LOG

```

--- esp-idf-monitor 1.2.1 on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (121) esp_image: segment 1: paddr=0002d384 vaddr=3fc916ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403784be
0x403784be: esp_cpu_wait_for_intr at /home/nikhi/idf5/esp/esp-idf/components/esp_hw_support/cpu.c:121

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (27) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (28) boot: compile time Oct  6 2023 16:59:52
I (28) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=1d35ch (119644) map
I (121) esp_image: segment 1: paddr=0002d384 vaddr=3fc91600 size=02908h ( 10504) load
I (124) esp_image: segment 2: paddr=0002fc94 vaddr=40374000 size=00384h (   900) load
I (127) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=48860h (297056) map
I (189) esp_image: segment 4: paddr=00078888 vaddr=40374384 size=0d23ch ( 53820) load
I (207) boot: Loaded app from partition at offset 0x10000
I (208) boot: Disabling RNG early entropy source...
I (219) cpu_start: Multicore app
I (219) cpu_start: Pro cpu up.
I (220) cpu_start: Starting app cpu, entry point is 0x40375344
0x40375344: call_start_cpu1 at /home/nikhi/idf5/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (237) cpu_start: Pro cpu start user code
I (237) cpu_start: cpu freq: 160000000 Hz
I (238) cpu_start: Application information:
I (240) cpu_start: Project name:     PeeCee-V1.0
I (246) cpu_start: App version:      f6edc7c-dirty
I (251) cpu_start: Compile time:     Oct  6 2023 16:59:45
I (258) cpu_start: ELF file SHA256:  d3f987a46cb5dbaa...
I (264) cpu_start: ESP-IDF:          v5.1.1-dirty
I (269) cpu_start: Min chip rev:     v0.0
I (273) cpu_start: Max chip rev:     v0.99 
I (278) cpu_start: Chip rev:         v0.1
I (283) heap_init: Initializing. RAM available for dynamic allocation:
I (290) heap_init: At 3FC95DD8 len 00053938 (334 KiB): DRAM
I (297) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (303) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (309) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (317) spi_flash: detected chip: generic
I (320) spi_flash: flash io: dio
W (324) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (339) sleep: Configure to isolate all GPIO pins in sleep state
I (344) sleep: Enable automatic switching of GPIO sleep configuration
I (351) app_start: Starting scheduler on CPU0
I (356) app_start: Starting scheduler on CPU1
I (356) main_task: Started on CPU0
I (366) main_task: Calling app_main()
E (366) EN: ERROR
I (376) main_task: Returned from app_main()

```

## SETUP

Target esp32s3
Idf version: ESP-IDF v5.1.1-dirty

##sdkconfig

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/12831570/sdkconfig.txt)

KonstantinKondrashov, 2023-10-06:
Hi @nikhil-robinson!
Before to burn a efuse block this functions checks whether the given efuse block is unused or not. Keep in mind that it will retun OK only the very first time. Could you check the efuse BLOCK_KEY4 first, is it empty? Run the `espefuse.py summary` command (provide the summary log).

nikhil-robinson, 2023-10-07:
Hi @KonstantinKondrashov 

```
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/421.
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/421.
/home/nikhi/idf5/esp/esp-idf/components/esptool_py/esptool/espefuse.py: 11: Syntax error: word unexpected (expecting ")")

````

this is the output of ``` espefuse.py summary ```

When i use the function  ``` esp_efuse_write_key ``` i am getting this error.

```
I (364) main_task: Calling app_main()
I (364) efuse: Batch mode of writing fields is enabled
I (374) efuse: Batch mode of writing fields is cancelled
E (384) EN: Error in efuses state, unused block not found
I (384) main_task: Returned from app_main()

```

KonstantinKondrashov, 2023-10-09:
`Error in efuses state, unused block not found` - This log means that your chip already uses the given efuse block (it can mean that 1-it is already written by you or 2-it was used before). You need to take a look at the efuse summary (run `espefuse.py summary`), the log that you gave is an error. I do not know why. Please fix it to be able to see the whole picture of the efuses.

KonstantinKondrashov, 2023-10-11:
I close this issue now. Feel free to reopen it if any remaining questions.

closed 2023-10-11 by KonstantinKondrashov

nikhil-robinson, 2023-10-18:
> `Error in efuses state, unused block not found` - This log means that your chip already uses the given efuse block (it can mean that 1-it is already written by you or 2-it was used before). You need to take a look at the efuse summary (run `espefuse.py summary`), the log that you gave has an error. I do not know why. Please fix it to be able to see the whole picture of the efuses.
> 
> Probably you can use the next efuse block BLOCK_KEY5 but I would recommend first taking a look at the efuse summary.

if you are not able solve please let someone else look into the issue.

KonstantinKondrashov, 2023-10-18:
Hi @nikhil-robinson!
Your original issue is that you are trying to write to an efuse block that is already written. Please find an unused one, it should resolve your issue, or if you want help from our side provide the[ `espefuse.py summary`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/efuse.html#espefuse-py) (it shows the state of all efuse blocks).

Once BLOCK_KEY5  block is written the following attempts to burn into the same block lead to an error.

> import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/421.
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/421.
/home/nikhi/idf5/esp/esp-idf/components/esptool_py/esptool/espefuse.py: 11: Syntax error: word unexpected (expecting ")")

Regarding this error please open a separate issue, seems something is not installed well, please take a look the [Installation chapter](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/index.html#installation)
Thanks.