espressif/esp-idf #12362: Find replacement for CONFIG_MBEDTLS_ALLOW_UNSUPPORTED_CRITICAL_EXT

torkleyy, 2023-10-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This flag is currently without any effect, as it has been removed upstream (see https://github.com/Mbed-TLS/mbedtls/pull/4477).

This feature is crucial as major vendors of certificates use such unsupported extensions (see https://github.com/Mbed-TLS/mbedtls/issues/2605).

I think there are two options to replace this functionality:

1. Replicate this functionality by implementing the callback (`mbedtls_x509_crt_parse_der_with_ext_cb`) and ignore unknown extensions (this is deemed insecure by mbedtls maintainers)
2. Allow the user to service such a callback

mahavirj, 2023-10-11:
Thanks for notifying. We will remove this config from ESP32 mbedtls port. However, the alternative solution for this must be discussed in the upstream community itself. One of the suggestion can be found here: https://github.com/Mbed-TLS/mbedtls/pull/3564#issuecomment-674825340

mahavirj, 2023-10-25:
Fixed with https://github.com/espressif/esp-idf/commit/0c3ed4f540cc6ecd9d148cc2fd1767eeb7d860a3

closed 2023-10-25 by mahavirj